riscv-collab/riscv-gnu-toolchain #184: Float ABI for struct mixed with int and float

kito-cheng, 2016-10-03:
Hi all:

I am testing RV32G for gcc, and confuse about the ABI for pass such struct:

```
typedef struct { float f; int i; } Sfi;

Sfi g1sSfi;
void checkSfi (Sfi x, int i)  __attribute__ ((noinline,noclone));
void checkSfi (Sfi x, int i)
{
  if (x.f != (float)i || x.i != i+1)
    __builtin_abort ();
}

void checkgSfi (void) __attribute__ ((noinline,noclone));
void checkgSfi (void)
{
  checkSfi ( g1sSfi, 1);
}
int main()
{
  g1sSfi.i = 2;
  g1sSfi.f = 1;
  checkgSfi();
  return 0;
}
```

and `struct Sfi` pass in `fa0` and `a1` in current gcc implementation,

I am ambiguous the struct part for the hard float part, can any one explain how the struct pass in hard float?

aswaterman, 2016-10-03:
I think it's doing the right thing...

See page 7 of ftp://www.linux-mips.org/pub/linux/mips/doc/ABI/MIPS-N32-ABI-Handbook.pdf, on which our ABI is based.  "If a chunk consists solely of a double float field (but not a double,
which is part of a union), it is passed in a floating point register" -- this is for a 64-bit machine, so replace "double" with "float" for RV32G, and you get this behavior.

We're open to changing this behavior if there's a good reason.

kito-cheng, 2016-10-04:
ok, thanks :)

closed 2016-10-04 by kito-cheng