riscv-collab/riscv-gnu-toolchain #185: RV32IMA build is broken

kito-cheng, 2016-10-14:
RV32IMA build is broken after https://github.com/riscv/riscv-gnu-toolchain/commit/55f83087a711a24b8704309b0dd32e6071fcafad

```
$ /home/kito/riscv-workspace/riscv-gnu-toolchain/configure --prefix=/home/kito/riscv-workspace/rv32i --with-xlen=32 --disable-float
$ make linux
...
/home/kito/riscv-workspace/rv32i/lib/gcc/riscv32-unknown-linux-gnu/6.1.0/../../../../riscv32-unknown-linux-gnu/bin/ld: /home/kito/riscv-workspace/rv32i-x/build-glibc-linux32/csu/init.o: can't link hard-float modules with soft-float modules
/home/kito/riscv-workspace/rv32i/lib/gcc/riscv32-unknown-linux-gnu/6.1.0/../../../../riscv32-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /home/kito/riscv-workspace/rv32i-x/build-glibc-linux32/csu/init.o
collect2: error: ld returned 1 exit status
Makefile:94: recipe for target '/home/kito/riscv-workspace/rv32i-x/build-glibc-linux32/csu/crt1.o' failed
make[3]: *** [/home/kito/riscv-workspace/rv32i-x/build-glibc-linux32/csu/crt1.o] Error 1
make[3]: Leaving directory '/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-glibc/csu'
Makefile:214: recipe for target 'csu/subdir_lib' failed
make[2]: *** [csu/subdir_lib] Error 2
make[2]: Leaving directory '/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-glibc'
Makefile:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/kito/riscv-workspace/rv32i-x/build-glibc-linux32'
Makefile:158: recipe for target 'stamps/build-glibc-linux32' failed
make: *** [stamps/build-glibc-linux32] Error 2
```

kito-cheng, 2016-10-14:
Btw, RV32G, RV64IMA RV64G is ok

palmer-dabbelt, 2016-10-14:
I think I at least have a workaround: pass "--with-arch=RV32IMA" instead of "--with-xlen=32 --disable-float"
````
$ ../configure --prefix=$RISCV --enable-linux --with-arch=RV32IMA
$ make
...
$ ./install/bin/riscv32-unknown-linux-gnu-objdump -d ./install/sysroot/lib32/libc-2.24.so | grep flw
````

There's two commits inside the submodule bumps that look like they could be relevant to me:
https://github.com/riscv/riscv-binutils-gdb/commit/b0946801b50b74d7b383b85e71173626d72847da
https://github.com/riscv/riscv-gcc/commit/657f5609d0f4128aa56f6f0f8145295ae5ab3861

The fact that changing "--with-arch" changes the behavior makes me think it's the GCC commit, as that's the one that explicitly parses an ISA string.  I'm trying this
````
diff --git a/gcc/config.gcc b/gcc/config.gcc
index 6f03b6a..c327d07 100644
--- a/gcc/config.gcc
+++ b/gcc/config.gcc
@@ -4000,12 +4000,10 @@ case "${target}" in
                        # Default to soft-float unless D extension is present
                        # (which is also the case if with_arch is not set).
                        case ${with_arch} in
-                               "")
-                                       ;;
                                *g* | *G* | *d* | *D*)
                                        with_float=hard
                                        ;;
-                               *)
+                               "" | *)
                                        with_float=soft
                                        ;;
                        esac
````
under the assumption that the comment is correct, but it's just a guess (I don't know what with_float="" means).

palmer-dabbelt, 2016-10-14:
OK, it looks like it's working for me: https://github.com/riscv/riscv-gcc/commit/1ab8b1e5f04c36e785b5962b5c7b7f127f71809d

kito-cheng, 2016-10-16:
It's break more build, the https://github.com/riscv/riscv-gcc/commit/1ab8b1e5f04c36e785b5962b5c7b7f127f71809d seem treat empty ISA string as IMA not G.

However my RV32IMA still build fail after follow change, and make RV64IMA build fail:
```
diff --git a/gcc/config.gcc b/gcc/config.gcc
index c327d07..766e401 100644
--- a/gcc/config.gcc
+++ b/gcc/config.gcc
@@ -4000,10 +4000,10 @@ case "${target}" in
                        # Default to soft-float unless D extension is present
                        # (which is also the case if with_arch is not set).
                        case ${with_arch} in
-                               *g* | *G* | *d* | *D*)
+                               "" | *g* | *G* | *d* | *D*)
                                        with_float=hard
                                        ;;
-                               "" | *)
+                               *)
                                        with_float=soft
                                        ;;
                        esac

```

kito-cheng, 2016-10-17:
The root cause seem the gcc no longer pass -msoft-float/-mhard-float, it's make *.s / *.S get wrong ABI flag, however it's introduce new test fail after fix [1].

```
                                    |     rv32ima   |       rv32g   |     rv64ima   |       rv64g   | 
gcc       # of expected failures    |   127 (    0) |   127 (    0) |   129 (    0) |   129 (    0) | 
          # of expected passes      | 82386 (    5) | 82393 (    8) | 83802 (    5) | 83791 (  -26) | 
          # of unexpected failures  |    62 (    0) |    69 (   -8) |    77 (    0) |   120 (   28) | 
          # of unexpected successes |     0 (    0) |     0 (    0) |     0 (    0) |     0 (    0) | 
          # of unresolved testcases |    10 (    0) |    10 (    0) |    10 (    0) |    20 (   10) | 
          # of unsupported tests    |  1643 (   -2) |  1636 (    0) |  1514 (   -2) |  1499 (    0) | 
g++       # of expected failures    |   303 (    0) |   303 (    0) |   303 (    0) |   303 (    0) | 
          # of expected passes      | 90602 (    0) | 90602 (    5) | 92482 (    0) | 92482 (    0) | 
          # of unexpected failures  |    30 (    0) |    30 (   -5) |    22 (    0) |    22 (    0) | 
          # of unexpected successes |     2 (    0) |     2 (    0) |     2 (    0) |     2 (    0) | 
          # of unresolved testcases |     0 (    0) |     0 (    0) |     0 (    0) |     0 (    0) | 
          # of unsupported tests    |  3876 (    0) |  3876 (    0) |  3853 (    0) |  3853 (    0) | 
```

[1] https://github.com/kito-cheng/riscv-gcc/commits/empty-isa-string
  - include fix in prev comment

kito-cheng, 2016-10-19:
Can you update document in `configure.ac` and consider to remove --with-xlen= and  --disable-float option from configure option if --with-arch is recommended.

Document for `Test Suite` is need to update too since configure with --disable-float will generate floating point instruction now, but it's not support by gdb-sim yet.

kito-cheng, 2016-10-20:
https://github.com/riscv/riscv-gcc/pull/9 is make it build-able with `--with-arch=RV32IMA`, but the new fail is not resolve yet, those fail will fix with `long double`.