riscv-collab/riscv-gnu-toolchain #186: Consider changing the multilib setup

sorear, 2016-10-14:
Fedora uses `/lib` for all 32-bit arches and `/lib64` for all 64-bit arches.  This is unlikely to change, so we are currently floating this patch against gcc:

```
--- gcc/config/riscv/t-linux64.riscv-fedora     2016-10-06 18:09:04.000000000 +0000
+++ gcc/config/riscv/t-linux64  2016-10-06 18:42:40.000000000 +0000
@@ -2,7 +2,7 @@
 
 MULTILIB_OPTIONS = m64/m32 mno-atomic
 MULTILIB_DIRNAMES = 64 32 no-atomic
-MULTILIB_OSDIRNAMES = ../lib ../lib32 no-atomic
+MULTILIB_OSDIRNAMES = ../lib64 ../lib no-atomic
 
 ifneq ($(with_float), soft)
 MULTILIB_OPTIONS += msoft-float
```

I have no good estimate of the number of users that want `/lib64` versus `/lib32` versus don't care; if most of them are like Fedora it might make sense to change this upstream.

palmer-dabbelt, 2016-10-15:
I'm OK with this.  I think @kito-cheng just had to add a workaround for this into the Makefile.in.  Can you submit a PR with what you want to GCC, and do you mind giving the Makefile.in a once-over to nuke any oddness that may have crept in?

sorear, 2016-10-15:
It turns out that this patch is only half of the story, we also need to change the program interpreter to point at `/lib64/ld.so.1`.  I'll try to put together a PR later this weekend.  Do you have an oddness in mind besides lib32 stuff?

palmer-dabbelt, 2016-10-15:
Now that I'm actually awake, I remember someone has patches in-flight about this: https://github.com/riscv/riscv-gnu-toolchain/pull/136 .   We should definitely get this fixed before upstreaming, but I suggest we move the discussion to the PR so I'm going to close this.

closed 2016-10-15 by palmer-dabbelt