riscv-collab/riscv-gnu-toolchain #187: Generated FPU instruction even -msoft-float

kito-cheng, 2016-10-18:
Source code: https://github.com/riscv/riscv-gnu-toolchain/commit/55f83087a711a24b8704309b0dd32e6071fcafad + https://github.com/kito-cheng/riscv-gcc/commit/93fb18c8e83badfee39ec99f24df73a1842ac31f

```
float foo(float a, float b){
  return a+b;
}
```

```
$ riscv64-unknown-linux-gnu-gcc float_add.c -o - -S  -msoft-float 
	.file	"float_add.c"
	.option nopic
	.option soft-float
	.text
	.align	2
	.globl	foo
	.type	foo, @function
foo:
	add	sp,sp,-32
	sd	s0,24(sp)
	add	s0,sp,32
	sw	a0,-20(s0)
	sw	a1,-24(s0)
	flw	fa4,-20(s0)
	flw	fa5,-24(s0)
	fadd.s	fa5,fa4,fa5
	fmv.x.s	a0,fa5
	ld	s0,24(sp)
	add	sp,sp,32
	jr	ra
	.size	foo, .-foo
	.ident	"GCC: (GNU) 6.1.0"
```

kito-cheng, 2016-10-18:
@aswaterman  It's seem need `-mno-float` option to disable floating point instruction now?

aswaterman, 2016-10-18:
> Source code: 55f8308
> + kito-cheng/riscv-gcc@93fb18c

-msoft-float describes calling convention, not ISA. If -march contains F,
the instructions are still legal. But arguments won't be passed/returned in
FP registers.

This is like soft-fp in ARM. Maybe we should change the GCC flags to be
more clear?

aswaterman, 2016-10-18:
Or -march doesn't have F.

kito-cheng, 2016-10-19:
My point is your change the backward compatibility, it's control float-point instruction generate or not before this commit https://github.com/riscv/riscv-gcc/commit/90bfbdd4bc88dc7b5e4fcfa0e0a3e6a1e832cd30, and ABI should control by -mabi= or -mfloat-abi= instead of -msoft-float and -mhard-float[1], it's a confusing name.

ARM's -msoft-float and -mhard-float are alias to -mfloat-abi= for backward compatibility and it's mark as undocumented for long time.

aswaterman, 2016-10-19:
> riscv/riscv-gcc@90bfbdd

So what do propose? I'm not too concerned about backwards compatibility at
this point.

I think -march is the most sensible way to control emission of FP
instructions for RISC-V. And we need some flag for ABI. That's why I picked
this scheme.

We could deprecate hard/soft and use -mfloat-abi, but that overlaps -march,
which is confusing. We would also need more settings for -mfloat-abi to
control single vs. double, instructions and calling conventions (so 6
options in total, I think).

kito-cheng, 2016-10-19:
I prefer to remove -msoft-float, -mhard-float if we don't care backwards compatibility since it's seem redundant and confusing if -march decide ABI.

And another suggestions for https://github.com/riscv/riscv-gcc/commit/90bfbdd4bc88dc7b5e4fcfa0e0a3e6a1e832cd30:

[1] Do not use InverseMask for `-mdouble-float` and `-msingle-float`, let `-mdouble-float` imply `-msingle-float`, and `-mquad-float` imply `-mdouble-float` and `-msingle-float` in future.
[2] Remove `-m[no-]float`, just check `-msingle-float`, `-mdouble-float`, `-mquad-float`

palmer-dabbelt, 2016-10-19:
I like getting rid of -m{hard,soft}-float in favor of -march, they've been flaky for a while.  I think passing -mabi is a sane way to do it.  The idea would be that "-march" controls the ISA subset that can be used to generate code, and "-mabi" determines the ABI to emit.

aswaterman, 2016-10-24:
I'm back from vacation.  I'll get this done ASAP, but first I want to be clear on what we're doing.  Here is the revised proposal:

- -march controls which instructions may be emitted, and does not affect ABI.
- -mfloat-abi controls which calling convention is used: "soft" means no args passed in registers; "single" means only single-precision values are passed in regisers; "double" means single- and double-precision values are passed in registers; and "quad" means single-, double-, and quad-precision values are passed in registers.
- GCC will issue an error if -mfloat-abi requires an extension not provided by -march, but vice-versa is OK.
- ELF flags will be expanded to cover the 4 float-abi cases.
- Default float-abi for RVI and RVIF is "soft"; default for RVIFD is "double".  "single" and "quad" are never used as default ABIs.

aswaterman, 2016-10-24:
I'm fine with -msingle/double/quad-float (without InverseMask), which cause F, FD, and FDQ to be enabled.  I think we'd still need -mno-float, which removes F.

commit a1e7a925 (2016-10-25): Overhaul floating-point command-line arguments

- -march controls which instructions may be emitted, and does not
affect ABI.
- -mno-float is equivalent to passing neither F for D to -march;
-msingle-float is equivalent to passing F but not D to -march;
and -mdouble-float is equivalent to passing F and D to -march.
- -mfloat-abi controls which calling convention is used: "soft" means
no args passed in registers; "single" means only single-precision
values are passed in regisers; "double" means single- and
double-precision values are passed in registers.
- -mfloat-abi defaults to "soft" if the  [... cut]
closed 2016-10-25 by aswaterman