riscv-collab/riscv-gnu-toolchain #188: Compilation with --disable-float fails on newlib/libc/machine/riscv/setjmp.S

TM1234, 2016-10-31:
See here:

https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/zPHwe8ih534

To summarise...

git clone --recursive  https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/toolchain --enable-multilib --disable-float
make

results in 

/home/user/Downloads/riscv/sifive/riscv-gnu-toolchain/src/newlib-gcc/newlib/libc/machine/riscv/setjmp.S:22:4: error: #error unsupported FLEN
 #  error unsupported FLEN
    ^~~~~
make[10]: *** [lib_a-setjmp.o] Error 1

Reverting to the previous version of setjmp.S allows the tools to build OK so the problem seems to be with the a1e7a925c16f8ff34c5a6d794bfe9b9d5660268f commit.

I don't know whether or not this is related or is a separate issue but even when I successfully build the tools with --disable-float using the previous version of setjmp.S I cannot link my program which passes -msoft-float to all tools (assemble, compile, link):

gcc/bin/../lib/gcc/riscv64-unknown-elf/6.1.0/../../../../riscv64-unknown-elf/lib/32/soft-float/libc.a(lib_a-memset.o): can't link hard-float modules with soft-float modules

aswaterman, 2016-10-31:
I just tried the same thing without issue.  However, I note that those macros used in setjmp.S were introduced very recently (in the commit you referenced).  I wonder if newlib is somehow being built with an older version of gcc.  Can you nuke all installed versions of gcc and try again?

If you look here https://github.com/riscv/riscv-gcc/blob/riscv-gcc-6.1.0/gcc/config/riscv/riscv.h#L72 you'll see that one of those three macros should definitely be defined.

With respect to your linking issue: if you provide --disable-float, then the hard-float multilibs don't get built.  If you drop that flag, you shouldn't get link errors anymore.

TM1234, 2016-10-31:
Hi Andrew - thanks for the update. I'm going to try again with a clean CentOS 6.8 virtual machine which will definitely have no vestigial RISC-V gcc tools. I'll post back later with results.

I'm not sure that I understand your second comment about the link problem. I presumed that --disable-float meant "disable hard float" so that the libs would use soft float support and my program code should be compiled with -msoft-float and should link with libs using only soft float support? My target RISC-V CPU implementation does not have hard float support.

aswaterman, 2016-10-31:
The linking problem may well be resolved by getting the most up-to-date gcc building.

TM1234, 2016-10-31:
OK - thanks Andrew. Still cloning the repo and will kick off a build and try it ASAP and report back.

TM1234, 2016-11-01:
OK - a clean clone, configure and build worked and the tools build successfully.
I had previously done a git pull to update my sources, make clean, configure and make but it seems as if this did not guarantee a clean build?
I didn't have any RISC-V tools on the path that would have been picked up during the build process.
And the toolchain build was 1GB+ in size whereas the totally clean one that I did just now was only 600MB+.

My remaining question is how can I build the tools so that the libs use soft and not hard float support?
I guess I misunderstood --disable-float and read it as --disable[-hard]-float!
So when I compile now with -msoft-float I get this:

riscv64-unknown-elf-gcc: error: unrecognized command line option '-msoft-float'; did you mean '-mno-float'?

Do I need to do --enable-soft-float (AND --disable-hard-float?!)?

TM1234, 2016-11-01:
Apologies for the running commentary but when I remove -msoft-float from the command line the program compiles and links OK. But it seems odd that if soft-float is built in (via configure --disable-float) then the compiler complains if passed msoft-float? Admittedly it's redundant if the only option is soft-float anyway but surely it's not incorrect per se?

By the way I'm using the Roa Logic RISC-V Eclipse plugins so had to modify the plugin to stop it adding -msoft-float to the command line(s):

https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/8dae3T-fS8A

aswaterman, 2016-11-01:
We changed the API.  -msoft-float and -mhard-float are no longer options; we removed them to clearly delineate the calling convention from the ISA.  The calling convention is selected by -mfloat-abi=soft, -mfloat-abi=single, or -mfloat-abi=double.  Independently, the ISA is set with -march.

TM1234, 2016-11-01:
Thanks a lot Andrew - that clarifies things a lot for me.
So in my case should I actually be using -mfloat-abi=soft?
Is -mfloat-abi=soft tools/libs built with configure --disable-float?
And in the general case should always specify one of the three -mfloat-abi=x options?
What does no -mfloat-abi=x option mean?

TM1234, 2016-11-01:
Hi Andrew - perhaps if you get a chance you could comment here to try to help eliminate confusion?

https://groups.google.com/a/groups.riscv.org/d/msg/sw-dev/8dae3T-fS8A/K1yqHG0yIgAJ

On the other hand if there is some spec/doc that already deals with these issues just point me at it.

Thanks a lot.

aswaterman, 2016-11-01:
I'm actually not entirely clear about what --disable-float is *supposed* to do.

The important thing (which I think is already understood in that google thread) is that -mfloat-abi sets the calling convention and -march sets which instructions are available.  It's possible to have floating-point instructions with a soft-float ABI (but impossible to have a hard-float ABI without floating-point instructions).

TM1234, 2016-11-01:
Thanks Andrew. Apologies if this is a dumb question or one that has already been covered but...

My target RISC-V CPU has no FP unit or f regs so (a) how should I build the RISC-V gcc etc. tools and (b) what -march and -mfloat-abi should I compile my code with?

aswaterman, 2016-11-01:
You can just pass -mno-float, which is the same as -mfloat-abi=soft and -march=[something that doesn't contain F or D, e.g., RV32IM].

aswaterman, 2016-11-01:
Also, it's hard to call anything a dumb question when the documentation is so lacking :)

TM1234, 2016-11-01:
Thanks a lot Andrew.
Just to be clear this (below) will work with the RISC-V gcc etc. tools built with configure --disable-float etc.?
My RISC-V is RV32IM as it happens.

> You can just pass -mno-float, which is the same as -mfloat-abi=soft and -march=[something that doesn't contain F or D, e.g., RV32IM].

I'm not clear if -mno-float is effectively redundant if it's the same as -mfloat-abi=soft?
I don't see how -march=... affects things here even if the arch does have "F/D".
After all presumably -mfloat-abi=soft is presumably allowed with an "F/D" arch even if that ignores hardware capabilities of the target?
(Maybe I'm overthinking this!?! :-|)

aswaterman, 2016-11-01:
-mno-float _is_ redundant with the combination of -mfloat-abi=soft and -march=RV32IM.  It's just there for convenience, and to simplify the multilib stuff.

closed 2016-11-14 by aswaterman

Kendidi, 2021-06-17:
I used the following to rebuild toolchain.  Does it looks OK?

    ./configure --prefix=/opt/riscv32 --with-arch=rv32imfc --with-abi=ilp32f
    sudo make linux

jim-wilson, 2021-06-17:
"make linux" will build a riscv32-unknown-linux-gnu-gcc which is not what you want.  make linux means the target is linux.  This won't build because the linux toolchain does not support single float only.  Also, linux requires the A (atomic) extension, so the build will also fail because that is missing.  You want to run "make" instead which will build a riscv32-unknown-elf-gcc.