riscv-collab/riscv-gnu-toolchain #193: Segment fault when relax enable

kito-cheng, 2016-11-04:
GCC: riscv/riscv-gcc@bfaf705
Binutils: riscv/riscv-binutils-gdb@6756174
Glibc: riscv/riscv-glibc@7fef429
Arch: rv32i, rv32g, rv64i and rv64g

How to reproduce:
```
$ riscv32-unknown-linux-gnu-g++ badalloc1.C 
$ qemu-riscv32  -L `riscv32-unknown-linux-gnu-g++ --print-sysroot` ./a.out
Segmentation fault (core dumped)

$ riscv32-unknown-linux-gnu-g++ badalloc1.C  -Wl,--no-relax
$ qemu-riscv32  -L `riscv32-unknown-linux-gnu-g++ --print-sysroot` ./a.out
$ echo $?
0
```

Code for badalloc1.C:
```
typedef __SIZE_TYPE__ size_t;
extern "C" void abort();
extern "C" void *memcpy(void *, const void *, size_t);

#ifdef STACK_SIZE
const int arena_size = 256 + 8 * 128;
#else
#if defined(__FreeBSD__) || defined(__sun__) || defined(__hpux__)
// FreeBSD, Solaris and HP-UX require even more space at initialization time.
// FreeBSD 5 now requires over 131072 bytes.
const int arena_size = 262144 + 72 * 1024;
#else
// Because pointers make up the bulk of our exception-initialization
// allocations, we scale by the pointer size from the original
// 32-bit-systems-based estimate.
const int arena_size = 32768 * ((sizeof (void *) + 3)/4) + 72 * 1024;
#endif
#endif

struct object
{
  size_t size __attribute__((aligned));
};

static char arena[arena_size] __attribute__((aligned));
static size_t pos;

// So we can force a failure when needed.
static int fail;

extern "C" void *malloc (size_t size)
{
  object *p = reinterpret_cast<object *>(&arena[pos]);

  if (fail)
    return 0;

  p->size = size;
  size = (size + __alignof__(object) - 1) & - __alignof__(object);
  pos += size + sizeof(object);

  // Verify that we didn't run out of memory before getting initialized.
  if (pos > arena_size)
    abort ();

  return p + 1;
}

extern "C" void free (void *)
{
}

extern "C" void *realloc (void *p, size_t size)
{
  void *r;

  if (p)
    {
      object *o = reinterpret_cast<object *>(p) - 1;
      size_t old_size = o->size;

      if (old_size >= size)
	{
	  r = p;
	  o->size = size;
	}
      else
	{
	  r = malloc (size);
	  memcpy (r, p, old_size);
	  free (p);
	}
    }
  else
    r = malloc (size);

  return r;
}

void fn_throw() throw(int)
{
  throw 1;
}

void fn_rethrow() throw(int)
{
  try{fn_throw();}
  catch(int a){
    throw;}
}

void fn_catchthrow() throw(int)
{
  try{fn_throw();}
  catch(int a){
    throw a + 1;}
}

int main()
{
  /* On some systems (including FreeBSD and Solaris 2.10),
     __cxa_get_globals will try to call "malloc" when threads are in
     use.  Therefore, we throw one exception up front so that
     __cxa_get_globals is all set up.  Ideally, this would not be
     necessary, but it is a well-known idiom, and using this technique
     means that we can still validate the fact that exceptions can be
     thrown when malloc fails.  */
  try{fn_throw();}
  catch(int a){}

  fail = 1;

  try{fn_throw();}
  catch(int a){}

  try{fn_rethrow();}
  catch(int a){}

  try{fn_catchthrow();}
  catch(int a){}
  
  return 0;
}

```

aswaterman, 2016-11-10:
This one's interesting... malloc relies on gp (because of relaxation).  gp is initialized in _start, i.e., after ld.so runs.  But ld.so calls malloc, before gp has been initialized...

I wonder how other ISAs that use gp handle this test?  I guess they don't automatically relax to gp, so it just happens to work?

sorear, 2016-11-11:
Judging by http://lists.llvm.org/pipermail/llvm-dev/2016-November/106889.html , most of the backends in LLVM (PPC EABI, MIPS, Hexagon, Lanai) use a small-data segment, and only symbols in that segment are allowed to do gp-relative addreses.  I don't have current information for gcc.

aswaterman, 2016-11-11:
Makes sense.

It looks like the Nios II and MIPS use ABI-specific ELF extensions for this purpose, which I would like to avoid.  I have a different fix, which is local to glibc and avoids a creating an ABI gadget: just use the preinit_array feature, which is a list of functions to be invoked before any shared object initializers are called.  

https://github.com/riscv/riscv-glibc/commit/362afedda5c2e011beb63ce2717fe6ab627eb2cb

aswaterman, 2016-11-11:
I'm closing this issue since the bug is fixed, but feel free to reopen for further discussion if you'd like.

closed 2016-11-11 by aswaterman