riscv-collab/riscv-gnu-toolchain #194: Exception handling through signal frames

kito-cheng, 2016-11-04:
I've test with -Wl,--no-relax but it's still fail, so maybe it's something wrong during ForcedUnwind.

GCC: riscv/riscv-gcc@bfaf705
Binutils: riscv/riscv-binutils-gdb@6756174
Glibc: riscv/riscv-glibc@7fef429
Qemu: https://github.com/riscv/riscv-qemu/commit/05ac735a4b4085e39678c967ecf7f47d45f459e0
Arch: rv32i, rv32g, rv64i and rv64g

Option:  -fexceptions -fnon-call-exceptions -O2

How to reproduce:
```
$ riscv32-unknown-linux-gnu-gcc cleanup-8.c -fexceptions -fnon-call-exceptions -O2
$ qemu-riscv32  -L `riscv32-unknown-linux-gnu-g++ --print-sysroot` ./a.out
Aborted (core dumped)
```

Code for cleanup-8.c
```
/* Verify that cleanups work with exception handling through signal
   frames.  */

#include <unwind.h>
#include <stdlib.h>
#include <signal.h>
#include <string.h>

static _Unwind_Reason_Code
force_unwind_stop (int version, _Unwind_Action actions,
                   _Unwind_Exception_Class exc_class,
                   struct _Unwind_Exception *exc_obj,
                   struct _Unwind_Context *context,
                   void *stop_parameter)
{
  if (actions & _UA_END_OF_STACK)
    abort ();
  return _URC_NO_REASON;
}

static void force_unwind ()
{
  struct _Unwind_Exception *exc = malloc (sizeof (*exc));
  memset (&exc->exception_class, 0, sizeof (exc->exception_class));
  exc->exception_cleanup = 0;
                   
#ifndef __USING_SJLJ_EXCEPTIONS__
  _Unwind_ForcedUnwind (exc, force_unwind_stop, 0);
#else
  _Unwind_SjLj_ForcedUnwind (exc, force_unwind_stop, 0);
#endif
                   
  abort ();
}

int count;
char *null;

static void counter (void *p __attribute__((unused)))
{
  ++count;
}

static void handler (void *p __attribute__((unused)))
{
  if (count != 2)
    abort ();
  exit (0);
}

static int __attribute__((noinline)) fn5 ()
{
  char dummy __attribute__((cleanup (counter)));
  force_unwind ();
  return 0;
}

static void fn4 (int sig)
{
  char dummy __attribute__((cleanup (counter)));
  fn5 ();
  null = NULL;
}

static void fn3 ()
{
  abort ();
}

static int __attribute__((noinline)) fn2 ()
{
  *null = 0;
  fn3 ();
  return 0;
}

static int __attribute__((noinline)) fn1 ()
{
  signal (SIGSEGV, fn4);
  signal (SIGBUS, fn4);
  fn2 ();
  return 0;
}

static int __attribute__((noinline)) fn0 ()
{
  char dummy __attribute__((cleanup (handler)));
  fn1 ();
  null = 0;
  return 0;
}

int main()
{ 
  fn0 ();
  abort ();
}

```

aswaterman, 2016-11-14:
OK, I fixed this; we needed to add the ABI-specific frame unwinding code to libgcc.  The program now works on Linux.  See https://github.com/riscv/riscv-gcc/commit/0fb771469ad266cfec835b070ea80a9b56022ba2

However, it does not work in QEMU.  I suspect that the signal handling code in QEMU is not ABI-compatible with the RISC-V Linux port, perhaps because the format of struct sigcontext doesn't match.  @sorear @sagark any interest in looking into that?

sorear, 2016-11-14:
Last time I tried to build gcc from the master branch under emulation, it failed at the end of stage1 / start of stage2 with a "C compiler cannot generate executables" error, probably due to multilib mismatch and I'm not sure what else.

Would I have better luck trying to use the whole toolchain repository?

If someone has a precompiled test program I can take a look.  Does this work for other architectures besides RISC-V?  There _is_ a significant difference in signal frame layout: QEMU does not provide a VDSO so it writes the sigreturn stub into the stack (relying on the fact that QEMU doesn't enforce PROT_EXEC).

kito-cheng, 2016-11-14:
@sorear aarch64 is ok, I can pass cleanup-8.c with linaro's toolchain and qemu in Fedora 23 .

And here is static link version for this case. https://drive.google.com/open?id=0B-huCVxC0XekSnRIdS1JNWEtLUk

```
$ qemu-riscv64  cleanup-8.rv64g 
```

AArch64:
```
$ qemu-aarch64 -version
qemu-aarch64 version 2.4.1 (qemu-2.4.1-11.fc23), Copyright (c) 2003-2008 Fabrice Bellard
$ wget https://releases.linaro.org/14.04/components/toolchain/binaries/gcc-linaro-aarch64-linux-gnu-4.8-2014.04_linux.tar.xz
$ tar -xf gcc-linaro-aarch64-linux-gnu-4.8-2014.04_linux.tar.xz
$ ./gcc-linaro-aarch64-linux-gnu-4.8-2014.04_linux/bin/aarch64-linux-gnu-gcc cleanup-8.c -fexceptions -fnon-call-exceptions -O2 -static-libgcc # -static-libgcc is needed since libgcc does not copy to sysroot in linaro's toolchain
$ qemu-aarch64 -L `./gcc-linaro-aarch64-linux-gnu-4.8-2014.04_linux/bin/aarch64-linux-gnu-gcc --print-sysroot` a.out 
$ echo $?
0
```

aswaterman, 2016-11-14:
Note, the RISC-V unwinder identifies signal frames by checking that the PC points to the sequence "li a7, 0x8b; ecall".  The other architectures do something analogous, but I figured it was worth pointing out.

lazyparser, 2021-03-04:
@kito-cheng Hello, would you like to update the status of this issue?

jim-wilson, 2021-03-04:
The testcase works on hardware and still fails on qemu.  This is believed to be a qemu bug.

closed 2022-11-11 by TommyMurphyTM1234