riscv-collab/riscv-gnu-toolchain #197: Linker does not output correctly for binary output format

ComponentFactory, 2016-11-16:
The following RISC-V assembly code (RV32) is used to show the problem...

    start:    jal end
    end:      jal start

I invoke the assembler using the following simple command...

    riscv32-unknown-elf-as -m32 example.s -o example.o

To check the generated code I disassemble it...

    riscv32-unknown-eft-objdump -D example.o

Which gives the following output...

    00000000 <start>:
        0:  004000ef    jal 4 <end>
    00000004 <end>
        4:  ffdff0ef    jal 0 <start>

This output looks correct. So I use the following linker command to generate the binary output...

    riscv32-unknown-elf-ld --oformat=binary example.o -o example

But this seems to have lost the relative addressing values because using the following command to look at the generated output bytes...

    xxd example

...gives the following result...

    00000000:  ef00 0000 eff0 ffff

Taking into account it is little endian it means that each set of four bytes is in reverse order compared to the disassembly seen earlier. We can see the first jump 000000ef has lost its jump offset. The second jump fffff0ef is also different from before and after applying twos compliment we get -1, which is definitely wrong!

The work around is to use...

    riscv32-unknown-elf-objcopy -O binary example.o example

..which does give the correct answer.

sorear, 2016-11-16:
(I was able to produce the same failure, FWIW)

aswaterman, 2016-11-16:
Yes, it seems like ld just does the wrong thing when not emitting ELF.  I don't have the cycles to look into it at the moment, but I'd happily review a patch if anyone else has the time.

jim-wilson, 2018-07-31:
The linker is using some ELF specific fields to control how it operates.  As a result, only ELF output is supported.  You must produce ELF and then use objcopy to convert to binary.  A number of other targets work the same way.  The upstream linker was fixed to give an error when you specify an output format other than ELF, instead of silently producing bad output.

closed 2018-07-31 by jim-wilson