riscv-collab/riscv-gnu-toolchain #199: "internal compiler error: in extract_insn, at recog.c:2287" when using float

TM1234, 2016-11-29:
I followed the instructions here for building the bare metal/newlib gcc and related tools:

https://github.com/sifive/riscv-gnu-toolchain

Specifically I built on CentOS 6.8 32 bit.
I cloned https://github.com/riscv/riscv-gnu-toolchain.
I installed the prerequisite tools.
I did configure --prefix=`pwd`/riscv-unknown-elf-gcc --enable-multilib --disable-float
Then I did make.

When I try to compile this code

```
float f;
f = 0.0;
if (f == 0.0)
{
    f = 1.0;
}

```

I get this (I'm using Eclipse and the Roa Logic RISC-V Eclipse plugin to compile hence the format of the gcc command line and logging output):

```
Building file: ../main.c
Invoking: RISC-V GCC/Newlib C Compiler
riscv64-unknown-elf-gcc -g3 -gdwarf-2 -m32 -DUSE_PLIC -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreGPIO" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreUARTapb" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/hal" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/riscv_hal" -O0 -Wall -c -MMD -MP -MF"main.d" -MT"main.d" -march=RV32IM -mmuldiv -mfloat-abi=soft -o "main.o" "../main.c"
../main.c: In function 'main':
../main.c:79:1: error: unrecognizable insn:
 }
 ^
(insn 34 33 35 2 (set (reg:SI 86)
        (eq:SI (reg:DF 82)
            (reg:DF 84))) ../main.c:62 -1
     (nil))
../main.c:79:1: internal compiler error: in extract_insn, at recog.c:2287
0x8565dc5 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
	/home/user/Downloads/riscv/riscv-gnu-toolchain/src/newlib-gcc/gcc/rtl-error.c:108
0x8565e23 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
	/home/user/Downloads/riscv/riscv-gnu-toolchain/src/newlib-gcc/gcc/rtl-error.c:116
0x853ab1a extract_insn(rtx_insn*)
	/home/user/Downloads/riscv/riscv-gnu-toolchain/src/newlib-gcc/gcc/recog.c:2287
0x8363845 instantiate_virtual_regs_in_insn
	/home/user/Downloads/riscv/riscv-gnu-toolchain/src/newlib-gcc/gcc/function.c:1582
0x8363845 instantiate_virtual_regs
	/home/user/Downloads/riscv/riscv-gnu-toolchain/src/newlib-gcc/gcc/function.c:1950
0x8363845 execute
	/home/user/Downloads/riscv/riscv-gnu-toolchain/src/newlib-gcc/gcc/function.c:1999
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [main.o] Error 1

09:59:10 Build Finished (took 766ms)
```

I'm not sure if I need to include any additional info?

BTW - I also tried

configure --prefix=`pwd`/riscv-unknown-elf-gcc --enable-multilib

(which enables float support by default) but it gave this:

```
Building file: ../main.c
Invoking: RISC-V GCC/Newlib C Compiler
riscv64-unknown-elf-gcc -g3 -gdwarf-2 -m32 -DUSE_PLIC -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreGPIO" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreUARTapb" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/hal" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/riscv_hal" -O0 -Wall -c -MMD -MP -MF"main.d" -MT"main.d" -march=RV32IM -mmuldiv -mfloat-abi=soft -o "main.o" "../main.c"
../main.c: In function 'main':
../main.c:79:1: error: unrecognizable insn:
 }
 ^
(insn 34 33 35 2 (set (reg:SI 86)
        (eq:SI (reg:DF 82)
            (reg:DF 84))) ../main.c:62 -1
     (nil))
../main.c:79:1: internal compiler error: in extract_insn, at recog.c:2287
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [main.o] Error 1
```

TM1234, 2016-12-07:
I got the latest sources today and built them on Ubuntu 14.04.5 32 bit using the default gcc 4.8.4:
```
git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/riscv-unknown-elf-gcc --enable-multilib --disable-float
make
```
When I try to compile main.c containing the following:
```
float f = 10.1;
f *= 2.2;
```
I get this - similar to but not the same as the original problem above:
```
Building file: ../main.c
Invoking: RISC-V GCC/Newlib C Compiler
riscv64-unknown-elf-gcc -g3 -gdwarf-2 -m32 -DUSE_PLIC -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreGPIO" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreUARTapb" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/hal" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/riscv_hal" -O0 -Wall -c -MMD -MP -MF"main.d" -MT"main.d" -march=RV32IM -mmuldiv -mfloat-abi=soft -o "main.o" "../main.c"
../main.c: In function 'main':
../main.c:76:1: error: unable to generate reloads for:
 }
 ^
(insn 33 32 34 2 (set (reg:DF 75 [ _13 ])
        (float_extend:DF (reg:SF 86))) ../main.c:62 101 {extendsfdf2}
     (expr_list:REG_DEAD (reg:SF 86)
        (nil)))
../main.c:76:1: internal compiler error: in curr_insn_transform, at lra-constraints.c:3564
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [main.o] Error 1
```

aswaterman, 2016-12-07:
We are working on this today.

TM1234, 2016-12-07:
Thanks Andrew.
I reduced my main.c to just this and still get the same problem as above (my second post):
```
int main(int argc, char **argv) {
     float f = 10.1;
     f *= 2.2;
     while (1)
     {
     }
}
```
and I get this:
```
make all 
Building file: ../main.c
Invoking: RISC-V GCC/Newlib C Compiler
riscv64-unknown-elf-gcc -g3 -gdwarf-2 -m32 -DUSE_PLIC -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreGPIO" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/drivers/CoreUARTapb" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/hal" -I"/home/user/Microsemi_SoftConsole_v5.0/extras/workspace.examples/systick-blinky/riscv_hal" -O0 -Wall -c -MMD -MP -MF"main.d" -MT"main.d" -march=RV32IM -mmuldiv -mfloat-abi=soft -o "main.o" "../main.c"
../main.c: In function 'main':
../main.c:87:1: error: unable to generate reloads for:
}
^
(insn 11 10 12 2 (set (reg:DF 72 [ _2 ])
        (float_extend:DF (reg:SF 77))) ../main.c:83 101 {extendsfdf2}
     (expr_list:REG_DEAD (reg:SF 77)
        (nil)))
../main.c:87:1: internal compiler error: in curr_insn_transform, at lra-constraints.c:3564
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [main.o] Error 1

23:06:51 Build Finished (took 1s.394ms)

```
Hope this helps.

aswaterman, 2016-12-07:
Thanks.  I can replicate the problem here.  We'll roll it into a refactoring of compiler flags handling ISA/ABI configuration, as that is the root of the problem.  Part of the compiler thinks floating-point is supported, and another part doesn't.

TM1234, 2016-12-07:
Thanks Andrew - I presume that this is not related?

https://groups.google.com/a/groups.riscv.org/d/msg/sw-dev/EAyXwZMbZbw/5FBiyU3dCQAJ

aswaterman, 2016-12-07:
Possibly.  We'll make sure that configuration works after these changes.

TM1234, 2016-12-07:
OK - thanks a lot.

TM1234, 2016-12-09:
Apropos of this, even if they don't work right now, what are the correct configuration options to build a 32 bit hosted toolchain (i.e. 32 bit gcc etc. executables build on a 32 bit Ubuntu), named riscv64-unknown-elf-* that can supports both 32 and 64 bit RISC-V compilation and libs and that also supports libs with soft float support?

I assumed that it's 

./configure --prefix=... --enable-multilib --disable-float

However others have said that I need to use --with-arch=RV32IM but that doesn't make sense to me since surely that would not cause 32 AND 64 bit libs to be built? In any case --enable-multilib and --with-arch=RV32IM causes the build to break at the moment anyway.

palmer-dabbelt, 2017-03-30:
I'm going to assume this is fixed, as we have much better multilib support now.

closed 2017-03-30 by palmer-dabbelt