riscv-collab/riscv-gnu-toolchain #200: Need to be able to create ELF files with 2M segment starting and alignment

rminnich, 2016-12-01:
The Harvey kernel, due to its HPC heritage, does not support or use 4K pages. On PPC, it uses 1M pages; on amd64 and riscv, it uses 2M pages.

I've hit two issues with the riscv toolchain. I can't tell it to use 2M pages via the 
-z common-page-size=xxxx
switch to ld; it issues a warning that it is ignoring the switch.
While I can use a ldscript to force alignment of program segments to 2M, it still sets the alignment element of the struct to 0x1000.

What would be ideal is to have it correctly implement the common-page-size switch, such that segments are 2M aligned and the alignment in the ELF program headers is 0x200000, not 0x1000

aswaterman, 2016-12-07:
I figured out the "common-page-size ignored" error comes from using a space instead of = between the flag and the size.

The real problem still stands.

aswaterman, 2016-12-07:
OK, so, the problem is that the max page size defaults to 0x1000 on RISC-V's BFD port.  I checked, and this is the most common definition; x86 is the odd one out, defining it to 0x200000.

Fortunately, there is a portable fix to your problem: -z max-page-size=0x200000 -z common-page-size=0x200000

closed 2016-12-07 by aswaterman