riscv-collab/riscv-gnu-toolchain #201: Tweak prerequisites instructions for tools build?

TM1234, 2016-12-07:
See here:

https://github.com/riscv/riscv-gnu-toolchain

As far as I can tell the following prerequisites are sufficient:

https://forums.sifive.com/t/e300-linux-make-tools-dependencies/86/3

viz.

sudo apt-get install automake libmpc-dev gawk bison flex texinfo libtool libusb-1.0-0-dev g++

Add dejagnu if the newlib test suite will be run

aswaterman, 2016-12-07:
I don't see much benefit in paring this down, since most of the other
packages will be pulled in by dependences.  It's more confusing to separate
out dependences based upon what features will be used (e.g., better to just
tell everyone to install dejagnu).

Once the ABI is stable and we have binary distributions of the cross
compilers, the point will be moot.

closed 2016-12-07 by aswaterman

TM1234, 2016-12-07:
OK - fair enough Andrew.
I just thought that a minimal set of prerequisites to install might make sense and help people keep their build machine as clean/lean as possible.
Not a major issue. :-)