riscv-collab/riscv-gnu-toolchain #202: unsupported ISA substring Xcustom

cliffordwolf, 2016-12-08:
With 7e48594 I used to be able to run gcc with `-march=RV32IMCXcustom` and then could use opcodes like `custom0 zero,0,0,4` in assembler code. Now (with 805ee1d) gcc rejects Xcustom

```
/opt/riscv32i/bin/riscv32-unknown-elf-gcc -c -m32 -march=RV32IMCXcustom -o firmware/start.o firmware/start.S
cc1: error: -march=RV32IMCXcustom: unsupported ISA substring Xcustom
```

and without it the assembler rejects my code.

```
/opt/riscv32i/bin/riscv32-unknown-elf-gcc -c -m32 -march=RV32IMC -o firmware/start.o firmware/start.S
firmware/start.S: Assembler messages:
firmware/start.S:37: Error: unrecognized opcode `custom0 zero,0,0,4'
firmware/start.S:38: Error: unrecognized opcode `custom0 zero,zero,0,3'
...
```

What is the right way to do this with the current tool chain?

seldridge, 2016-12-08:
See: https://github.com/riscv/riscv-gnu-toolchain/issues/190

tl;dr: Support removed during the upstreaming process as RoCC is a custom extension. You can get around this with some preprocessor macros (https://github.com/riscv/riscv-gnu-toolchain/issues/190#issuecomment-264083212) which gcc will cleanup to be at lest reasonably efficient.

palmer-dabbelt, 2016-12-08:
We removed support for Xcustom and the custom opcodes because they're a Berkeley extension and we didn't want to give undue priority to our stuff (it wouldn't be fair to everyone).  There's another issue about this

  https://github.com/riscv/riscv-gnu-toolchain/issues/190

that has various suggestions for how to write CPP macros that produce the desired behavior.  I don't want to regress on anyone, so if those solutions aren't sufficient then we'll figure out a way to make your stuff work, but I'm going to close this issue to keep all the discussion in one place.  Feel free to re-open the other issue if you need to.

closed 2016-12-08 by palmer-dabbelt