riscv-collab/riscv-gnu-toolchain #205: ld: can't relax section: Success

cliffordwolf, 2016-12-10:
I've derived the following test case from the `gcc -E` output for `auipc.S` from riscv-tests:

```
.text
.global main
main:
addi a0,zero,0
addi a1,zero,0
jal zero,main
.ascii "mytest"
.align 3
```

With 805ee1d (built with --with-arch=RV32I) I get the following error:

```
$ /opt/riscv32i/bin/riscv32-unknown-elf-gcc test.S
/opt/riscv32i/lib/gcc/riscv32-unknown-elf/6.1.0/../../../../riscv32-unknown-elf/bin/ld: can't relax section: Success
collect2: error: ld returned 1 exit status
```

The test case is minimal in the sense that commenting out any of the three instructions, the .ascii statement, or the .align statement makes the problem disappear.

cliffordwolf, 2016-12-15:
Can you guys reproduce this? Is anyone working on it? Is there anything I can do to help?

After updating to e3e50c5 (and changing the `configure` parameter `--with-arch=RV32I` to `--with-arch=rv32i` to fix the build) I still get the same `can't relax section: Success` error for my `test.S` file.

I'm currently not updating riscv-gnu-toolchain for PicoRV32 because of this problem.

sorear, 2016-12-15:
@palmer-dabbelt Do any of your recent relocation fixes impact this test?

aswaterman, 2016-12-15:
I'll look into it

aswaterman, 2016-12-15:
The problem manifests because the text section lost word alignment.  Currently, .align on text sections only does the right thing if the text section is already 4-byte aligned (or 2-byte aligned for RVC).  You can work around this by putting the string in the data section.

Working on a fix...

aswaterman, 2016-12-16:
Should be fixed in riscv-binutils-gdb.  Re-running test suite, then we can push to riscv-gnu-toolchain.

closed 2016-12-16 by aswaterman