riscv-collab/riscv-gnu-toolchain #209: Newlib entry point is not 0x10000 anymore

cliffordwolf, 2016-12-17:
It used to be that the entry point used by the default newlib linker script is 0x10000. However, with rev 34e199d of riscv-gnu-toolchain (with cherry-picked riscv/riscv-binutils-gdb@a5971ec) there does not seem to be a fixed entry point anymore. (My toolchain is built with `--with-arch=rv32i`.)

E.g. using my version of dhrystone (see https://github.com/cliffordwolf/picorv32/tree/riscv-gnu-toolchain-update/dhrystone for build scripts):

```
$ riscv32-unknown-elf-readelf -h dhry.elf
...
  Entry point address:               0x15590
...
```

(If I manually set my processor entry point to 0x15590 then I can execute dhrystone from this binary. With the default entry point of 0x10000 it just crashes of course.)

If I build an empty program and link it with newlib, I get yet another entry point:

```
$ echo 'int main() { return 0; }' > test.c
$ riscv32-unknown-elf-gcc test.c 
$ riscv32-unknown-elf-readelf -h a.out 
...
  Entry point address:               0x114f8
...
```

Is the static 0x10000 newlib entry point deprecated? Is there a convenient way to set the entry point address without writing a custom linker script?

Afaics my test cases that do not use the default linker script work fine.

Also: riscv-gnu-toolchain used to install a `<prefix>/riscv32-unknown-elf/lib/riscv.ld` linker script that could be used as a starting point for custom scripts. This file seems to be gone now. Any reason for that? Should I now use the linker script produced by `riscv32-unknown-elf-ld -verbose` as a reference? Is `riscv32-unknown-elf-ld -z nocombreloc -verbose` better?

aswaterman, 2016-12-18:
It is deprecated.  Our newlib toolchain used to use a custom linker script always, but in the upstreaming process we decided it was vastly preferable to use LD's built-in linker script instead, which keeps the various toolchains more similar to each other and to other ELF toolchains.

The default linker script is suitable for systems that employ a proper ELF loader that can jump to the encoded entry point.  Embedded systems that don't have this property usually require custom linker scripts anyway, so I don't view this as a major regression.

Here's the old linker script, by the way: https://raw.githubusercontent.com/riscv/riscv-gnu-toolchain/87ec00322ed6caa207747ce00ed38ca15995ec63/newlib/libgloss/riscv/riscv.ld

cliffordwolf, 2017-02-26:
Is there a reason this issue is still open? I'm closing it now. Thanks @aswaterman for the explanation.

Btw: I'm now using [this](https://github.com/cliffordwolf/picorv32/blob/master/firmware/riscv.ld) linker script, which is a [horribly hacked](https://github.com/cliffordwolf/picorv32/blob/master/firmware/riscv_ld.diff) version of the binutils default linker script used by riscv-binutils.

closed 2017-02-26 by cliffordwolf