riscv-collab/riscv-gnu-toolchain #210: Top of git doesn't build rv32 on MSYS2/Windows10

mcnamarad1971, 2016-12-19:
Gents,

git revision 34e199df17d4620c8e25e5932a25a51b25a787cd is not building, after using the following ./configure && make lines.

./configure --with-arch=rv32im, make -j4
or
./configure --with-arch=rv32g, make -j4

N.B. ./configure --enable-multilib --with-arch=rv64g fails at the same point as it starts building the 32-bit libs for the newlib variant.

Same line works on ubuntu.  

64-bit variants appear to build okay on MSYS2/Windows10 so it seems to be something local to the 32-bit RISCV variant?

The issue being reported is in newlib, libstdc++-v3, src, c++11 part of the build when it tries to build cxx11-locale-inst.cc.  

==============

The compile line is:
/bin/sh ../../libtool --tag CXX --tag disable-shared   --mode=compile /<a_path>/riscv-gnu-toolchain/build-gcc-newlib/./gcc/xgcc -shared-libgcc -B/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/./gcc -nostdinc++ -L/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/src -L/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/src/.libs -L/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/libsupc++/.libs -nostdinc -B/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/newlib/ -isystem /<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /<a_path>/riscv-gnu-toolchain/src/newlib-gcc/newlib/libc/include -B/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libgloss/libnosys -L/<a_path>/riscv-gnu-toolchain/src/newlib-gcc/libgloss/riscv32 -B/opt/riscv32/riscv32-unknown-elf/bin/ -B/opt/riscv32/riscv32-unknown-elf/lib/ -isystem /opt/riscv32/riscv32-unknown-elf/include -isystem /opt/riscv32/riscv32-unknown-elf/sys-include    -I/<a_path>/riscv-gnu-toolchain/src/newlib-gcc/libstdc++-v3/../libgcc -I/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/include/riscv32-unknown-elf -I/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/include -I/<a_path>/riscv-gnu-toolchain/src/newlib-gcc/libstdc++-v3/libsupc++   -std=gnu++11   -fno-implicit-templates  -Wall -Wextra -Wwrite-strings -Wcast-qual -Wabi  -fdiagnostics-show-location=once   -ffunction-sections -fdata-sections  -frandom-seed=cxx11-locale-inst.lo -g -O2  -c -o cxx11-locale-inst.lo /<a_path>/riscv-gnu-toolchain/src/newlib-gcc/libstdc++-v3/src/c++11/cxx11-locale-inst.cc

It appears to generate the object file (cxx-locale-inst.o) okay, but the compiler reports an error back to make

========
The error being reported is:
In file included from /<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/include/bits/locale_facets.h:48:0,
                 from /<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/include/locale:40,
                 from /<a_path>/riscv-gnu-toolchain/src/newlib-gcc/libstdc++-v3/src/c++11/locale-inst.cc:38,
                 from /<a_path>/src/riscv-gnu-toolchain/src/newlib-gcc/libstdc++-v3/src/c++11/cxx11-locale-inst.cc:39:
/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/include/bits/streambuf_iterator.h: In member function '_InIter std::__cxx11::time_get<_CharT, _InIter>::do_get_date(std::__cxx11::time_get<_CharT, _InIter>::iter_type, std::__cxx11::time_get<_CharT, _InIter>::iter_type, std::ios_base&, std::ios_base::iostate&, tm*) const [with _CharT = char; _InIter = std::istreambuf_iterator<char, std::char_traits<char> >]':
/<a_path>/riscv-gnu-toolchain/build-gcc-newlib/riscv32-unknown-elf/libstdc++-v3/include/bits/streambuf_iterator.h:198:40: internal compiler error: Aborted
  return traits_type::eq_int_type(_M_get(), __eof);

===============
Here are some details of the original native gcc that I used
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-msys/5.3.0/lto-wrapper.exe
Target: x86_64-pc-msys
Configured with: /msys_scripts/gcc/src/gcc-5.3.0/configure --build=x86_64-pc-msys --prefix=/usr --libexecdir=/usr/lib --enable-bootstrap --enable-shared --enable-shared-libgcc --enable-static --enable-version-specific-runtime-libs --with-arch=x86-64 --with-tune=generic --disable-multilib --enable-__cxa_atexit --with-dwarf2 --enable-languages=c,c++,fortran,lto --enable-graphite --enable-threads=posix --enable-libatomic --enable-libcilkrts --enable-libgomp --enable-libitm --enable-libquadmath --enable-libquadmath-support --enable-libssp --disable-win32-registry --disable-symvers --with-gnu-ld --with-gnu-as --disable-isl-version-check --enable-checking=release --without-libiconv-prefix --without-libintl-prefix --with-system-zlib --enable-linker-build-id --with-default-libstdcxx-abi=gcc4-compatible
Thread model: posix
gcc version 5.3.0 (GCC)
==============

I'd appreciate any ideas for debug/workarounds, etc for the time being.

aswaterman, 2016-12-19:
It's of course possible that there's a portability bug somewhere.  Alas, I don't own a windows system, so I won't be able to replicate the problem, but maybe I can walk you through diagnosing the problem.

First of all, gcc is the compiler driver, and it invokes potentially several processes; one of those is likely crashing.  Add the --verbose and -save-temps flags to the gcc command line to see which process is crashing, and to preserve the temporary files so that you can invoke that process directly on the same inputs.  You can then use your debugger to see where (and possibly why) it is crashing, which would provide us with quite a bit more information.

mcnamarad1971, 2016-12-22:
Thanks Andrew, I've worked around by getting a mingw32 build going instead.

aswaterman, 2016-12-23:
Would you be so kind as to briefly write up your approach so that someone who isn't a fluent Windows developer (i.e., me) can incorporate it into the README?  (e.g., I installed packages X, Y, Z on cygwin version V, and had to jump through hoops A, B, C)

cr1901, 2017-01-20:
@mcnamarad1971 I am a prospective MSYS2 builder. When you say "getting a mingw32 build going instead", I assume you mean that you open a bash prompt with `MSYSTEM` set to `MINGW32` (or alternatively, a bash prompt that is spawned from mingw32_shell.bat)?

palmer-dabbelt, 2017-05-04:
I've cross compiled this for Windows recently, so I'm going to consider this closed.

closed 2017-05-04 by palmer-dabbelt