riscv-collab/riscv-gnu-toolchain #211: double a, b, c; return -(a*b-c); suboptimal

tommythorn, 2017-01-06:
I was just fooling around when I happened upon a strange inconsistency; I expected to see fnmsub.d.
Oddly the other three cases are handled correctly.

```
tommy@xps15:~/freedom-e-sdk$ echo 'double f(double a, double b, double c) { return -a*b-c; }' > gmn.c &&./toolchain/bin/riscv32-unknown-elf-gcc -march=rv32g -mabi=ilp32d -O2 -ffast-math -S gmn.c  && cat gmn.s
	.file	"gmn.c"
	.option nopic
	.text
	.align	2
	.globl	f
	.type	f, @function
f:
	fneg.d	fa0,fa0
	fmsub.d	fa0,fa0,fa1,fa2
	ret
	.size	f, .-f
	.ident	"GCC: (GNU) 6.1.0"
tommy@xps15:~/freedom-e-sdk$ echo 'double f(double a, double b, double c) { return -(a*b-c); }' > gmn.c &&./toolchain/bin/riscv32-unknown-elf-gcc -march=rv32g -mabi=ilp32d -O2 -ffast-math -S gmn.c  && cat gmn.s
	.file	"gmn.c"
	.option nopic
	.text
	.align	2
	.globl	f
	.type	f, @function
f:
	fneg.d	fa0,fa0
	fmadd.d	fa0,fa0,fa1,fa2
	ret
	.size	f, .-f
	.ident	"GCC: (GNU) 6.1.0"
tommy@xps15:~/freedom-e-sdk$ echo 'double f(double a, double b, double c) { return -(a*b+c); }' > gmn.c &&./toolchain/bin/riscv32-unknown-elf-gcc -march=rv32g -mabi=ilp32d -O2 -ffast-math -S gmn.c  && cat gmn.s
	.file	"gmn.c"
	.option nopic
	.text
	.align	2
	.globl	f
	.type	f, @function
f:
	fnmadd.d	fa0,fa0,fa1,fa2
	ret
	.size	f, .-f
	.ident	"GCC: (GNU) 6.1.0"
```

kito-cheng, 2017-01-06:
Thank for report that, the code gen is optimize now, I will push the fix after test :)
```
[kito@iCan riscv-workspace]$ cat x1.c 
double f(double a, double b, double c) { return -a*b-c; }
[kito@iCan riscv-workspace]$ ./rv64imafd/bin/riscv64-unknown-linux-gnu-gcc x1.c -o - -S -O3 
	.file	"x1.c"
	.option nopic
	.text
	.align	2
	.globl	f
	.type	f, @function
f:
	fnmsub.d	fa0,fa0,fa1,fa2
	ret
	.size	f, .-f
	.ident	"GCC: (GNU) 6.1.0"
[kito@iCan riscv-workspace]$ cat x2.c 
double f(double a, double b, double c) { return -(a*b-c); }
[kito@iCan riscv-workspace]$ ./rv64imafd/bin/riscv64-unknown-linux-gnu-gcc x2.c -o - -S -O3 
	.file	"x2.c"
	.option nopic
	.text
	.align	2
	.globl	f
	.type	f, @function
f:
	fnmsub.d	fa0,fa0,fa1,fa2
	ret
	.size	f, .-f
	.ident	"GCC: (GNU) 6.1.0"
[kito@iCan riscv-workspace]$ cat x3.c 
double f(double a, double b, double c) { return -(a*b+c); }
[kito@iCan riscv-workspace]$ ./rv64imafd/bin/riscv64-unknown-linux-gnu-gcc x3.c -o - -S -O3 
	.file	"x3.c"
	.option nopic
	.text
	.align	2
	.globl	f
	.type	f, @function
f:
	fnmadd.d	fa0,fa0,fa1,fa2
	ret
	.size	f, .-f
	.ident	"GCC: (GNU) 6.1.0"
```

aswaterman, 2017-01-06:
Thanks y'all.

closed 2017-01-27 by aswaterman