riscv-collab/riscv-gnu-toolchain #212: Cannot compile toolchain using Arch Linux

GuzTech, 2017-01-11:
For some time now I cannot compile the toolchain under Arch Linux. I get the following error:

```
syslex_wrap.o: In function `yylex':
/home/guztech/Projects/FPGA/RISCV/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib/binutils/syslex.c:1097: undefined reference to `yywrap'
```

I do have flex installed, and I can compile generated C files and link then using the `-lfl` flag. `syslex.c` has a define `yywrap` and if I remove it, then I can continue until I get the same error in other files. It seems like the linker does not use the `-lfl` flag or puts it in the wrong place. I also tried it on Ubuntu 16.04 in a VM and it works just fine.

[Arch] GCC version 6.2.1
[Arch] Flex version 2.6.3
[Ubuntu] GCC version 5.4.x
[Ubuntu] Flex version 2.6.0

aswaterman, 2017-01-11:
Seems like a generic toolchain build issue (i.e. not RISC-V-specific), so there's not a lot we can do. I'll leave the issue open for a while, in case others have a solution.

GuzTech, 2017-01-11:
Allright, I'll also post all my findings and extra information here for posterity.

Using GCC version 5.4.0 on Arch does not fix the problem.

GuzTech, 2017-01-11:
`syslex.c` has a bunch of defines that other files:

```
    #define yy_create_buffer yy_create_buffer

    #define yy_delete_buffer yy_delete_buffer
...
    #define yywrap yywrap

    #define yyalloc yyalloc

etc etc
```

When I run flex like in the Makefile by hand, the version in Ubuntu 16.04 does not generate these defines, but the version in Arch does. The only difference is the version of flex (2.6.0 vs 2.6.3).

palmer-dabbelt, 2017-01-12:
Looks like flex-2.6.3 breaks a bunch of stuff with exactly this error message

https://bugs.gentoo.org/show_bug.cgi?id=598186
https://bugs.gentoo.org/show_bug.cgi?id=604848
https://bugs.gentoo.org/show_bug.cgi?id=604732

There seems to be an upstream issue about it already

https://github.com/westes/flex/issues/155

so I vote we just close this one.  @GuzTech @aswaterman any opposition to closing this bug?

GuzTech, 2017-01-12:
Thanks for finding that out, my Google-Fu failed me this time. I'm fine with closing this bug.

palmer-dabbelt, 2017-01-12:
No problem -- I just checked my local Gentoo machine, noticed it was on 2.6.1, and figured there'd be some blocker 2.6.3 wasn't stable yet.  Sure enough, the blocker was exactly this bug.  That's frequently the case :).

closed 2017-01-12 by palmer-dabbelt