riscv-collab/riscv-gnu-toolchain #213: Tool Chain installation

susi655, 2017-02-01:
Sorry for reopening the issue #174 

I wanted to update my tool chain. So, I downloaded the latest files and started installing. I used the build.sh which worked previously. But the following error popped up.

Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
make: *** [mtrap.o] Error 1

aswaterman, 2017-02-01:
That error message is coming from the x86 compiler; it looks like the RISC-V compiler is not getting invoked.  Maybe a PATH issue?

susi655, 2017-02-01:
echo $PATH

/home/smart/bin:/home/smart/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/smart/smart/Tools/Vivado/2016.2/bin:/home/smart/smart/Tools/14.7/ISE_DS/PlanAhead/bin:/home/smart/smart/Tools/14.7/ISE_DS/ISE/bin/lin64:/snap/bin:/home/smart/riscv-tools/riscv/bin:**/home/smart/riscv-tools/riscv/bin**

parvez0701, 2017-02-09:
I am working on 64 bit Linux machine. I am not getting any output for my hello program:
[pahmad@ibxdv-pahmad ~]$ riscv32-unknown-linux-gnu-gcc -o hello hello.c
[pahmad@ibxdv-pahmad ~]$ spike pk hello
[pahmad@ibxdv-pahmad ~]$

palmer-dabbelt, 2017-03-30:
I'm not sure what's going on with this bug, so I'm closing it.

closed 2017-03-30 by palmer-dabbelt