riscv-collab/riscv-gnu-toolchain #215: build error with freshly cloned source

Silur, 2017-02-10:
System is archlinux 4.9.7-1, toolchain source freshly cloned and build started with:
```
./configure --prefix=/opt/riscv --enable-multilib
make linux
```

Error is the following:
```
ada-exp.o: In function `ada_yyparse()':
/tmp/riscv-gnu-toolchain/build-binutils-linux/gdb/ada-exp.c:1586: undefined reference to `ada_yylex()'
```

All dependencies listed in README.md is installed

bgamari, 2017-02-17:
There is an [upstream ticket](https://sourceware.org/bugzilla/show_bug.cgi?id=21057) for this.

Silur, 2017-02-17:
how reliable is this patch? did the core developers review it, is it a quick-n-dirty hack that probably breaks the runtime itself or an actual patch?

palmer-dabbelt, 2017-03-30:
I'm closing this as an upstream issue.

closed 2017-03-30 by palmer-dabbelt