riscv-collab/riscv-gnu-toolchain #217: "Register 834 is not available" error for breakpoint

slavaim, 2017-02-15:
I has received an error "Register 834 is not available" when trying to set a breakpoint in GDB attached to a QEMU running the Linux kernel . The step instructions s/si/n were functional but sometimes the same error was displayed. 

QEMU has been compiled according to instructions from https://riscv.org/software-tools/riscv-qemu/ 

Any ideas?

Below is an example of a GDB session with an error. I intentionally tried to set a breakpoint to a function that had been called  and was visible in a call stack to narrow the possible options.

```
(gdb) target remote localhost:1234
Remote debugging using localhost:1234
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xffffffff8012395c in ?? ()
(gdb) bt
#0  0xffffffff8012395c in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) file /work/risc-v/linux/linux-4.6.2/vmlinux
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from /work/risc-v/linux/linux-4.6.2/vmlinux...done.
(gdb) bt
#0  format_decode (fmt=0xffffffff8026c4fa "%s", spec=0xffffffff8028de48 <init_thread_union+7752>) at lib/vsprintf.c:1867
#1  0xffffffff8012555c in vsnprintf (buf=0xffffffff802cb7ba <textbuf+2> "", size=992, fmt=0xffffffff8026c4fa "%s", args=0xffffffff8026c4fa) at lib/vsprintf.c:344
#2  0xffffffff80125f88 in vscnprintf (buf=<optimized out>, size=992, fmt=<optimized out>, args=<optimized out>) at lib/vsprintf.c:1660
#3  0xffffffff8003b74c in vprintk_emit (facility=-2144942854, level=-1, 
    dict=0xffffffff80231000 "\b]\022\200\324]\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200D[\022\200\b[\022\200 Z\022\200\324Z\022\200 Z\022\200\240\\\022\200\324\\\022\200\364[\022\200t[\022\200 Z\022\200 Z\022\200 Z\022\200\240]\022\200 Z\022\200 Z\022\200 Z\022\200 Z\022\200<]\022\200\240\\\022\200\214d\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200Da\022\200"..., dictlen=18446744071564608762, 
    fmt=0x2 <error: Cannot access memory at address 0x2>, args=0x25) at kernel/printk/printk.c:2347
#4  0xffffffff80057050 in printk (fmt=<optimized out>) at kernel/printk/printk.c:1854
#5  0xffffffff80000754 in start_kernel () at init/main.c:488
#6  0xffffffff80000040 in _start () at arch/riscv/kernel/head.S:36
Backtrace stopped: frame did not save the PC
(gdb) b start_kernel
Register 834 is not available
(gdb) set architecture riscv:rv64
The target architecture is assumed to be riscv:rv64
(gdb) b start_kernel
Register 834 is not available
(gdb) p start_kernel
$1 = {void (void)} 0xffffffff800006f0 <start_kernel>
```

slavaim, 2017-02-17:
I did some research and I believe gdb_handle_packet was not called at QEMU when breakpoint command was being executed by GDB. This points to a potential problem in GDB. For other GDB commands QEMU's gdb_handle_packet is called as expected

```
#0  gdb_handle_packet (s=0x5604b7b0d040, line_buf=0x5604b7b0d05c "g") at /work/risc-v/riscv-qemu/gdbstub.c:788
#1  0x00005604b66ed0cd in gdb_read_byte (s=0x5604b7b0d040, ch=55) at /work/risc-v/riscv-qemu/gdbstub.c:1460
#2  0x00005604b66ed1ed in gdb_chr_receive (opaque=0x0, 
    buf=0x7ffdcb5ff120 "+$g#67;c#a8028dfc0,40#5cs+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df", size=6)
    at /work/risc-v/riscv-qemu/gdbstub.c:1668
#3  0x00005604b675f20b in qemu_chr_be_write_impl (s=0x5604b7b0c330, 
    buf=0x7ffdcb5ff120 "+$g#67;c#a8028dfc0,40#5cs+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df", len=6) at qemu-char.c:388
#4  0x00005604b675f269 in qemu_chr_be_write (s=0x5604b7b0c330, 
    buf=0x7ffdcb5ff120 "+$g#67;c#a8028dfc0,40#5cs+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df", len=6) at qemu-char.c:400
#5  0x00005604b6763691 in tcp_chr_read (chan=0x5604b7a228a0, cond=G_IO_IN, opaque=0x5604b7b0c330) at qemu-char.c:2966
#6  0x00005604b68e71b2 in qio_channel_fd_source_dispatch (source=0x5604b7b0cfc0, callback=0x5604b676353b <tcp_chr_read>, user_data=0x5604b7b0c330) at io/channel-watch.c:84
#7  0x00007f94a1bbc04a in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00005604b6865f2e in glib_pollfds_poll () at main-loop.c:213
#9  0x00005604b6866009 in os_host_main_loop_wait (timeout=0) at main-loop.c:258
#10 0x00005604b68660b6 in main_loop_wait (nonblocking=1) at main-loop.c:506
#11 0x00005604b676cb9b in main_loop () at vl.c:1947
#12 0x00005604b6774331 in main (argc=10, argv=0x7ffdcb600678, envp=0x7ffdcb6006d0) at vl.c:4662
```

slavaim, 2017-02-18:
I continued with the research and located the point where the exception was thrown by GDB 

```
#0  0x0000000000710acd in __cxa_throw ()
#1  0x00000000004d729c in throw_exception_cxx (exception=...) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/common/common-exceptions.c:303
#2  0x00000000004d7325 in throw_exception (exception=...) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/common/common-exceptions.c:317
#3  throw_it(return_reason, errors, const char *, typedef __va_list_tag __va_list_tag *) (reason=reason@entry=RETURN_ERROR, error=<optimised out>, fmt=<optimised out>, ap=ap@entry=0x7ffcfaae49e0)
    at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/common/common-exceptions.c:373
#4  0x00000000004d7556 in throw_verror (error=<optimised out>, fmt=<optimised out>, ap=ap@entry=0x7ffcfaae49e0) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/common/common-exceptions.c:379
#5  0x00000000004d7614 in throw_error (error=error@entry=NOT_AVAILABLE_ERROR, fmt=fmt@entry=0x7eed80 "Register %d is not available")
    at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/common/common-exceptions.c:394
#6  0x0000000000533a59 in frame_unwind_register (frame=<optimised out>, regnum=834, buf=<optimised out>) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/frame.c:1167
#7  0x0000000000533c2c in frame_unwind_register_unsigned (frame=0x1b2aab0, regnum=regnum@entry=834) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/frame.c:1275
#8  0x0000000000533c6c in get_frame_register_unsigned (frame=<optimised out>, regnum=regnum@entry=834) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/frame.c:1282
#9  0x00000000004097f2 in riscv_breakpoint_kind_from_pc (gdbarch=0x1547600, pcptr=<optimised out>) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/riscv-tdep.c:170
#10 0x00000000004a50b1 in default_breakpoint_from_pc (gdbarch=0x1547600, pcptr=<optimised out>, lenptr=0x7ffcfaae4bac) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/arch-utils.c:847
#11 0x00000000004b8962 in program_breakpoint_here_p (gdbarch=<optimised out>, address=18446744071562069744) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9087
#12 0x00000000004b8b0c in bp_loc_is_permanent (loc=0x152c5f0) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9130
#13 add_location_to_breakpoint (b=b@entry=0x1531f70, sal=sal@entry=0x7ffcfaae4c40) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9067
#14 0x00000000004b8b7d in init_raw_breakpoint (b=0x1531f70, gdbarch=<optimised out>, sal=..., bptype=bp_breakpoint, ops=<optimised out>)
    at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:7574
#15 0x00000000004b9acf in init_breakpoint_sal (b=b@entry=0x1531f70, gdbarch=gdbarch@entry=0x1547600, location=location@entry=0x1a1cb10, filter=filter@entry=0x0, cond_string=0x0, extra_string=0x0, 
    type=bp_breakpoint, disposition=disp_donttouch, thread=-1, task=0, ignore_count=0, ops=0xb92300 <bkpt_breakpoint_ops>, from_tty=1, enabled=1, flags=0, display_canonical=0, internal=0, sals=...)
    at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9274
#16 0x00000000004bfe1f in create_breakpoint_sal (display_canonical=0, flags=0, internal=0, enabled=1, from_tty=1, ops=0xb92300 <bkpt_breakpoint_ops>, ignore_count=0, task=0, thread=-1, 
    disposition=disp_donttouch, type=bp_breakpoint, extra_string=0x0, cond_string=0x0, filter=0x0, location=0x1a1cb10, gdbarch=0x1547600, sals=...)
    at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9410
#17 create_breakpoints_sal (gdbarch=0x1547600, canonical=0x7ffcfaae4f00, cond_string=0x0, extra_string=0x0, type=bp_breakpoint, disposition=disp_donttouch, thread=-1, task=0, ignore_count=0, 
    ops=0xb92300 <bkpt_breakpoint_ops>, from_tty=1, enabled=1, internal=0, flags=0) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9464
#18 0x00000000004c058a in create_breakpoint (gdbarch=0x1547600, location=location@entry=0x1a1cc50, cond_string=cond_string@entry=0x0, thread=-1, thread@entry=0, extra_string=0x0, 
    extra_string@entry=0x1a0ac4e "", parse_extra=parse_extra@entry=1, tempflag=0, type_wanted=bp_breakpoint, ignore_count=0, pending_break_support=AUTO_BOOLEAN_AUTO, ops=0xb92300 <bkpt_breakpoint_ops>, 
    from_tty=1, enabled=1, internal=0, flags=0) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9886
#19 0x00000000004c0a29 in break_command_1 (arg=0x1a0ac4e "", flag=<optimised out>, from_tty=1) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/breakpoint.c:9994
#20 0x0000000000440ae3 in cmd_func (cmd=0x14072a0, args=0x1a0ac42 "start_kernel", from_tty=1) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/cli/cli-decode.c:1913
#21 0x00000000005f3d20 in execute_command (p=<optimised out>, p@entry=0x1a0ac40 "b start_kernel", from_tty=1) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/top.c:674
#22 0x00000000005270bc in command_handler (command=0x1a0ac40 "b start_kernel") at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/event-top.c:590
#23 0x00000000005273d8 in command_line_handler (rl=<optimised out>) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/event-top.c:780
#24 0x00000000005266cc in gdb_rl_callback_handler (rl=0x1a01ac0 "") at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/event-top.c:213
[... cut ...]
#34 0x0000000000527ce3 in catch_errors (func=func@entry=0x5766c0 <captured_command_loop(void*)>, func_args=func_args@entry=0x0, errstring=errstring@entry=0x7a29b4 "", mask=mask@entry=RETURN_MASK_ALL)
    at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/exceptions.c:236
#35 0x000000000057754e in captured_main (data=0x7ffcfaae52e0) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/main.c:1148
#36 gdb_main (args=args@entry=0x7ffcfaae5410) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/main.c:1158
#37 0x00000000004082d5 in main (argc=<optimised out>, argv=<optimised out>) at /work/risc-v/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/gdb.c:32
```

slavaim, 2017-02-19:
I temporarily fixed the problem by hard coding the returned value for ```riscv_breakpoint_kind_from_pc```

```
static int
riscv_breakpoint_kind_from_pc (struct gdbarch *gdbarch, CORE_ADDR *pcptr)
{
  //
  // a temporary fix to allow GDB to work with QEMU gdbstub
  //
  gdbarch_tdep (gdbarch)->supports_compressed_isa = SUP_NO;
  return 4;

......
}
```

sandeepvijay, 2017-10-18:
I got the same issue and this helped me. Thank you.

You can also disable the usage of compressed breakpoint in GDB using "set riscv use_compressed_breakpoint off".

michaeljclark, 2018-02-08:
The problem go away when you disable compressed breakpoints?

I'm not exactly sure how the GDB/QEMU breakpoint system works, however, this piece of code comes to mind:

https://github.com/riscv/riscv-qemu/blob/c41459f3d1641281ef99c9aa510e53b933b1aa62/target/riscv/translate.c#L1875-L1885

which increments the PC by 4, and wouldn't work if the breakpoint was a 2-byte c.ebreak.

You could test by letter the debugger set compressed breakpoints and change the increment to two.

sorear, 2018-02-10:
@michaeljclark That block of code is used for "hardware" breakpoints, not ebreak instructions

michaeljclark, 2018-02-10:
@sorear still curious why that code has a constant offset of 4

sorear, 2018-02-10:
Looks like it was copied from somewhere and 1 should work just as well.

michaeljclark, 2018-02-10:
BTW the QEMU GDB stub only handles registers 0-64

- 0 (hardwired zero register)
- 1-31 (integer registers)
- 32 (program counter)
- 33 - 64 (fpu registers)

Is 834 a CSR perhaps? I'd like to see the GDB XML register descriptor for RISC-V

michaeljclark, 2018-02-11:
It's a CSR. misa I believe.

Reading GDB source we see that the CSR register offset is 65.

https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.29/gdb/riscv-tdep.h#L112

0x301 (misa) is 769 in decimal + 65 = 834

I can modify the gdb stubs in riscv-qemu to add CSR support:

https://github.com/riscv/riscv-qemu/blob/riscv-next/target/riscv/gdbstub.c

michaeljclark, 2018-02-11:
I've made a commit that should fix this in QEMU:

https://github.com/michaeljclark/riscv-qemu/commit/ff56fb3002882ce9348b304f152daaf163d7c7d7

vsiles, 2018-05-09:
Sorry to dig up an old thread, but I installed riscv-tools/riscv-qemu a couple days back, and I had the same kind of error when trying to set a breakpoint with GDB:
```
(gdb) b *0x80000010
Register 3921 is not available
```

Disabling the compressed breakpoints seems to fix the problem.

lazyparser, 2021-03-04:
> Sorry to dig up an old thread, but I installed riscv-tools/riscv-qemu a couple days back, and I had the same kind of error when trying to set a breakpoint with GDB:
> 
> ```
> (gdb) b *0x80000010
> Register 3921 is not available
> ```
> 
> Disabling the compressed breakpoints seems to fix the problem.

Hi @vsiles ,

Have you observed this error recently? Can we close this issue safely and mark it as fixed?

vsiles, 2021-03-04:
Hi !
I'm afraid I can't tell. I didn't work with qemu/riscv for 3 years now. Last I tried (3 years ago) the error was still there.

Can't repro now, the code is in my old company :D

lazyparser, 2021-03-04:
> Hi !
> I'm afraid I can't tell. I didn't work with qemu/riscv for 3 years now. Last I tried (3 years ago) the error was still there.
> 
> Can't repro now, the code is in my old company :D

Never mind. Thanks for you reply :-)

jim-wilson, 2021-03-04:
This was fixed when I added an xml gdb stub to the RISC-V qemu port.  That happened about 2 and a half years ago I think.  There should be (maybe closed) issues elsewhere like riscv-binutkils-gdb or riscv-qemu that explains this in more detail, but the details are probably not important.

closed 2021-03-05 by lazyparser