riscv-collab/riscv-gnu-toolchain #218: lr and sc in atomic

susi655, 2017-02-27:
how to specify lr and sc values in assembly instruction for atomic operations.
Tried writing like amoadd.w 0, t2, t1, (t0) illegal operands. 
amoadd.w t2, t1, (t0) is working.

aswaterman, 2017-02-27:
Not sure I understand the question exactly, but:

lr.w t1, (t0) is the correct syntax for lr.w.

sc.w t2, t1, (t0) is the correct syntax for sc.w.

susi655, 2017-02-28:
Sorry for miscommunication!
I wanted to ask regarding aq and rl in atomic instructions.
without specifying aq and rl the assembly instruction is **amoadd.w t2, t1, (t0).** This instruction is taken by tool chain. But where to keep values for aq and rl. Tried **amoadd.w 0, t2, t1, (t0)** . aqrl value is 0. Tool chain is giving error "illegal operands".

[image]

aswaterman, 2017-02-28:
It is part of the instruction name: amoadd.w.aq, amoadd.w.rl, amoadd.w.aqrl

susi655, 2017-02-28:
I tried above instructions. amoadd.w.aq, amoadd.w.rl are working. But amoadd.w.aqrl is giving unrecognized opcode.

aswaterman, 2017-02-28:
you must be using an older version of the assembler (you can use "sc" instead of "aqrl" in the mean time)

susi655, 2017-02-28:
it's working. thank you

susi655, 2017-03-17:
Where can I find assembly instruction that can be properly detected by tool chain. Currently I am facing issue with writing assembly instruction of branch.
ex: blt t0,t1,8 is not giving what is expected.

palmer-dabbelt, 2017-03-30:
Here's the opcode list https://github.com/riscv/riscv-binutils-gdb/blob/riscv-next/opcodes/riscv-opc.c

I'm closing this, as I think the original issue is done.

closed 2017-03-30 by palmer-dabbelt