riscv-collab/riscv-gnu-toolchain #219: Debug information produced by riscv-gcc has incorrect line numbers

markww, 2017-02-28:
I'm having some problems with riscv64 gcc producing incorrect line numbers in the debug line table. I'm using rv64g gcc from riscv-gnu-toolchain commit hash 914224 (the revision pointed to by the head of the riscv-tools repo). In the following small example program (which I will refer to from this point on as reproducer.c):

```C
static void fn(void) {
}

static void marker() {}

int main() {
  fn();
  marker();
  return 0;
}
```

the function call to "marker" is reported to be on line 7 when it should be on line 8. You can watch this occur yourself by running the following script (adapted from one I used to minimize the reproducer for this problem):

```bash
#!/bin/bash

set -e
set -x

rm -f marker_grep_result.txt objdump-out.txt jump-to-marker-line.txt reproducer.o reproducer.exe addr2line_out.txt

${RISCV}/bin/riscv64-unknown-elf-gcc -O0 -g -o reproducer.o -c reproducer.c
${RISCV}/bin/riscv64-unknown-elf-gcc -o reproducer.exe  reproducer.o
fgrep -n 'marker();' reproducer.c > marker_grep_result.txt
MARKER_LINE_NO=`cat marker_grep_result.txt | cut -d ':' -f 1 | xargs`
${RISCV}/bin/riscv64-unknown-elf-objdump --disassemble reproducer.exe > objdump-out.txt
grep 'jal.*<marker>' objdump-out.txt > jump-to-marker-line.txt
MARKER_JUMP_ADDRESS=`cat jump-to-marker-line.txt | cut -d ':' -f 1 | xargs`
${RISCV}/bin/riscv64-unknown-elf-addr2line --exe=reproducer.exe ${MARKER_JUMP_ADDRESS} > addr2line_out.txt
REPORTED_LINE=`cat addr2line_out.txt | cut -d ':' -f 2 | grep -o '^[0-9]*' | xargs`
if [ ${REPORTED_LINE} -eq ${MARKER_LINE_NO} ] ; then
  exit 1
fi

```

Essentially, this script compiles reproducer.c into an executable. It finds the line number upon which the call to marker occurs. Then, it looks in the objdump of that executable for the address at which the jump and link to marker occurs, and asks addr2line where the debug information claims the call to marker occurs. This script exits with 0 if all commands pass and the line numbers differ.

Although the line number skew is only 1 in the submitted reproducer, the differences pile up in realistic programs and produce very inaccurate debug line numbers. This has become a serious impediment on a RISCV project I am helping out with, and I would greatly appreciate assistance with fixing this issue.

aswaterman, 2017-02-28:
I won't have time to look into this any time soon - @kito-cheng any ideas if we are missing another DWARF linker relaxation?

markww, 2017-03-01:
Just for reference, it appears that this issue first appeared in riscv-gnu-toolchain commit ef50015

kito-cheng, 2017-03-01:
We have found this problem last month and working for that now, it's seem cause by wrong DW_CFA_advance_loc info after relax.

palmer-dabbelt, 2017-03-30:
I think this is fixed by Kito's commit to binutils.

closed 2017-03-30 by palmer-dabbelt