riscv-collab/riscv-gnu-toolchain #222: Unexistant riscv-binutils-gdb version

rbarzic, 2017-03-28:
When  trying to checkout with  the command :
git clone --recursive https://github.com/riscv/riscv-gnu-toolchain  

I get the following message:

Unable to checkout '4057c3f7252f50c71074e5be51043aa39ab02441' in submodule path 'riscv-binutils-gdb'

Looks like the reference to riscv-binutils-gdb has been changed very recently but points to a non-existing hash

rbarzic, 2017-03-28:
Running "git checkout master"  in riscv-binutils-gdb 
and running:
../configure --with-arch=rv32g --prefix=/opt/riscv32i-2017-04-28
fails with the following error message:

checking whether makeinfo --split-size=5000000 supports @click... yes
checking for default auto-load directory... $debugdir:$datadir/auto-load
checking for default auto-load safe-path... $debugdir:$datadir/auto-load
configure: error: configuration riscv32-unknown-elf is unsupported.
Makefile:10424: recipe for target 'configure-gdb' failed
make[2]: *** [configure-gdb] Error 1

palmer-dabbelt, 2017-03-28:
Sorry about that.  This commit has been archived: https://github.com/riscv/riscv-binutils-gdb/commit/4057c3f7252f50c71074e5be51043aa39ab02441

closed 2017-03-28 by palmer-dabbelt