riscv-collab/riscv-gnu-toolchain #223: "-mcmodel=medany" doesn't pass the GCC test suite

palmer-dabbelt, 2017-03-31:
We've got a handful of additional failures when running the GCC tests with "-mcmodel-medany".  Here's the patch I used to run it

````
diff --git a/Makefile.in b/Makefile.in
index 2901ac5..8c35c8d 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -284,7 +284,7 @@ check-gcc-linux: stamps/build-gcc-linux-stage2
        export DEJAGNULIBS=$(srcdir)/riscv-dejagnu && \
        export RISC_V_SYSROOT=$(SYSROOT) && \
        cd build-gcc-linux-stage2 && \
-       $(MAKE) check-gcc "RUNTESTFLAGS=--target_board=riscv-sim"
+       $(MAKE) check-gcc "RUNTESTFLAGS=--target_board=riscv-sim/-mcmodel=medany"
 
 clean:
        rm -rf build-* $(addprefix src/,$(PACKAGES)) stamps
````

aswaterman, 2017-03-31:
With or without the new relaxation?

palmer-dabbelt, 2017-03-31:
Without, and probably also with (it's running now).

palmer-dabbelt, 2017-03-31:
````
                                      |      rv32ima   |    rv32imafd   |      rv64ima   |    rv64imafd   |     rv32imaf   |     rv64imaf   |
gcc         # of expected failures    |    179 (    0) |    179 (    0) |    178 (    0) |    178 (    0) |    179 (    0) |    178 (    0) |
            # of expected passes      |  90931 (  -17) |  91038 (  -20) |  91891 (  -54) |  91995 (  -57) |  91042 (  -14) |  91999 (  -54) |
            # of unexpected failures  |     92 (   85) |    102 (   88) |    123 (  116) |    137 (  119) |     97 (   82) |    131 (  116) |
            # of unexpected successes |      2 (    0) |      2 (    0) |      3 (    0) |      3 (    0) |      2 (    0) |      3 (    0) |
            # of unresolved testcases |      3 (    3) |      3 (    3) |      1 (    1) |      3 (    1) |      3 (    3) |      1 (    1) |
            # of unsupported tests    |   2084 (    2) |   2026 (    2) |   1892 (    2) |   1832 (    2) |   2026 (    2) |   1834 (    2) |
g++         # of expected failures    |    368 (    0) |    368 (    0) |    368 (    0) |    368 (    0) |    368 (    0) |    368 (    0) |
            # of expected passes      |  98754 (   64) |  98754 (   64) | 100636 (   64) | 100636 (   64) |  98754 (   64) | 100636 (   64) |
            # of unexpected failures  |     12 (    0) |     12 (    0) |     12 (    0) |     12 (    0) |     12 (    0) |     12 (    0) |
            # of unexpected successes |      0 (    0) |      0 (    0) |      0 (    0) |      0 (    0) |      0 (    0) |      0 (    0) |
            # of unresolved testcases |      0 (    0) |      0 (    0) |      0 (    0) |      0 (    0) |      0 (    0) |      0 (    0) |
            # of unsupported tests    |   4421 (   18) |   4421 (   18) |   4393 (   18) |   4393 (   18) |   4421 (   18) |   4393 (   18) |
````

palmer-dabbelt, 2017-05-04:
It does now.

closed 2017-05-04 by palmer-dabbelt