riscv-collab/riscv-gnu-toolchain #225: Provide clear guidance for riscv64ilp32

sorear, 2017-04-04:
1. In any situation where memory bandwidth and memory size are cost drivers but register file size is not, there will be a desire to use 32-bit pointers on hardware with 64-bit ALUs and memory systems.
2. Closed non-ABI systems like Java have private mechanisms for use of 32-bit pointers (`UseCompressedOops`).  These mechanisms are preferable, anyway, since they support much larger process sizes (32GB for the Java version).
3. Use of 32-bit pointers in C and in languages without specific support (e.g. Perl) requires splitting the workload into pieces (usually processes) of 4GB or smaller, and then using an ABI which has 32-bit pointers.
4. The RISC-V ISA provides two choices here: RV32G (with the UXL mechanism), or RV64G with an ILP32 ABI.
5. RV32G is supported by the toolchain as it exists right now but appears to be inferior in other respects.  In particular it denies access to 64-bit integers for functions that benefit from them and (for systems enough total memory for it to matter) requires CPU and toolchain support for two mutually incompatible ISAs.  RV32GC does provide a more compact encoding for FLW but int64 operations are likely to be more common than float32 on the systems this ticket is describing.
6. Is the above analysis agreed on?  If so, we can start laying the groundwork for that, by documenting what a riscv64ilp32 ELF executable looks like in the ELF spec, and adding wording to the ISA spec that "RV32 is intended for embedded systems, larger systems which desire small pointers should use RV64 with an appropriate ABI".  In particular let's make clear that the differences between RV32 and RV64 are on purpose and in no sense an "unforced error".  Then toolchain support can be implemented by interested parties as time passes.
7. Basic principles of a calling convention for that would be 32-bit callee-saved registers, 32-bit *signed* pointers (with a possible future [-2GiB,0) user-mode carveout; negative user pointers are already common on 32-bit systems), all kernel-loaded ELF segments in the 32-bit region, mmap either forced to 32-bit region or implementing `MAP_32BIT`.
8. If the above analysis is disagreed with and all `sizeof(void*)==4` code should be RV32, then that should also be indicated with wording in the ISA spec and in the ABI documentation.
9. The above considerations recur with 64-bit code on 128-bit systems, although 128-bit systems may never be numerous enough for standardization to matter.

sorear, 2017-04-04:
and someone just did the ELF research so I don't have to.  Quoth https://sourceware.org/ml/binutils/2017-04/msg00019.html:

> x86-64 has the x32 sub-mode, but that is (bogusly imo) indicated
by using a 32-bit ELF instead of the default 64-bit ELF one. IA-64
uses a bit in e_flags (and it looks like some other architectures,
like MIPS, do too), which is the approach I'd expect to be used
everywhere when mixed bitness in a single executable is not
intended to be permitted. I can't seem to find examples of section
flags indicating bitness - ARM's Thumb mode, for example, works
by inserting special marker labels instead (thus allowing mixed
code inside a section).

aswaterman, 2017-04-04:
I agree with points 1-5.

The signed pointer issue seems problematic, though.  Load/store effective addresses are computed in 64-bit arithmetic, so e.g. ```lb t0, 1(t0)```, where ```t0=0x000000007fffffff```, loads from address ```0x0000000080000000```, not ```0xffffffff80000000```.  There's a similar issue with AUIPC addressing.

I guess you could alias the ```0x0000000080000000``` and ```0xffffffff80000000``` ranges so array indexing works across the 2 GiB boundaries?

I agree there are problems with zero-extending pointers but sign-extending all other types.  I'm just not yet sure which is worse.

sorear, 2017-04-04:
@aswaterman For the purposes of this comment I am treating the ISA as a fixed given.  Since LWU has no compressed forms, using it for pointers would be seriously disadvantaged.

The anomalies that result from mixing 64-bit adds in effective address generation with 32-bit sign-extending adds for pointer arithmetic only appear if objects are allowed to span the watershed from 2GB to -2GB.  I propose that they wouldn't, and that the page at address `0x7ffff000` be banned from allocation in MAP_32BIT.  (We don't want objects in the last page itself due to a LUI anomaly and due to the C one-past-the-end rule.)

aswaterman, 2017-04-04:
Very true.

closed 2021-03-08 by lazyparser