riscv-collab/riscv-gnu-toolchain #238: Permission deniened when building toolchain

sheepduke, 2017-05-05:
## Background
I am writing an ebuild (package specificaion for Gentoo Linux)  for `riscv-tools`. I created a individual package for `riscv-gnu-toolchain` but it failed to compile, because Gentoo requires that package cannot touch any fire or directory outside its working directory.

## Expected Behavior
For almost all softwares, `make` and `install` should be two different stages. Thus, it is possible to `make` a software locally, then `make install` it with `sudo` privilege.

However, `riscv-gnu-toolchain` does not follow this pattern. During its installation it needs to create some directory which requires `sudo` privilege.

It may be because of the nature of cross compiling tools. But is it possible to solve this?

## How to Reproduce
1. Compile `riscv-gnu-toolchain` as normal user.
2. After ~1 minutes on i5-4670K, it will just fail.

## Relevant Trace
make[2]: Nothing to be done for 'install-target'.
mkdir -p -- /usr/local/riscv64-unknown-elf /usr/local/riscv64-unknown-elf
mkdir: cannot create directory ‘/usr/local/riscv64-unknown-elf’: Permission denied
mkdir: cannot create directory ‘/usr/local/riscv64-unknown-elf’: Permission denied
make[2]: *** [Makefile:2501: installdirs] Error 1
make[2]: Leaving directory '/home/sheep/temp/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib'
make[1]: *** [Makefile:2256: install] Error 2
make[1]: Leaving directory '/home/sheep/temp/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib'
make: *** [Makefile:201: stamps/build-binutils-newlib] Error 2

## Thank you
This section is used for expressing appreciation ;-)

palmer-dabbelt, 2017-05-05:
Can you just use crossdev to build the RISC-V toolchain on Gentoo?  riscv-gnu-toolchain is essentially a distro in its own right, so I don't think it makes a whole lot of sense to use it other distros.  If you use crossdev you'll also have the advantage of getting your toolchain built like the rest of the Gentoo toolchains, and you'll have riscv64-unknown-linux-gnu-emerge so you can build software :).

Here's my Gentoo scripts and packages, they're pretty out of date but now that RISC-V is upstream most of this should go away

  https://github.com/palmer-dabbelt/riscv-gentoo-infra
  https://github.com/palmer-dabbelt/riscv-gentoo-infra

sheepduke, 2017-05-06:
On 2017-05-05 17:52, Palmer Dabbelt <notifications@github.com> wrote:
> Can you just use crossdev to build the RISC-V toolchain on Gentoo?  riscv-gnu-toolchain is essentially a distro in its own right, so I don't think it makes a whole lot of sense to use it other [...]

Well, I'm not sure if my understanding is correct, but what I need is to
build *bare metal* cross compiler on a Gentoo host, not to build Gentoo
for RISC-V architecture.

I did not found any RISC-V support in crossdev arch list...Is it
possible for crossdev?

P.S. Anyway, at least I can use the build.sh script and install it to
some place...So it's not really a big problem for me ;)

Thanks.

closed 2017-05-08 by sheepduke