riscv-collab/riscv-gnu-toolchain #244: Builds on systems with guile fail

ilg-ul, 2017-05-17:
The GDB configure script picks the guile installed in the system folders instead of the desired local guile.

If the versions differ, the build fails.

This happened on ARCH, where the system guile is pretty new.

Suggestion:
- fix the GDB `configure` script to avoid using the system guile, and always use the version that comes with the GDB sources.

aswaterman, 2017-05-17:
Hey Liviu,

We could try to work around this in the riscv-gnu-toolchain/Makefile.in; is
it possible to address the problem there? (e.g. --with-guile)?

Modifying GDB's configure script is outside of my competence, but if you
can get it fixed in upstream GDB, we'll rebase on top of it.

ilg-ul, 2017-05-17:
Hi Andrew,

I guess using the `--with-guile` option in makefile might be enough until the upstream GDB is fixed.

aswaterman, 2017-05-17:
Do you mind prototyping it and submitting a PR?  We don't use Arch and so can't replicate the problem locally.

ilg-ul, 2017-05-17:
> can't replicate the problem locally

the problem is not ARCH specific, just install a recent guile (apt-get install guile-2.0-dev on Ubuntu) and the behaviour should be the same.

promach, 2017-06-02:
I have done the autoconf after doing git pull on the new riscv-gnu-toolchain commit, but the error persists.

Did I miss anything else ?

jim-wilson, 2017-12-23:
You have to specify --with-guile=no when running configure to disable guile.  Or alternatively, install an older version of guile that gdb will work with, and then use --with-guile=X to specify that version number.  Disabling guile is simpler.

jim-wilson, 2018-01-25:
Fixed by the --with-guile configure flag.

closed 2018-01-25 by jim-wilson