riscv-collab/riscv-gnu-toolchain #245: Unable to build riscv-pk with riscv64-unknown-linux-gnu-gcc

liujed, 2017-05-19:
I am trying to follow the [Linux/RISC-V Installation Manual](https://riscv.org/software-tools/#linuxman) to get Linux running in the RISC-V ISA simulator. The last step is to build riscv-pk against glibc:
> `../configure --prefix=$RISCV --host=riscv64-unknown-linux-gnu --with-payload=<riscv-linux>/vmlinux`

When I do this, however, I get what appear to be parse errors for the `#define`s of `stdin` and friends in `file.h`:
```
riscv64-unknown-linux-gnu-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"bbl_payload\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/frontend.c
In file included from ../pk/syscall.c:5:0:
../pk/file.h:17:23: error: expected ')' before '+' token
 #define stdin  (files + 0)
                       ^
../pk/file.h:18:23: error: expected ')' before '+' token
 #define stdout (files + 1)
                       ^
../pk/file.h:19:23: error: expected ')' before '+' token
 #define stderr (files + 2)
                       ^
Makefile:302: recipe for target 'syscall.o' failed
make: *** [syscall.o] Error 1
make: *** Waiting for unfinished jobs....
```
Things seem to build fine with Newlib (i.e, if I use `--host=riscv64-unknown-elf` instead).

palmer-dabbelt, 2017-05-19:
It's fine to build pk/bbl with riscv64-unknown-elf, that's what I always to.  I'm looking into the build failure.

palmer-dabbelt, 2017-05-19:
I can't reproduce this locally.  Are you on the latest version of riscv-tools?

liujed, 2017-05-20:
I was on riscv-tools @ 1b27f68 from a couple days ago. Now, when I do a recursive clone of riscv-tools, I can't seem to find commit hash 3f21b5c for riscv-binutils-gdb to initialize `riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb`.

liujed, 2017-05-20:
Things appear to be building now. Thanks.

closed 2017-05-20 by liujed