riscv-collab/riscv-gnu-toolchain #248: Wrong line information for undefined symbols

palmer-dabbelt, 2017-06-22:
When building the following code

````
$ cat ~/main.c 
void func();
void foo();

int main(void)
{
        int i;

        func();

    foo();

    return 0;
}
````

I get worse error messages on RISC-V than on x86: specifically our line information just points to main instead of to the correct line.

````
$ riscv64-unknown-linux-gnu-gcc ~/main.c -g
/tmp/ccHnRjU2.o: In function `main':
/home/palmer/main.c:5: undefined reference to `func'
/home/palmer/main.c:5: undefined reference to `foo'
collect2: error: ld returned 1 exit status
$ gcc ~/main.c -g                                                                                                                                                                      
/tmp/ccVAYKMq.o: In function `main':
/home/palmer/main.c:8: undefined reference to `func'
/home/palmer/main.c:10: undefined reference to `foo'
````

I think this might have something to do with our internal labels showing up all over the place

````
$ riscv64-unknown-linux-gnu-objdump -d -l main.o 

main.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <main>:
main():
/home/palmer/main.c:8
   0:   1141                    addi    sp,sp,-16

0000000000000002 <.LCFI0>:
   2:   e406                    sd      ra,8(sp)
   4:   e022                    sd      s0,0(sp)

0000000000000006 <.LCFI1>:
   6:   0800                    addi    s0,sp,16

0000000000000008 <.L0 >:
/home/palmer/main.c:10
   8:   00000097                auipc   ra,0x0
   c:   000080e7                jalr    ra

0000000000000010 <.L0 >:
  10:   00000097                auipc   ra,0x0
  14:   000080e7                jalr    ra

0000000000000018 <.L0 >:
/home/palmer/main.c:12
  18:   4781                    li      a5,0

000000000000001a <.L0 >:
  1a:   853e                    mv      a0,a5
  1c:   60a2                    ld      ra,8(sp)

000000000000001e <.LCFI3>:
  1e:   6402                    ld      s0,0(sp)

0000000000000020 <.LCFI4>:
  20:   0141                    addi    sp,sp,16

0000000000000022 <.LCFI5>:
  22:   8082                    ret
````

````
$ riscv64-unknown-linux-gnu-objdump -d -l -r main.o                                                                                                                                    

main.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <main>:
main():
/home/palmer/main.c:8
   0:   1141                    addi    sp,sp,-16

0000000000000002 <.LCFI0>:
   2:   e406                    sd      ra,8(sp)
   4:   e022                    sd      s0,0(sp)

0000000000000006 <.LCFI1>:
   6:   0800                    addi    s0,sp,16

0000000000000008 <.L0 >:
/home/palmer/main.c:10
   8:   00000097                auipc   ra,0x0
                        8: R_RISCV_CALL func
                        8: R_RISCV_RELAX        *ABS*
   c:   000080e7                jalr    ra

0000000000000010 <.L0 >:
  10:   00000097                auipc   ra,0x0
                        10: R_RISCV_CALL        foo
                        10: R_RISCV_RELAX       *ABS*
  14:   000080e7                jalr    ra

0000000000000018 <.L0 >:
/home/palmer/main.c:12
  18:   4781                    li      a5,0

000000000000001a <.L0 >:
  1a:   853e                    mv      a0,a5
  1c:   60a2                    ld      ra,8(sp)

000000000000001e <.LCFI3>:
  1e:   6402                    ld      s0,0(sp)

0000000000000020 <.LCFI4>:
  20:   0141                    addi    sp,sp,16

0000000000000022 <.LCFI5>:
  22:   8082                    ret
````

but I'm not 100% sure this is a binutils bug and not a GCC bug, so I'm just filing it here for now.

palmer-dabbelt, 2017-07-17:
This appears to be a binutils bug.  The generated FDE looks like this

````
00000010 0000000000000034 00000000 FDE cie=00000000 pc=0000000000000000..0000000000000000
  DW_CFA_advance_loc4: 0 to 0000000000000000
  DW_CFA_def_cfa_offset: 16
  DW_CFA_advance_loc4: 2 to 0000000000000002
  DW_CFA_offset: r1 at cfa-8
  DW_CFA_advance_loc4: 4 to 0000000000000006
  DW_CFA_restore: r1
  DW_CFA_advance_loc4: 22 to 000000000000001c
  DW_CFA_def_cfa_offset: 0
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
````

which contains a PC range of 0:0, which is incorrect.  This causes the various tools to blow up looking up symbols.

The FDE's PC range is fine in the linked binary because it's filled out by relocation.  I think the fix here is to fill out a non-zero PC before relocation, which reflects the object file, while continuing to relocate the corresponding .8byte like we're doing now so the debug info is correct in the executable.

palmer-dabbelt, 2017-07-17:
@kuanlinchentw Do you happen to know how to both emit non-zero bits and generate a fixup in GAS?

kuanlinchentw, 2017-07-18:
@palmer-dabbelt In our internal build toolchain, the line number is correct. I'll try to figure out what happen in the github version and fix it.

jim-wilson, 2017-12-01:
Reproduced, and confirmed that it is fixed by issue #98.  So closing.

closed 2017-12-01 by jim-wilson