riscv-collab/riscv-gnu-toolchain #251: configure: WARNING: unrecognized options: --with-xlen

salasjad, 2017-06-27:
I am trying to install the RV32I toolchain with the riscv32-unknown-elf- prefix. However, adding `--with-xlen=32` or `--enable-multilib` doesn't seem to work.

`./configure --with-xlen=32 --with-guile=no --prefix=/opt/riscv`
`make`

Building on Arch Linux, gcc version 7.1.1

aswaterman, 2017-06-27:
I think you want ```--with-arch=rv32i``` rather than ```--with-xlen=32```

salasjad, 2017-06-27:
Thanks @aswaterman,`--with-arch=rv32i` creates the riscv32-unknown-elf- files. I get an error saying `riscv64-unknown-elf-as: unrecognized option '-m32'`, when I try to compile the code. 

From Makefile:
`CFLAGS=-m32 -msoft-float`

It this related to the toolchain build process?

TM1234, 2017-08-25:
-m32 and -msoft-float were deprecated a while back in favour of -march and -mabi as far as I recall.

jim-wilson, 2018-01-25:
The answer was provided by tm1234.

closed 2018-01-25 by jim-wilson