riscv-collab/riscv-gnu-toolchain #252: error while building riscv-gnu-toolchain

CircleLee114, 2017-07-07:
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:644:11: error: ‘B1200’ was not declared in this scope
     1200, B1200
           ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:648:11: error: ‘B1800’ was not declared in this scope
     1800, B1800
           ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:652:11: error: ‘B2400’ was not declared in this scope
     2400, B2400
           ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:656:11: error: ‘B4800’ was not declared in this scope
     4800, B4800
           ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:660:11: error: ‘B9600’ was not declared in this scope
     9600, B9600
           ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:594:16: error: ‘EXTA’ was not declared in this scope
 #define B19200 EXTA
                ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:664:12: note: in expansion of macro ‘B19200’
     19200, B19200
            ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:598:16: error: ‘EXTB’ was not declared in this scope
 #define B38400 EXTB
                ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:668:12: note: in expansion of macro ‘B38400’
     38400, B38400
            ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c: In function ‘int hardwire_setbaudrate(serial*, int)’:
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:741:28: error: aggregate ‘hardwire_ttystate state’ has incomplete type and cannot be defined
   struct hardwire_ttystate state;
                            ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c: In function ‘int hardwire_setstopbits(serial*, int)’:
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:780:28: error: aggregate ‘hardwire_ttystate state’ has incomplete type and cannot be defined
   struct hardwire_ttystate state;
                            ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c: In function ‘int hardwire_setparity(serial*, int)’:
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:825:28: error: aggregate ‘hardwire_ttystate state’ has incomplete type and cannot be defined
   struct hardwire_ttystate state;
                            ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:837:19: error: ‘PARENB’ was not declared in this scope
       newparity = PARENB | PARODD;
                   ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:837:28: error: ‘PARODD’ was not declared in this scope
       newparity = PARENB | PARODD;
                            ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:862:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c: In function ‘int hardwire_setstopbits(serial*, int)’:
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:818:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c: In function ‘int hardwire_setbaudrate(serial*, int)’:
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:775:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c: In function ‘int hardwire_noflush_set_tty_state(serial*, serial_ttystate, serial_ttystate)’:
/home/circle/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/ser-unix.c:243:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[3]: *** [ser-unix.o] Error 1
make[2]: *** [all-gdb] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-binutils-newlib] Error 2

palmer-dabbelt, 2017-07-08:
This looks like a generic upstream problem, not a problem with the RISC-V port.  This guy had the same problem <http://www.math.fsu.edu/~okhanmoh/gnu_linux_notes.html> but doesn't have a solution.

This comment <https://github.com/riscv/riscv-binutils-gdb/blob/riscv-next/gdb/ser-unix.c#L455> indicates it's some esoteric stuff.  Maybe your system headers don't define B1200 and friends?

jim-wilson, 2018-01-25:
B1200 and other macros are defined in <termios.h> and/or <termio.h>.  Gdb configure autodetects whether these header files are available, and uses them if available.  If they aren't available, and you try to compile this file, then yes, you will get an error.  You need to provide the missing files, or don't build a unix gdb.  It looks like gdb could be improved to not use these macros when both termios.h and termio.h are missing, but that is a gdb problem not a RISC-V problem.  If you can reproduce, it should be reported to the gdb developers at
    https://sourceware.org/gdb/bugs/

closed 2018-01-25 by jim-wilson