riscv-collab/riscv-gnu-toolchain #253: Default ISA does not match documentation

stkaplan, 2017-07-07:
README.md says the default ISA is RV64G, but configure.ac sets the default to build RV64GC. Not sure which of those is wrong.

palmer-dabbelt, 2017-07-07:
The readme is wrong.

cbatten, 2017-07-12:
Is there a reason why RV64GC is the default? I could imagine many architectures will support RGV64G but not RV64GC, so maybe RV64G is a better default? Just a thought.

aswaterman, 2017-07-12:
It represents our guess that the C extension will be standard for general-purpose implementations running Linux.  I think most cores that omit C will be unusual in at least one additional way (custom accelerators, research prototypes, incomplete educational implementations) and so using a compiler target other than the default won't be much of an additional onus.

cbatten, 2017-07-12:
Hmmm ... why do you think the C extension will be standard for general-purpose implementations running linux? The C extension's primary benefit is in code footprint which seems less critical if we are running full-blow linux.

cbatten, 2017-07-12:
Plus the riscv-gnu-toolchain is used not just for linux but also for bare-metal embedded systems which may or may not include the C extensions. Since G = general, it just seemed like that was the right variant to support by default.

cbatten, 2017-07-12:
I think you are going to say "Waterman shows that RVC fetches 25%-30% fewer instruction bits, which reduces instruction cache misses by 20%-25%, or roughly the same performance impact as doubling the instruction cache size [28]" so that the C extensions are useful for general-purpose implementations too :)

cbatten, 2017-07-12:
I guess another question is: Does the RISCV linux port _require_ RV64GC? or it only requires RV64G? It it only requires RV64G then that seems like it would be the right default for the gcc toolchain. If it requires RV64GC then I guess that would be the right default .. although I would also wonder if the linux port _really_ requires RV64G.

aswaterman, 2017-07-12:
I wasn't planning on quoting myself, but yes, that was the point I was going to make :)  In any case, this repo supports building the other compilers with an argument to configure, so I don't think this is a big deal.

While the Linux kernel port doesn't require C, I believe the Linux distros plan to ship GC binaries by default (a decision I support, but not a decision that is mine to make).

cbatten, 2017-07-12:
Ah -- well, if the Linux distros plan to ship RV64GC binaries then RV64GC will indeed become the "defacto" standard for general-purpose RISC-V systems.

palmer-dabbelt, 2017-07-12:
Our Linux port requires IMA, but could easily be made to just require I: removing M only requires importing some libgcc functions, and the A dependency was only added on Monday when upstream suggested it wasn't worth bothering with non-A systems (which removed two special syscalls and ~100 LOC).  I don't think anyone building Linux-capable systems thinks M or A impose a reasonable cost, so I don't plan on bothering with this.

Linux is largely oblivious to the C extension: there's a Kconfig option to enable C, but all it does is compress the kernel code.  A non-C kernel should run C userspace fine, and vice-versa (assuming the C extension is present in hardware).  As far as userspace goes, the default set of multilibs on Linux systems are
 * rv32imac
 * rv32imafdc
 * rv64imac
 * rv64imafdc
but that's an easy change to make -- like Andrew said, we're really leaving this up to distributions.  I expect that the first distribution ports that emerge will be RV64GC and that they'll later release multilib versions.

In embedded land the default ISA doesn't really matter: anything we pick would be wrong.  Users are expected to just set -march according to their hardware.

cbatten, 2017-07-12:
Sounds good -- exciting to see RISC-V already upstream in binutils and gcc, and to see you and Andrew's hard work on getting RISC-V upstream in linux and glibc. Thanks for your hard work on this!

stkaplan, 2017-07-20:
Looks like this got fixed. Thanks!

closed 2017-07-20 by stkaplan