riscv-collab/riscv-gnu-toolchain #254: problem installing newlib

esolera, 2017-07-10:
I got this problem when a trying installing Newlib

touch stamp-system-gdbinit
make[5]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/data-directory'
make[5]: Entering directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib'
make[6]: Entering directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib'
make[7]: Entering directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib/import'
make  all-recursive
make[8]: Entering directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib/import'
make[9]: Entering directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib/import'
make[9]: Nothing to be done for 'all-am'.
make[9]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib/import'
make[8]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib/import'
make[7]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib/import'
make[6]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib'
make[5]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb/build-gnulib'
make[4]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb'
make[3]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib/gdb'
make[2]: Nothing to be done for 'all-target'.
make[2]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
make[1]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
make -C build-binutils-newlib install
make[1]: Entering directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
make[2]: Entering directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
/bin/bash /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb/mkinstalldirs /opt/riscv /opt/riscv
mkdir -p -- /opt/riscv /opt/riscv
mkdir: cannot create directory ‘/opt/riscv’: Permission denied
mkdir: cannot create directory ‘/opt/riscv’: Permission denied
Makefile:2501: recipe for target 'installdirs' failed
make[2]: *** [installdirs] Error 1
make[2]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
Makefile:2255: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-gnu-toolchain/build-binutils-newlib'
Makefile:210: recipe for target 'stamps/build-binutils-newlib' failed
make: *** [stamps/build-binutils-newlib] Error 2

palmer-dabbelt, 2017-07-10:
It appears you don't have write access to /opt/riscv.

AntonKrug, 2017-08-14:
Wouldn't be better to close the issue? Looks like it was permission denied issue and esolera doesn't respond.

palmer-dabbelt, 2017-08-14:
Works for me.

closed 2017-08-14 by palmer-dabbelt

TONG369, 2020-08-13:
I have meet this problem, what should I do to solve?
@palmer-dabbelt @esolera @AntonKrug @asb @arunthomas