riscv-collab/riscv-gnu-toolchain #255: Undocumented branch instruction in libgcc

Mimoja, 2017-07-12:
gcc rv32i emits (invalid?) bnez and beqz instructions

`btype opcode: 0x63 funct3: 0x2 rs1: 0x16 rs2: 0x0 imm: 0xFFFFFFCC`
instruction in libgcc:

Toolchain was build with: 

> ./configure --prefix=/opt/riscv -with-arch=rv32i
> sudo make -j8 linux
> sudo make install

`/opt/riscv/bin/riscv32-unknown-elf-gcc math.c -o math_broken -nostdlib -lgcc`
is used to compile this math.c:

`
int _start(){
    int a = 2;
    int b = 5;
    int c = a + b;
    int d = a * c;
    int e = d / 4;
    return d;
}
`

`/opt/riscv/riscv32-unknown-linux-gnu/bin/objdump -d math_broken`
>math_broken:     file format elf32-littleriscv
Disassembly of section .text:
>
>00010054 <_start>:
   10054:	fd010113          	addi	sp,sp,-48
   10058:	02112623          	sw	ra,44(sp)
   1005c:	02812423          	sw	s0,40(sp)
   10060:	03010413          	addi	s0,sp,48
   10064:	00200793          	li	a5,2
   10068:	fef42623          	sw	a5,-20(s0)
   1006c:	00500793          	li	a5,5
   10070:	fef42423          	sw	a5,-24(s0)
   10074:	fec42703          	lw	a4,-20(s0)
   10078:	fe842783          	lw	a5,-24(s0)
   1007c:	00f707b3          	add	a5,a4,a5
   10080:	fef42223          	sw	a5,-28(s0)
   10084:	fec42783          	lw	a5,-20(s0)
   10088:	fe442703          	lw	a4,-28(s0)
   1008c:	00070593          	mv	a1,a4
   10090:	00078513          	mv	a0,a5
   10094:	03c000ef          	jal	ra,100d0 <__mulsi3>
   10098:	00050793          	mv	a5,a0
   1009c:	fef42023          	sw	a5,-32(s0)
   100a0:	fe042783          	lw	a5,-32(s0)
   100a4:	41f7d713          	srai	a4,a5,0x1f
   100a8:	00377713          	andi	a4,a4,3
   100ac:	00f707b3          	add	a5,a4,a5
   100b0:	4027d793          	srai	a5,a5,0x2
   100b4:	fcf42e23            	sw	a5,-36(s0)
   100b8:	fe042783          	lw	a5,-32(s0)
   100bc:	00078513          	mv	a0,a5
   100c0:	02c12083          	lw	ra,44(sp)
   100c4:	02812403          	lw	s0,40(sp)
   100c8:	03010113          	addi	sp,sp,48
   100cc:	00008067          	ret
>
>000100d0 <__mulsi3>:
   100d0:	00050613          	mv	a2,a0
   100d4:	00000513          	li	a0,0
   100d8:	0015f693          	andi	a3,a1,1
   100dc:	00068463          	beqz	a3,100e4 <__mulsi3+0x14>
   100e0:	00c50533          	add	a0,a0,a2
   100e4:	0015d593          	srli	a1,a1,0x1
   100e8:	00161613          	slli	a2,a2,0x1
   100ec:	fe0596e3          	bnez	a1,100d8 <__mulsi3+0x8>
   100f0:	00008067          	ret

At 0x100dc lies the problem. 0x00068463 contains a valid BRANCH opcode with invalid funct3: 0b010
With comparison to page 104 of the riscv-2.2 spec this is not defined. Interpreted as different typed does not make it any more clear:
rtype opcode: 0x63 rd: 0xD funct3: 0x1 rs1: 0xB rs2: 0x0 funct7: 0x7F
itype opcode: 0x63 rd: 0xD funct3: 0x1 rs1: 0xB imm: 0xFFFFFFE0
btype opcode: 0x63 funct3: 0x2 rs1: 0x16 rs2: 0x0 imm: 0xFFFFFFCC
stype opcode: 0x63 funct3: 0x1 rs1: 0xB rs2: 0x0 imm: 0xFFFFFFED
utype opcode: 0x63 rd: 0xD imm: 0xFE059000
jtype opcode: 0x63 rd: 0xD imm: 0xFFF597E0

Am i decoding wrong or is this a bug in gcc?

aswaterman, 2017-07-12:
Those are pseudoinstructions.  ```bnez``` is ```bne``` with rs2=x0.

Mimoja, 2017-07-12:
I know. Bug was on my side, when decoding funct3, not in understanding BNEZ

closed 2017-07-12 by Mimoja