riscv-collab/riscv-gnu-toolchain #257: Unusable GCC from 32 bit install

TheThirdOne, 2017-07-22:
# Setup

After a fresh install using
```
./configure --prefix=$INSTALLDIR --with-arch=rv32g --with-abi=ilp32d
make linux
```

# Issues

The gcc installed seems to be entirely unusable.

Compiling c files results in:
```
$ riscv32-unknown-linux-gnu-gcc -o test.o test.c
riscv32-unknown-linux-gnu-gcc: error trying to exec 'cc1': execvp: No such file or directory
```

Linking into an executable fails:
```
$ riscv32-unknown-linux-gnu-gcc -o test test.o
riscv32-unknown-linux-gnu-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found
compilation terminated.
```

And compiling assembly results in:
```
$ riscv32-unknown-linux-gnu-gcc -o test.o test.s 
Assembler messages:
Fatal error: invalid -march= option: `rv32g'
```

Despite AS working fine:
```
$ riscv32-unknown-linux-gnu-as -o test.o test.s 
$
```

The reason for the last issue appears to be that it is using my system GNU assembler rather than the riscv32 one. Running with verbose results in:

```
gcc version 7.1.1 20170509 (GCC) 
COLLECT_GCC_OPTIONS='-o' 'test.o' '-v' '-march=rv32g' '-mabi=ilp32d'
 as -v --traditional-format -march=rv32g -mabi=ilp32d -o /tmp/ccv36jBv.o test.s
GNU assembler version 2.26.1 (x86_64-linux-gnu) using BFD version (GNU Binutils for Ubuntu) 2.26.1
Assembler messages:
Fatal error: invalid -march= option: `rv32g'
``` 

# Context

I have successfully installed the 64 bit gcc and it works just fine (on all of the issues above), but I need a 32 bit version.

I would like this resolved fairly quickly, so if there are workarounds which aren't proper solutions that's fine. I only really need assemble and link with libc. It should be possible to link with libc manually, but I don't know how to find the right files to link with. 
 
If it matters, this is on Ubuntu 16.04.2 LTS (Xenial Xerus)

jim-wilson, 2018-01-25:
It isn't clear what went wrong, and probably not possible to debug this late after the bug was reported.
The compiler should be using the assembler in $INSTALLDIR/riscv32-unknown-linux-gnu/bin/as for instance.  Does this file exist?

You can get the linker command by adding the gcc -v option to your compile command.  But that assumes that gcc is working.

TheThirdOne, 2018-01-26:
This was a while ago so I'm not complete sure, but I believe the assembler and linker were both present and worked correctly. I don't remember what workaround I found, but I found something to allow me to move forward. 

I remember saving an image of the broken installation, but I can't find it right now. I'll try to reproduce the error tomorrow.

jim-wilson, 2018-07-31:
No comment in 6 months, so close.

closed 2018-07-31 by jim-wilson