riscv-collab/riscv-gnu-toolchain #260: problem with multilib

esolera, 2017-07-29:
I do this comand : 

 ./configure --prefix=/opt/riscv --enable-multilib
make linux

and i have that issue:
checking for C compiler default output file name... 
configure: error: in `/home/edgar/Desktop/manual/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/lib32/ilp32/libgomp':
configure: error: C compiler cannot create executables
See `config.log' for more details.
Makefile:13569: recipe for target 'configure-target-libgomp' failed
make[2]: *** [configure-target-libgomp] Error 1
make[2]: Leaving directory '/home/edgar/Desktop/manual/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage2'
Makefile:891: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/edgar/Desktop/manual/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage2'
Makefile:184: recipe for target 'stamps/build-gcc-linux-stage2' failed
make: *** [stamps/build-gcc-linux-stage2] Error 2

palmer-dabbelt, 2017-08-01:
It's expected this works.  Can you try:
````
/configure --prefix=/opt/riscv --enable-multilib --enable-linux
make
````

if that works, we have some makefile/autoconf bug in riscv-gnu-toolchain.  If that doesn't work, then I'll need to see the config.log in /home/edgar/Desktop/manual/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/lib32/ilp32/libgomp

jim-wilson, 2018-01-25:
No config.log file provided, so no further help can be provided.

closed 2018-01-25 by jim-wilson

gengtsh, 2019-06-18:
i encounter the same problem: configure: error: in `/home/XX/Work/RISCV/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/lib32/ilp32/libgomp':
configure: error: C compiler cannot create executables
Makefile:13569: recipe for target 'configure-target-libgomp' failed

please see my config. log
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ../configure --prefix=/home/XX/Work/RISCV --enable-multilib --enable-linux

## --------- ##
## Platform. ##
## --------- ##

hostname = XX-thinkpad
uname -m = x86_64
uname -r = 4.18.0-21-generic
uname -s = Linux
uname -v = #22~18.04.1-Ubuntu SMP Thu May 16 15:07:19 UTC 2019

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games
PATH: /snap/bin
PATH: /home/XX/Work/RISCV/bin
PATH: /home/XX/Work/Chisel3/verilator/bin
PATH: /home/XX/Work/Compiler/Cool/class/bin
PATH: /home/XX/Work/RISCV/bin
PATH: /home/XX/Work/Chisel3/verilator/bin
PATH: /home/XX/Work/Compiler/Cool/class/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1895: checking for gcc
configure:1911: found /usr/bin/gcc
configure:1922: result: gcc
configure:2151: checking for C compiler version
configure:2160: gcc --version >&5
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2171: $? = 0
configure:2160: gcc -v >&5
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
[... 201 lines cut ...]
psdir='${docdir}'
runstatedir='${localstatedir}/run'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''
with_guile=''
with_system_zlib='--with-system-zlib'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
#define HAVE_LIBGMP 1
#define HAVE_LIBMPFR 1
#define HAVE_LIBMPC 1

configure: exit 0

jim-wilson, 2019-06-19:
Current theory is that there is non-multilib compiler already built with the same prefix, and this is confusing the build process into accidentally using the installed compiler instead of the just built stage2 to compile the 32-bit libgomp, which then fails because the installed compiler is non-multilib.

The workaround is to use a --prefix that doesn't already have a compiler in it.

A possible fix might be to build all supported languages in the stage1 gcc build, so that we fully overwrite any old compiler before starting the stage2 gcc build.