riscv-collab/riscv-gnu-toolchain #261: Is native run of the toolchain possible?

AntonKrug, 2017-08-01:
I was gonna compile the toolchain so it can run natively on the riscv linux.

I used this options just to try it out:

> ../configure --host=riscv64-unknown-linux-gnu --prefix=/opt/riscv-native

But the options were ignored and all the binaries got compiled with **"--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf"**. Is the native run of the compiler possible/supported?

palmer-dabbelt, 2017-08-01:
We don't quite implement the "--host" argument in the riscv-gnu-toolchain autoconf because we don't want it to go everywhere (we always want to build glibc for RISC-V, for example).  We added another argument for this that I use to build Windows toolchains, IIRC it's something like "--with-host".

AntonKrug, 2017-08-02:
With the --with-host I had some progress. I do following:

```
cd ~/riscv-gnu-toolchain
cd riscv-gcc
./contrib/download_prerequisites   # to add the GMP, MPFR and MPC
cd ../

rm -R -f native-build
mkdir native-build
cd native-build
 ../configure --with-host=riscv64-unknown-linux-gnu --without-system-zlib --prefix=/opt/riscv-native
make -j 2

```

And in the end, I get some binaries, but the build fails when the make file tries to run the native binaries in the later stages of the build process:

```
riscv64-unknown-linux-gnu-g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -fno-common  -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o xgcc gcc.o gcc-main.o ggc-none.o \
  c/gccspec.o  libcommon-target.a \
   libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
riscv64-unknown-linux-gnu-g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -fno-common  -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cpp gcc.o gcc-main.o ggc-none.o \
  c-family/cppspec.o  libcommon-target.a \
   libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
riscv64-unknown-elf-gcc -dumpspecs > tmp-specs
/bin/bash: /opt/riscv-native/bin/riscv64-unknown-elf-gcc: cannot execute binary file: Exec format error
Makefile:1994: recipe for target 'specs' failed
make[2]: *** [specs] Error 126
make[2]: *** Waiting for unfinished jobs....
rm gcov-tool.pod gcov-dump.pod gpl.pod cpp.pod gfdl.pod gcov.pod gcc.pod fsf-funding.pod
make[2]: Leaving directory '/home/antonkrug/riscv-gnu-toolchain/native-build/build-gcc-newlib-stage1/gcc'
Makefile:4233: recipe for target 'all-gcc' failed
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory '/home/antonkrug/riscv-gnu-toolchain/native-build/build-gcc-newlib-stage1'
Makefile:223: recipe for target 'stamps/build-gcc-newlib-stage1' failed
make: *** [stamps/build-gcc-newlib-stage1] Error 2
```

Now I ignore the error and I copied the binaries to the riscv root and booted it up. Running the binaries only prints on the console whatever argument I typed to run it and nothing else:

"gcc -v" will just output echo back "gcc -v" nothing else. Either my root is too minimalistic, or I forgot/messed up something in the process. Plus they are only **riscv64-unknown-elf-** and not **riscv64-unknown-linux-gnu-** so even if I will get them working they will still not produce binaries I will be able to run on native riscv linux.

What you reckon, should I probably ask on the mailing list, if anybody has an idea?

kito-cheng, 2017-08-02:
You still need --target= for cross build:
```
 ../configure --with-host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu --host=riscv64-unknown-linux-gnu --enable-languages=c,c++ --enable-tls --enable-shared
```
And then you can get what you want, btw I can't compile it with qemu-riscv64 since execve is not correctly work :P
```
$ file cross-native/bin/gcc
cross-native/bin/gcc: ELF 64-bit LSB executable, UCB RISC-V, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-riscv64-lp64d.so.1, for GNU/Linux 3.0.0, not stripped
$ qemu-riscv64 -L `riscv64-unknown-linux-gnu-gcc --print-sysroot` cross-native/bin/gcc  -v
Using built-in specs.
COLLECT_GCC=cross-native/bin/gcc
COLLECT_LTO_WRAPPER=/tmp/kito/build-world/cross-native/cross-native/bin/../libexec/gcc/riscv64-unknown-linux-gnu/8.0.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --host=riscv64-unknown-linux-gnu --prefix=/home/kito/riscv-workspace/build-world/cross-native/build/../cross-native --enable-languages=c,c++ --enable-tls --enable-shared
Thread model: posix
gcc version 8.0.0 20170509 (experimental) (GCC) 
```

AntonKrug, 2017-08-02:
With your advice I did this:

` ../configure --with-host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu --host=riscv64-unknown-linux-gnu --enable-languages=c,c++ --enable-tls --enable-shared --without-system-zlib --prefix=/opt/riscv-native
`

And got to this error:

```
Making all in argz
make[6]: Entering directory '/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/newlib/libc/argz'
riscv64-unknown-elf-gcc -B/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/antonkrug/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/antonkrug/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"2.5.0\" -DPACKAGE_STRING=\"newlib\ 2.5.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/antonkrug/riscv-gnu-toolchain/native-build/../riscv-newlib/newlib/libc/argz -DMISSING_SYSCALL_NAMES -fno-builtin      -Os  -mcmodel=medlow -c -o lib_a-dummy.o `test -f 'dummy.c' || echo '/home/antonkrug/riscv-gnu-toolchain/native-build/../riscv-newlib/newlib/libc/argz/'`dummy.c
riscv64-unknown-elf-gcc -B/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/antonkrug/riscv-gnu-toolchain/riscv-newlib/newlib/libc/include -B/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/antonkrug/riscv-gnu-toolchain/riscv-newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"2.5.0\" -DPACKAGE_STRING=\"newlib\ 2.5.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/antonkrug/riscv-gnu-toolchain/native-build/../riscv-newlib/newlib/libc/argz -DMISSING_SYSCALL_NAMES -fno-builtin      -Os  -mcmodel=medlow -c -o lib_a-argz_add.o `test -f 'argz_add.c' || echo '/home/antonkrug/riscv-gnu-toolchain/native-build/../riscv-newlib/newlib/libc/argz/'`argz_add.c
/bin/bash: /opt/riscv-native/bin/riscv64-unknown-elf-gcc: cannot execute binary file: Exec format error
Makefile:393: recipe for target 'lib_a-dummy.o' failed
make[6]: *** [lib_a-dummy.o] Error 126
/bin/bash: /opt/riscv-native/bin/riscv64-unknown-elf-gcc: cannot execute binary file: Exec format error
make[6]: *** Waiting for unfinished jobs....
Makefile:399: recipe for target 'lib_a-argz_add.o' failed
make[6]: *** [lib_a-argz_add.o] Error 126
make[6]: Leaving directory '/home/antonkrug/riscv-gnu-toolchain/native-build/build-newlib/riscv64-unknown-elf/newlib/libc/argz'

```

The size of the final binaries didn't changed (I deleted the output directory in meantime) and the binaries behave the same.

kito-cheng, 2017-08-02:
Humm, I am using iscv-gcc/configure to build a native gcc for riscv64-unknown-linux-gnu, it seems like your are using riscv-gnu-toolchain/configure? If so, Try to configure/make/make install with riscv-gcc/configure?

AntonKrug, 2017-08-02:
Yes the riscv-gnu-toolchain, so it looks I'm in the wrong repository. I will try your suggestion, but should I continue in this thread, or should I start a new thread in the riscv-gcc repository?

kito-cheng, 2017-08-02:
No, just keep in this thread is fine, I am writing rule for build native riscv linux toolchian, and under testing now.

kito-cheng, 2017-08-02:
I've create `native-toolchain` branch, you could try it with `make linux-native`, and then you can found they on $PREFIX/native/*.

AntonKrug, 2017-08-03:
I tried the risc-gcc and got linker errors first and on second run errors such as:

```
In file included from /home/antonkrug/riscv-gcc/native-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable.h:35:0,
                 from /home/antonkrug/riscv-gcc/native-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/unordered_map:47,
                 from /home/antonkrug/riscv-gcc/libstdc++-v3/include/precompiled/stdc++.h:117:
/home/antonkrug/riscv-gcc/native-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h: In member function 'std::pair<bool, long unsigned int> std::__detail::_Power2_rehash_policy::_M_need_rehash(std::size_t, std::size_t, std::size_t)':
/home/antonkrug/riscv-gcc/native-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h:604:25: error: could not convert 'std::make_pair<bool, int>(false, 0)' from 'std::pair<bool, int>' to 'std::pair<bool, long unsigned int>'
    return std::make_pair(false, 0);
           ~~~~~~~~~~~~~~^~~~~~~~~~
/home/antonkrug/riscv-gcc/native-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h:607:23: error: could not convert 'std::make_pair<bool, int>(false, 0)' from 'std::pair<bool, int>' to 'std::pair<bool, long unsigned int>'
  return std::make_pair(false, 0);

```

Then I checkout to your branch and did 

```
./configure --prefix=/opt/riscv
make linux-native
```

And got this error:

`make: *** No rule to make target '/home/antonkrug/native-branch/riscv-gnu-toolchain/riscv-gcc', needed by 'stamps/build-gcc-linux-native'.  Stop.
`

austinharris, 2017-09-01:
@kito-cheng FYI I had to add --with-native-system-header-dir=$(INSTALL_DIR)/native/include to the gcc configure otherwise it would incorrectly pick up sys/sdt.h from the host on RHEL7 if you have the systemtap-devel package installed.

palmer-dabbelt, 2017-09-01:
Can someone PR all this in so we don't lose it?

kito-cheng, 2017-09-02:
`sys/sdt.h` is a upstream bug, I've this issue in other ports too.

palmer-dabbelt, 2017-09-02:
Ya, and it's been there for a while.  I'd just been uninstalling systemtap, but this solution is much better -- that's why I wanted it committed somewhere, that way I won't forget :).

jim-wilson, 2018-07-31:
I believe this is fixed.  I've been able to use the linux-native target to build toolchain binaries, and then run them on a riscv64-linux system.

closed 2018-07-31 by jim-wilson

AntonKrug, 2018-08-01:
Is this including the gdb and gdb server?

jim-wilson, 2018-08-01:
No.  I have a gdb port in progress, It requires multiple linux kernel patches, and one glibc patch.  I haven't looked at gdb server yet.
https://github.com/jim-wilson/riscv-linux-native-gdb
See the README.md file for info on linux kernel and gdb patches.

snathicktechno, 2019-05-27:
Hi jim,
I had successfully built the riscv-linux native toolchain now i want the check the native toolchain on the target, i mean target which is riscv running ubuntu.
i copied the native folder to target and when running ./riscv64-uknown-linux-gnu test.c 
it is throwing an error command not found 
could you please resolve me this issue 

Thanking you,
Snathick

AntonKrug, 2019-05-27:
Ubuntu is ported to RISCV? I was aware only of arm/powerpc/ibm.

jim-wilson, 2019-05-27:
If you are using riscv-gnu-toolchain to build a native compiler, then you are almost certainly doing something wrong, as this is one of the worst possible ways to try to build a native compiler.  It would be better if you filed a new issue, and explained from the beginning what you are trying to do, and we can point you at a better way to do this.

When you get a "file not found" error from the shell, and the file does obviously exist, then it is the file interpreter that is missing.  For a shell script, this is the program specified on the first line.
% cat tmp.sh
#!/bin/nonexistent
exit 0
% ./tmp.sh
sh: 15: ./tmp.sh: not found
% 
This is the error message from a simple shell implementation.  Bash gives a better error.  For an ELF binary, this is the program specified in the .interp section, which for a dynamically linked binary will be the ld.so program.
rohan:2013$ objdump --section .interp --full-contents /bin/ls

/bin/ls:     file format elf64-x86-64

Contents of section .interp:
 0238 2f6c6962 36342f6c 642d6c69 6e75782d  /lib64/ld-linux-
 0248 7838362d 36342e73 6f2e3200           x86-64.so.2.    
rohan:2014$ 
So what the error message is telling you is that your binary specifies an ld.so file name that does not exist on your target.

When you cross compile code, the cross compiler must use the exact same version of glibc as the target, and it must be configured the exact same way.  Otherwise, programs that you compile will not run on the target.  Using the native target in riscv-gnu-toolchain will only work if the target OS was built using the same exact glibc version which is in riscv-gnu-toolchain.  And because this is an old obsolete pre-upstream version, any real linux distro will not have a matching glibc.  If you use the legacy v1_0 branch of freedom-u-sdk, you can build a matching linux system, but this is an old obsolete pre-upstream very limited buildroot system that isn't useful for any real development work.  It is only useful as a tutorial on how to build a riscv linux system.

So as I indicated at the beginning, whatever you are trying to do, you are apparently doing it the wrong way, and you should restart from the beginning and explain what you want to do.

jim-wilson, 2019-05-27:
No, there isn't a RISC-V Ubuntu port yet.

snathicktechno, 2019-05-28:
Thank you jim i will go through it

fkladdu, 2019-06-13:
How can we get a cross native Toolchain of RISC-V.

jim-wilson, 2019-06-13:
1) Use a real distro like Fedora/Debian/OpenSuse/OpenEmbedded that comes with a native compiler.  Basically, anything other than whatever unspecified distro you are using.
2) Try the info at
    https://gcc.gnu.org/ml/gcc-help/2019-06/msg00042.html
   This assumes that you already have a C library on the target.
3) Try using crosstool-ng if you don't already have a C library on the target.

Building a toolchain this way is complicated.  If you really want to do it, you should expect to spend at least a few weeks learning how to do it.  It is much easier to use a toolchain that someone else built if you don't want to become a toolchain building expert.

AntonKrug, 2019-06-13:
I concur with Jim, the crosstool-ng is great tool, but it's not coffee break task. If you are learning and you are struggling with the stuff, first simplify your project and requirements. And then use work of others to get you there 90% when you have proof of concept working and understand it better, then you can be building it from scratch. If you can just use real distros as much as you can. Even if you build your toolchain, will you need other libraries/packages/dependencies? Look at Manuel's work, doing it from scratch it's very involved process https://people.debian.org/~mafm/posts/2018/20180402_debian-gnulinux-port-for-risc-v-64-bit-riscv64-in-debian-infrastructure-debian-ports/

I'm maybe misunderstanding things and maybe missing something or you are mixing the terms.  

> I had successfully built the riscv-linux native toolchain now i want the check the native tool-chain on the target, i mean target which is riscv running ubuntu.  

It's not fully clear what you done and what you want to do, it confused me that you have running riscv port of ubuntu and that you already have on some riscv target running tool-chain which can compiled another toolchain which can target riscv.

https://crosstool-ng.github.io/docs/toolchain-types/

> How can we get a cross native Toolchain of RISC-V.

You mean you want to get toolchain which was not build on riscv, but will run on riscv and produce riscv binaries? If you previously said that you build native toolchain where build == host == target, why you would need tool-chain where build != host == target?

ImagineMiracle-wxn, 2019-07-12:
`./../riscv-gcc-riscv-gcc-8.2.0/configure --with-host=riscv64-unknown-linux-gcc --host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu --prefix=/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/opt/ --with-sysroot=/media/imaginemiracle/ImagineMiracle/riscv_project/freedom-u-sdk_bk/freedom-u-sdk/toolchain/sysroot --without-system-zlib --enable-shared --enable-tls --enable-languages=c,c++ --disable-libmudflap --disable-libssp --disable-libquadmath --disable-nls --disable-bootstrap --disable-checking --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc`

When I follow the above steps to build the Makefile, execute make.
Then an error, how can I solve such a problem?

`/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h:582:5:   required from here
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:131:12: error: 'value' is not a member of 'std::is_copy_assignable<bool>'
     struct __and_<_B1, _B2>
            ^~~~~~~~~~~~~~~~
In file included from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_algobase.h:64:0,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/char_traits.h:39,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/ios:40,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/istream:38,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/sstream:38,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/complex:45,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/ccomplex:39,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-riscv-gcc-8.2.0/libstdc++-v3/include/precompiled/stdc++.h:52:
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h: In instantiation of 'struct std::pair<bool, long unsigned int>':
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h:582:5:   required from here
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:367:7: error: 'value' is not a member of 'std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<long unsigned int> >'
       operator=(typename conditional<
       ^~~~~~~~
In file included from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/move.h:55:0,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/nested_exception.h:40,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-riscv-gcc-8.2.0/libstdc++-v3/libsupc++/exception:144,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/ios:39,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/istream:38,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/sstream:38,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/complex:45,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/ccomplex:39,
                 from /media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-riscv-gcc-8.2.0/libstdc++-v3/include/precompiled/stdc++.h:52:
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits: In instantiation of 'struct std::__not_<std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<long unsigned int> > >':
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:378:7:   required from 'struct std::pair<bool, long unsigned int>'
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h:582:5:   required from here
/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:142:31: error: 'value' is not a member of 'std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<long unsigned int> >'
[... cut ...]
Makefile:418: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3'
Makefile:12115: recipe for target 'all-target-libstdc++-v3' failed
make[1]: *** [all-target-libstdc++-v3] Error 2
make[1]: Leaving directory '/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build'
Makefile:909: recipe for target 'all' failed
make: *** [all] Error 2`

ImagineMiracle-wxn, 2019-07-12:
`Makefile:1742: recipe for target 'riscv64-unknown-linux-gnu/bits/stdc++.h.gch/O2ggnu++0x.gch' failed
make[4]: *** [riscv64-unknown-linux-gnu/bits/stdc++.h.gch/O2ggnu++0x.gch] Error 1
make[4]: Leaving directory '/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include'
Makefile:511: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3'
Makefile:418: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3'
Makefile:12115: recipe for target 'all-target-libstdc++-v3' failed
make[1]: *** [all-target-libstdc++-v3] Error 2
make[1]: Leaving directory '/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/riscv-gcc-build'
Makefile:909: recipe for target 'all' failed
make: *** [all] Error 2`

ImagineMiracle-wxn, 2019-07-12:
I want to build gcc that can run on the RISCV architecture.

ImagineMiracle-wxn, 2019-07-12:
`riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:131:12: error: 'value' is not a member of 'std::is_copy_assignable<bool>'
     struct __and_<_B1, _B2>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:367:7: error: 'value' is not a member of 'std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<long unsigned int> >'
       operator=(typename conditional<
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:142:31: error: 'value' is not a member of 'std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<long unsigned int> >'
     : public __bool_constant<!bool(_Pp::value)>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:378:7: error: 'value' is not a member of 'std::__not_<std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<long unsigned int> > >'
       operator=(typename conditional<
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:131:12: error: 'value' is not a member of 'std::is_move_assignable<bool>'
     struct __and_<_B1, _B2>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:384:7: error: 'value' is not a member of 'std::__and_<std::is_move_assignable<bool>, std::is_move_assignable<long unsigned int> >'
       operator=(typename conditional<
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:131:12: error: 'value' is not a member of 'std::is_copy_assignable<bool>'
     struct __and_<_B1, _B2>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:367:7: error: 'value' is not a member of 'std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<int> >'
       operator=(typename conditional<
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:142:31: error: 'value' is not a member of 'std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<int> >'
     : public __bool_constant<!bool(_Pp::value)>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:378:7: error: 'value' is not a member of 'std::__not_<std::__and_<std::is_copy_assignable<bool>, std::is_copy_assignable<int> > >'
       operator=(typename conditional<
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:131:12: error: 'value' is not a member of 'std::is_move_assignable<bool>'
     struct __and_<_B1, _B2>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:384:7: error: 'value' is not a member of 'std::__and_<std::is_move_assignable<bool>, std::is_move_assignable<int> >'
       operator=(typename conditional<
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:131:12: error: 'value' is not a member of 'std::is_constructible<bool, const bool&>'
     struct __and_<_B1, _B2>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:101:45: error: 'value' is not a member of 'std::__and_<std::is_constructible<bool, const bool&>, std::is_constructible<long unsigned int, const int&> >'
         is_constructible<_T2, const _U2&>>::value;
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:102:7: error: body of constexpr function 'static constexpr bool std::_PCC<<anonymous>, _T1, _T2>::_ConstructiblePair() [with _U1 = bool; _U2 = int; bool <anonymous> = true; _T1 = bool; _T2 = long unsigned int]' not a return-statement
       }
/riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/type_traits:131:12: error: 'value' is not a member of 'std::is_constructible<bool, bool&&>'
     struct __and_<_B1, _B2>
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:115:40: error: 'value' is not a member of 'std::__and_<std::is_constructible<bool, bool&&>, std::is_constructible<long unsigned int, int&&> >'
         is_constructible<_T2, _U2&&>>::value;
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/stl_pair.h:116:7: error: body of constexpr function 'static constexpr bool std::_PCC<<anonymous>, _T1, _T2>::_MoveConstructiblePair() [with _U1 = bool; _U2 = int; bool <anonymous> = true; _T1 = bool; _T2 = long unsigned int]' not a return-statement
       }
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h:594:25: error: could not convert 'std::make_pair<bool, int>(false, 0)' from 'std::pair<bool, int>' to 'std::pair<bool, long unsigned int>'
    return std::make_pair(false, 0);
riscv-gcc/riscv-gcc-build/riscv64-unknown-linux-gnu/libstdc++-v3/include/bits/hashtable_policy.h:597:23: error: could not convert 'std::make_pair<bool, int>(false, 0)' from 'std::pair<bool, int>' to 'std::pair<bool, long unsigned int>'
  return std::make_pair(false, 0);
`

jim-wilson, 2019-07-12:
I would not recommend using the native rules in the riscv-gnu-toolchain Makefile.  It was a useful experiment a long time ago when we didn't have working distros, but it is one of the least useful ways to build a native compiler, and is now just a distraction.  IMHO I think we should drop the native rules from the Makefile so people stop trying to use them.

Note that a native compiler will only work if built with the same glibc version as the target, which has been configured the same way as the target.  Since riscv-gnu-toolchain is unlikely to have the same glibc version and configuration as the target, if you did succeed in building a compiler, it is unlikely that it would actually run on the target.

The best way to get a native compiler is to grab any distro that comes with one.  Fedora, Debian, OpenSuse, Open Embedded, etc.  And then build natively.

If you already have an OS running on your target, and want a compiler that works with it, then the best way to get a native compiler is to create a sysroot.  Traditionally this is /usr/lib, /lib, and /usr/include, but it depends on the OS and may require other dirs like /lib64 and /usr/lib64.  Once you have a sysroot, you can pass it to a binutils configure using --with-sysroot, build and install.  Then do the same for gcc using the same prefix.  This gives you a cross compiler that matches the target.  Now you can use this cross compiler to build a native compiler, again using the sysroot when configuring.

Note that cross building a native compiler is complicated, and if you have never done it before it may take a few weeks to learn enough about the compiler build process to learn how to do it.  It is far easier to just install a distro that includes a compiler.

You can also use crosstool-ng to cross build a native compiler, but you would have to learn how crosstool-ng configuration works which takes some time.  But crosstool-ng does have some nice docs that explains how the compiler build process works.  Even if you don't use crosstool-ng, it is worth reading them.
http://crosstool-ng.github.io/docs/toolchain-construction/

Also, note, if you have a Hifive Unleashed, and are still running the system that it shipped with, then you are wasting your time trying to build a native compiler.  It ships with a very limited buildroot system that isn't useful for anything other than proving that the board can boot linux.  You need to replace this with something else if you want to do anything useful with the board.  You can drop a distro rootfs onto the sdcard and chroot to it.  Or you can try using current github.com/sifive/freedom-u-sdk to build a better system that can boot directly into debian.  This requires changing the boot switches on the board though.  See the docs.

Please file new issues for new problems.

ImagineMiracle-wxn, 2019-07-12:
Thanks for your answer, I have built a cross-compilation tool. My current idea is to use this tool to compile riscv-gcc to generate a native compiler.what should I do？

ImagineMiracle-wxn, 2019-07-12:
Now, I am using the following configuration method, using your suggestion --with-sysroot

I try again.

`./../riscv-gcc-riscv-gcc-8.2.0/configure --with-host=riscv64-unknown-linux-gcc --host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu --prefix=/media/imaginemiracle/ImagineMiracle/riscv_project/riscv-gcc/opt/ --with-sysroot=/media/imaginemiracle/ImagineMiracle/riscv_project/freedom-u-sdk_bk/freedom-u-sdk/toolchain/sysroot --without-system-zlib --enable-shared --enable-tls --enable-languages=c,c++ --with-abi=lp64d --with-arch=rv64imafdc`

jim-wilson, 2019-07-13:
The only time I ever cross-built a native RISC-V compiler was when I built the first native gcc Ada compiler, which requires cross compilation because it is written in Ada.  Otherwise, it really is better to use a distro or something like buildroot or open-embedded or crosstoool-ng that has scripts that already know how to cross build a native compiler.  Anyways, when I cross-built my native Ada compiler I used these scripts.  You can ignore the --enable-languages setting since you aren't building ada.
[scripts.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3389234/scripts.txt)
So conceptually this is pretty simple, the hard part is figuring out what do when something goes wrong.  The easiest way to handle gmp, mpfr, and mpc is to drop them into the gcc source tree alongside libgcc and libstdc++, and just let the gcc Makefile build them as libraries.

ImagineMiracle-wxn, 2019-07-15:
@jim-wilson I am very grateful to you, I will try it out with your method.

cghalibkhan, 2019-08-01:
https://github.com/riscv/riscv-gnu-toolchain/issues/261#issuecomment-496161754
* How did u build riscv-linux native toolchain.
* can u plz tell me the steps.

Thank's
ghalib khan.