riscv-collab/riscv-gnu-toolchain #263: How do I know what combination of ISA is supported by toolchain now? what about RV32IMFDC?

zhenbohu, 2017-08-09:
Hi, Dear Experts

I have a very idiot question, but I cannot find the answer from the github README here.

How do I know what combination of ISA is supported by toolchain now? actually I just want to use RV32IMFDC arch to develop a core, but not sure if the toolchain right now is fully support it or not?

Can you point me a clear answer for this? or just kindly help to confirm if RV32IMFDC is supported or not by toolchain now?

Thanks
Bob

palmer-dabbelt, 2017-08-10:
I've written a somewhat detailed description of the GCC arch/abi support that should go live on Monday morning on SiFive's website that describes this in much greater detail: https://www.sifive.com/blog/

The short answer is: there is good support for "-march=rv32imafdc" in the toolchain now.  If you're doing embedded stuff it's well tested, if you're looking to do Linux it's expected to work, but gets less testing than the rv64 port.  We're getting Linux a lot more stable so that should be sorted out by the end of the year.

zhenbohu, 2017-08-11:
Hi, Palmer

Thanks for your information. I have designed a processor core without the "A" extension instrucitons supported. So I actually dont want to use the "A" extension. So what I want to use is architecture "IMFDC"  not the "IMAFDC". Do you know if the "IMFDC" combination is supported by the tools now?

Thanks
Bob

palmer-dabbelt, 2017-08-11:
rv32imfdc (and rv64imfdc) are supported by the toolchain, but we don't consider them first-tier targets for support.  This means:
* There won't be a multilib built for this target.  You should build your toolchain with "--disable-linux --disable-multilib --with-arch=rv32imfdc --with-abi=rv32d", which will give you a non-multilib toolchain (along with libgcc and newlib's libc).
* The Linux port currently requires the A extension.  We can relax this requirement without too much work.
* We don't test this target as part of our regression suite.  You should check out the regression branch of risc-gnu-toolchain and make sure it passes the GCC test suite (run "make report"), but since rv32imafdc is supported and the A extension isn't really used by the compiler you should be fine.

jim-wilson, 2018-01-25:
Was answered by Palmer.

closed 2018-01-25 by jim-wilson