riscv-collab/riscv-gnu-toolchain #264: build on windows msys2 failed

manageryzy, 2017-08-11:
```
build/gengtype.exe  \
                    -S /c/work/riscv-gnu-toolchain/riscv-gcc/gcc -I gtyp-input.list -w tmp-gtype.state
/c/work/riscv-gnu-toolchain/riscv-gcc/gcc/../libcpp/include/line-map.h: No such file or directory

```

version : v20170612
command : ` ./configure --prefix=/c/work/port && make`
error at : build-gcc-newlib-stage1/gcc/Makefile: s-gtype
 `x86_64-pc-msys-gcc` linked `fopen` in mscvrt . unix style path got error

palmer-dabbelt, 2017-08-11:
Sorry, we just cross-compile for Windows.  Do you know if the upstream GCC builds?  This doesn't look like a RISC-V-specific problem.

TM1234, 2017-08-25:
If I recall correctly you need to modify Makefile.in to work around this build problem on MinGW-64/MSYS2. Change this:
```
stamps/build-binutils-newlib: $(srcdir)/riscv-binutils-gdb
	rm -rf $@ $(notdir $@)
	mkdir $(notdir $@)
	cd $(notdir $@) && $</configure \
		--target=$(NEWLIB_TUPLE) \
		$(CONFIGURE_HOST) \
		--prefix=$(INSTALL_DIR) \
		@with_guile@ \
		--disable-werror
	$(MAKE) -C $(notdir $@)
	$(MAKE) -C $(notdir $@) install
	mkdir -p $(dir $@) && touch $@
```
to add --src=../riscv-gcc:

```
stamps/build-binutils-newlib: $(srcdir)/riscv-binutils-gdb
	rm -rf $@ $(notdir $@)
	mkdir $(notdir $@)
	cd $(notdir $@) && $</configure \
		--target=$(NEWLIB_TUPLE) \
		$(CONFIGURE_HOST) \
		--prefix=$(INSTALL_DIR) \
		@with_guile@ \
		--disable-werror \
		--src=../riscv-gcc
	$(MAKE) -C $(notdir $@)
	$(MAKE) -C $(notdir $@) install
	mkdir -p $(dir $@) && touch $@
```

palmer-dabbelt, 2017-08-25:
Can you submit a patch?

TM1234, 2017-08-25:
Sorry I was just posting from memory but I don't build on MinGW64-MSYS2 anymore - I now use a variant of Liviu's GNU MCU Eclipse scripted Docker build process to build on Linux for Linux and Windows. So I don't really have a patch to submit.

mmicko, 2017-11-30:
@palmer-dabbelt  Created pull request with similar solution, needed to alter so we support all variations of actual build folder

jim-wilson, 2017-12-19:
Fixed by #304 which has been merged in.

closed 2017-12-19 by jim-wilson