riscv-collab/riscv-gnu-toolchain #267: undefined reference to __modsi3

codelec, 2017-08-19:
riscv64-unknown-elf-gcc -I./../env -I./common -I./median -I./qsort -I./rsort -I./towers -I./vvadd -I./multiply -I./mm -I./dhrystone -I./spmv -I./mt-vvadd -I./mt-matmul -I./pmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -fno-common -fno-builtin-printf -march=rv32i -mabi=ilp32 -static -nostdlib -nostartfiles -lgcc -T ./common/test.ld -o median.riscv ./median/median_main.c ./median/median.c ./common/syscalls.c ./common/crt.S
/tmp/ccc310MB.o: In function .L47': syscalls.c:(.text+0x12c): undefined reference to__umodsi3'
/tmp/ccc310MB.o: In function .L56': syscalls.c:(.text+0x154): undefined reference to__udivsi3'
syscalls.c:(.text+0x164): undefined reference to `__umodsi3'
collect2: error: ld returned 1 exit status
Makefile:54: recipe for target 'median.riscv' failed
make: *** [median.riscv] Error 1

palmer-dabbelt, 2017-08-21:
These functions are in libgcc.  You're building with "-nostdlib" so you're not getting libgcc.

codelec, 2017-08-21:
thanks a lot. removing -nostdlib fixes it 
why is -nostdlib there here https://github.com/riscv/riscv-tests/blob/master/benchmarks/Makefile#L42

closed 2017-08-21 by palmer-dabbelt

Lustrew, 2019-05-19:
Hello, I have met this new error. Could someone help me. I have change https://github.com/riscv/riscv-tests/blob/master/benchmarks/Makefile#L42 to 
```makefile
RISCV_LINK_OPTS ?= -static -nostartfiles -lgcc -T $(src_dir)/common/test.ld
```
Here is the error.
```shell
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccF99tIO.o: in function `.L5':
mm.c:(.text+0x15c): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: mm.c:(.text+0x18c): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: mm.c:(.text+0x1bc): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: mm.c:(.text+0x1f4): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccF99tIO.o: in function `.L7':
mm.c:(.text+0x288): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccF99tIO.o:mm.c:(.text+0xcf0): more undefined references to `fma' follow
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/cchddMes.o: in function `.L0 ':
mm_main.c:(.text+0x30): undefined reference to `__sync_fetch_and_add_4'
collect2: error: ld returned 1 exit status
Makefile:56: recipe for target 'mm.riscv' failed
make: *** [mm.riscv] Error 1
```

jim-wilson, 2019-05-19:
I see no __modsi3 error here.  This appears to be completely unrelated to the original problem.  New issues for new problems please.

fma is a math library function.  Add -lm to your linker options.  This was in RISCV_LINK_OPTS before you modified it.

__sync_fetch_and_add_4 is a 4-byte synchronization primitive.  It will be emitted inline if you compile for a target with the A extension.  If you compile for a target without the A extension, then it isn't possible to implement it.  The problem here is that you can't run a multithreaded testcase on a target without the A extension.

Lustrew, 2019-05-19:
ok, thank you!