riscv-collab/riscv-gnu-toolchain #280: SSL subject name and target host name mismatch

akhileshian, 2017-09-18:
Hey guys,
I'm trying to run the following command:
git submodule update --init --recursive

after cloning the riscv repository and cd ing into it.

However, I'm seeing the following error:
  error: SSL: certificate subject name 'qemu.org' does not match target host name 'git.qemu-project.org' while accessing https://git.qemu-project.org/dtc.git/info/refs

  fatal: HTTP request failed
  Clone of 'git://git.qemu-project.org/dtc.git' into submodule path 'dtc' failed
  Failed to recurse into submodule path 'riscv-qemu'

Can someone please provide a fix, or suggest an alternate method to clone the repository?

palmer-dabbelt, 2017-09-19:
This isn't reproducing when I clone the repo, but I do get an SSL error when going to that link you mentioned.

akhileshian, 2017-09-20:
Could you please suggest a fix, to rectify the SSL error?

palmer-dabbelt, 2017-09-20:
The SSL error is at QEMU's website, so I can't fix it.  Since I can't reproduce this I'm not going to file a but there, but you could try emailing someone (maybe "admin@qemu-project.org") to see if they can fix it.

Unless I can reproduce a bug it's very hard to fix it, sorry.

akhileshian, 2017-09-20:
Thanks for trying anyway :)

kito-cheng, 2017-09-21:
I guess it's because use git(ssh) protocol, it not work on some network environment (For example, I can't ssh to outside in my company). I suggest we can change to https or relative path just like other submodule.

akhileshian, 2017-09-21:
Kito, could you please give an example?
I have already added the following to my ~/.gitconfig file:
[url "https://"]
        insteadOf = git://
Is this the issue you are referring to? Because this is already taken care of, and I'm facing an issue ONLY for the 'qemu' sub-module.

Regards,
Akhilesh.

jhcloos, 2017-09-23:
Using https://git.qemu.org instead of https://git.qemu-project.org works.

Specifically, cloning https://git.qemu.org/git/dtc.git succeeds.

akhileshian, 2017-09-23:
@jhcloos could you please tell me how can I clone git.qemu instead of git.qemu-project? I changed the links in the .gitmodules files in the riscv-gnu-toolchain directory, but it is erroring out, as follows:

$ git submodule update --init --recursive
Submodule 'riscv-binutils-gdb' (https://github.com/riscv/riscv-binutils-gdb.git) registered for path 'riscv-binutils-gdb'
Submodule 'riscv-dejagnu' (https://github.com/riscv/riscv-dejagnu.git) registered for path 'riscv-dejagnu'
Submodule 'riscv-gcc' (https://github.com/riscv/riscv-gcc.git) registered for path 'riscv-gcc'
Submodule 'riscv-glibc' (https://github.com/riscv/riscv-glibc.git) registered for path 'riscv-glibc'
Submodule 'riscv-newlib' (https://github.com/riscv/riscv-newlib.git) registered for path 'riscv-newlib'
No submodule mapping found in .gitmodules for path 'riscv-qemu'

jim-wilson, 2018-01-25:
You can do this with a .gitconfig file change similar to the one you already have.

[url "https://git.qemu.org/git"]
     insteadOf = https://git.qemu-project.org

This allows "git clone https://git.qemu-project.org/dtc.git" to succeed.

closed 2018-01-25 by jim-wilson

SunYongshuai, 2018-03-09:
I did this also but still it's not working
[url "https://git.qemu.org/git"]
insteadOf = https://git.qemu-project.org

but I change to

[url "https://github.com/qemu/"]
insteadOf = https://git.qemu-project.org/

It works!
Because https://github.com/qemu/ does work, but others not.