riscv-collab/riscv-gnu-toolchain #284: riscv64-unknown-elf-ld: final link failed: Symbol needs debug section which does not exist

karimhasebou, 2017-10-08:
[link.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1367155/link.txt)

palmer-dabbelt, 2017-10-09:
Can you provide the entire stdout/stderr from the linker?  What you've posted probably isn't the first error message.  I suspect you need to add "-mcmodel=medany".

karimhasebou, 2017-10-09:
@palmer-dabbelt  tneo_rv32i_riscv64-unknown-elf-gcc.a(tn_arch_risc_rv32.o): In function `.L0 ':
(.text+0x1c): relocation truncated to fit: R_RISCV_JAL against `*UND*'
riscv64-unknown-elf-ld: final link failed: Symbol needs debug section which does not exist

karimhasebou, 2017-10-09:
@palmer-dabbelt I have tried the flag you provided but am still getting the same error.

evancox10, 2017-10-27:
I'm seeing this too. Version of ld is as follows;
./riscv32-unknown-elf/bin/ld --version
GNU ld (GNU Binutils) 2.29.51.20170726
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

palmer-dabbelt, 2017-10-28:
Can I have all those input files?  The error message is actually pretty generic, but at least I'll be able to tell if it's a bug or an invalid input.

evancox10, 2017-10-30:
I've attached the input assembly file, the linker script, and the dump of the unlinked .o file. For some reason the dump contains binary characters, so YMMV when viewing it.

To reproduce, run the following command:
riscv32-unknown-elf-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles --save-temps -march=rv32im -mabi=ilp32 -I. -I./macros/scalar -T./link.ld test.S -o test.elf

(file extensions have been changed to .txt to make github happy)

[link.ld.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1427904/link.ld.txt)
[test.o.dump.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1427905/test.o.dump.txt)
[test.S.txt](https://github.com/riscv/riscv-gnu-toolchain/files/1427906/test.S.txt)

lrusak, 2017-11-20:
I'm also getting something funk here compiling linux

```
kernel/kallsyms.o: In function `.L0 ':
kallsyms.c:(.text+0x12): relocation truncated to fit: R_RISCV_PCREL_HI20 against undefined symbol `kallsyms_offsets'
/home/lukas/libreelec/build.LibreELEC-HiFive.riscv64-9.0-devel/toolchain/bin/riscv64-libreelec-linux-gnu-ld: final link failed: Symbol needs debug section which does not exist
```

command is
```
/home/lukas/libreelec/build.LibreELEC-HiFive.riscv64-9.0-devel/toolchain/bin/riscv64-libreelec-linux-gnu-ld -melf64lriscv --build-id -o .tmp_vmlinux1 -T ./arch/riscv/kernel/vmlinux.lds --whole-archive built-in.o --no-whole-archive --start-group lib/lib.a arch/riscv/lib/lib.a --end-group
```

and linker version
```
./build.LibreELEC-HiFive.riscv64-9.0-devel/toolchain/bin/riscv64-libreelec-linux-gnu-ld -version
GNU ld (GNU Binutils) 2.29
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

full log http://sprunge.us/WVGW

palmer-dabbelt, 2017-11-20:
@lrusak the error message is actually super generic: it happens whenever the linker is unable to fill out a relocation.  That specific symbol is a bit of a special case: it's a weak undefined symbol, which is meant to have the value 0.  You're probably compiling Linux with a PAGE_OFFSET that doesn't result in the absolute address 0 being within a 2GB offset of the PC when referencing `kallsyms_offsets`, which is throwing your error.

We have a fix for this, which is on master and has been backported to our 2.29 branch (the one we do releases off of).  If you're trying to do a distro port, I'd recommend taking all the patches listed here

  https://github.com/riscv/riscv-binutils-gdb/commits/riscv-binutils-2.29

but the specific one that will fix this bug is 

  https://github.com/riscv/riscv-binutils-gdb/commit/1bfb4ecbd84c4a463d787e5501b416ff51479ac8

lrusak, 2017-11-20:
@palmer-dabbelt Thanks! that fixed it for me. I really appreciate the help. Cheers :beers:

ben-k, 2017-11-21:
FYI, I encountered this error as well, when trying to compile the RISC-V tests using a base address greater than 0x80000000:

```
/tmp/cc5U21AC.o:(.text.init+0x20): relocation truncated to fit: R_RISCV_PCREL_HI20 against undefined symbol `mtvec_handler'
($RISCV)/riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
```

Patching binutils and rebuilding the toolchain seems to fix it - thanks for the tip.

palmer-dabbelt, 2017-11-22:
@ben-k: that's probably a different instance of the same error message, like I said it's pretty generic.  You've probably got some other issues and this has just started happening to work because we now allow some low addresses to be referenced from medany mode.

I'll try and get some time to look at the original bug report.

jim-wilson, 2018-02-06:
The problems reported by @lrusack and @ben-k were already fixed.

The problem reported by @karimhasebou appears to be the same problem and hence is already fixed.

The problem reported by @evancox10 occurred because he is creating random instructions that include a jal with an odd address, and that is not a valid instruction.  This is something we could catch at assembly time, but since it is obscure, it isn't clear whether it is worth the trouble.  The same problem can be seen with branches if you use something like .+1 to get an odd address into a branch.

The interesting problem is the "symbol needs debug section" error.  This is more complicated to explain.  Bfd is a library, and goes to a lot of trouble to avoid reporting bugs directly, since the application may want to control when/where/how/etc errors are reported.  So bfd only reports errors indirectly.  It has two mechanisms to do this.  It can call a callback provided by the application.  Or it can set bfd_error (which works like errno) and return false back to the application.  The problem here is that the riscv port is doing both, and hence confusing the application into printing two errors when only one occurred.  When the first error is reported via a callback, the bfd library tries to print a better message by looking for debug info to get a function name and/or line number.  If you compile without debug info, it will try to open a separate debug info file, fail, and set bfd_error, but since it isn't an error to not have a separate debug info file in this case, that error gets ignored.  The riscv bfd then returns false, indicating that there is an error that hasn't been reported yet,  The linker then sees that bfd_error was set, and reports that as an error which is wrong.  I'm working on a fix to get rid of the second error which should be pretty simple,  just don't return false after calling an error callback function.

jim-wilson, 2018-02-07:
Committed a fix upstream.

closed 2018-02-07 by jim-wilson

leeehui, 2018-09-28:
Hi,
I met the same problem when I use ld version 2.30 with link address larger than 0x80000000

GNU ld (GNU Binutils) 2.30
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

the error message:
/riscv-gnu-toolchain_c3ad555__newlib__rv64gc_lp64/bin/../lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libg_nano.a(lib_a-malloc.o): In function `malloc':
malloc.c:(.text.malloc+0x0): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata._impure_ptr section in /riscv-gnu-toolchain_c3ad555__newlib__rv64gc_lp64/bin/../lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libg_nano.a(lib_a-impure.o)
/riscv-gnu-toolchain_c3ad555__newlib__rv64gc_lp64/bin/../lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
CMakeFiles/nvdla.elf.dir/build.make:546: recipe for target 'nvdla.elf' failed
make[2]: *** [nvdla.elf] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/nvdla.elf.dir/all' failed
make[1]: *** [CMakeFiles/nvdla.elf.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

jim-wilson, 2018-09-28:
@leeehui This doesn't appear related to any of the other problems reported here.  The debug section error is a cascading error which was fixed upstream 7-8 months ago.  This is fixed in binutils-2.31 which is in riscv-gnu-toolchain.  But this should always be ignored because it is a cascading error.

The real problem is the relocation truncated to fix.  This is a generic error, and there are a great many different ways to hit it.  A possible explanation here is that you forgot to compile libraries with -mcmodel=medany which is required when linking at 0x80000000.  You can configure a toolchain to build medany libraries by using --with-cmodel=medany when configuring.  Or it could be a mistake in your code or linker script.  If you need more help, file a new bug report, with enough info to reproduce.

leeehui, 2018-09-28:
@jim-wilson I didn't get the point " configure a toolchain to build medany libraries by using --with-cmodel=medany when configuring" from previous comments.  I just added CFLAGS: -mcmodel=medany and it did not work.
I will try it, thanks a lot!

jim-wilson, 2018-09-28:
Before you build riscv-gnu-toolchain, you have to configure it.  You can pass --with-cmodel=medany to configure to get a medany toolchain.  Passing -mcmodel=medany at compile time won't work, because none of the libraries (libgcc, newlib/glibc, libstdc++, etc) were compiled with -mcmodel=medany.  Malloc is coming from newlib, which is not medany code.

leeehui, 2018-09-28:
clear enough! appreciate your explanation!
I will try and feedback here.

leeehui, 2018-09-29:
Hi, 
it works well after I compile the tools with --with-cmodel=medany.

BTW, 
I wonder why the default configuration of --with-cmodel is medlow rather than medany?
Are there any special considerations here?

palmer-dabbelt, 2018-09-29:
It's a bit of a mixed bag: `-mcmodel=medany` allows users to link their code in more places, but `-mcmodel=medlow` is better tested and faster.  We default to `-mcmodel=medlow` because it's the right option when it's possible, but the newlib binaries I ship are always `-mcmodel=medany` for 64-bit systems.

There's really no right answer here, it depends on what your system looks like.