riscv-collab/riscv-gnu-toolchain #285: Using -march=rv32e givs an error of invalid string

abdallah93as, 2017-10-11:
Hello guys

Simple question?

How do I compile for rv32e extension. Is there something I should have configured before I built the toolchain??

jim-wilson, 2018-01-25:
The RV32E spec is not final yet.  There might be changes to the spec before it is finalized, and we want to avoid people using and relying on old incompatible versions of the spec.  Hence, there is no support in the release branches as yet.

But there are work in progress patches on the riscv-next branch in the riscv-binutils-gdb and riscv-gcc repositories.

closed 2018-01-25 by jim-wilson

abdallah93as, 2018-01-25:
Thank you very much for your answer altough its been a while since i asked. But it was still important for me to know why it dissapeared in later versions