riscv-collab/riscv-gnu-toolchain #291: bgeu converted to bltu + j instruction

evancox10, 2017-10-27:
I'm seeing the following `bgeu` instruction get assembled into two instructions in the final binary:

```
# test.S:
BGEU         x10, x15, -274
```

```
# test.dump:
 138: 00f56463        bltu     a0,a5,140 <_start+0x140>
 13c: db3ff06f        j        fffffeee <fromhost+0xfffffc2e>
```

Why is this happening? `bgeu` shouldn't be a pseudo op, so why am I getting two instructions? `bgeu` and `bltu` are both opcode `1100011`, but they have distinct funct3 codes, `111`, and `110`.

aswaterman, 2017-10-27:
If you branch more than 4KiB away, or if you branch to a label that's in a
different section or object file, the assembler uses a "far branch", which
consists of an inverted branch around a jump, which has a longer (1 MiB)
reach.

aswaterman, 2017-10-27:
Also, branching to constant offsets is not supported in the general case,
due to linker relaxations. You should always branch to a label unless you
can prove that the offset will not change at link time (which, given that
the offset is 274 bytes, seems improbable).

evancox10, 2017-10-28:
These are randomly generated assembly instructions, the idea of meaningful labels doesn't really apply. The intent is that each assembly instruction should be directly convertible to a machine code instruction in a 1-to-1 manner. For example, when I generate a branch, I make sure to only use 12 bits of immediate so that it stays within the 4KiB space. I specifically don't use any pseudo-ops. So I think for this case, I really am writing code where no offsets should change.

Can you explain more about what linker relaxation may be going on here? Why is this necessary in this case? Can it be turned off? I guess I can just do the assembly step (`as`) and not do the linker step (`ld`)? But somehow I have to get Spike to read it, the easiest way to do that is get it in an ELF, which requires linking (I guess?).

BTW there are applications where developers need to know with high confidence that they get exactly the instructions they specify in assembly. For example, constant-time cryptography.

evancox10, 2017-10-28:
Also, there is only the single assembly file. So, nothing really to link afaict.

evancox10, 2017-10-29:
Sorry, I missed this: "if you branch to a label that's in a
different section or object file, the assembler uses a far 'branch'". That's probably what happening in this case.

So, my other question stands, is there any way I can turn this off? If this isn't currently exposed in the tools, can you help point me to where I would need to look to change it?

palmer-dabbelt, 2017-10-30:
@evancox10 I wrote up a pair of blog entries that describe what's going on here

https://www.sifive.com/blog/2017/08/21/all-aboard-part-2-relocations/
https://www.sifive.com/blog/2017/08/28/all-aboard-part-3-linker-relaxation-in-riscv-toolchain/

On RISC-V systems you almost certainly won't be able to run a non-linked object file.  You can turn off linker relaxation, IIRC it's "-Wl,--no-relax", but you'll always get the two-instruction addressing sequences.

evancox10, 2017-10-31:
@palmer-dabbelt thanks for the links, I have been reading those and your other blog posts. They are very informative, thanks! I get the general idea (now) of leveraging the linker for relocations and relaxations, although the posts don't mention the specific relocation I'm running into: R_RISCV_JAL.

So is there really no way to turn off this conversion of branches to branch+jump? If not, can you point me to the approximate relevant section of the assembler code so that I can see about disabling it myself?

Also, I have to wonder why it is necessary for the assembler to modify the code, rather than just sticking the relocation symbol in and then letting the linker patch it as necessary? Even if I could get the unlinked object file to execute (I'm sure I could, just takes time), the code has already been munged before I even get to the linker. I understand why gcc itself would emit the longer sequence coming from C, but not why the assembler would modify things when a short sequence was explicitly used. It seems like the best thing to do in this case would be to let the user do what they asked for, and then fail at link stage if it doesn't work. Or provide some option for this to happen.

Tangentially related, I would again like to stress the importance in having a way for an assembly programmer to get exactly the instructions they specify. For example, here's how ARM allows [instruction width selection in Thumb](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0473m/dom1359731171400.html). I see the issue here as a case of instruction width selection, although it's two instructions vs one, rather than 32-bit vs 16-bit. (Although the [compressed vs non-compressed selection issue](https://github.com/riscv/riscv-toolchain-conventions/issues/5) has also been raised.)

evancox10, 2017-11-07:
Hi @palmer-dabbelt ,
I don't mean to harass y'all about what is an admittedly corner case scenario here, but wanted to see if we could come to a resolution on this. I've been poking through the gas and ld code, trying to see what it would take to add an option for the assembler to not modify branches. Is [this](https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.29/gas/config/tc-riscv.c#L2464) the code that does the conversion? Looks like it, but I can't find any code that calls this function. (Besides `md_convert_frag()`, which is just a pass-through.)

Maybe there is a simpler hack I can do, like defining a filler section or an absolute symbol? Not sure.

Thanks,
Evan

aswaterman, 2017-11-07:
Yes, and also https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.29/gas/config/tc-riscv.c#L694 and https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.29/gas/config/tc-riscv.c#L357 -- there is a lot of machinery involved

jim-wilson, 2017-11-07:
You want to call add_fixed_insn instead of add_relaxed_insn in append_insn.  This will break compiler generated code and anything that gets relaxed by the linker.  The easy way to do this is to disable the special handling for BFD_RELOC_12_PCREL and BFD_RELOC_RISCV_JMP here.  With that change, in the objdump -d output, I see a bleu which is an alias for bgeu.  That looks like a minor bug in the disassembler, it should prefer the insn, not the alias.

closed 2021-03-08 by lazyparser