riscv-collab/riscv-gnu-toolchain #298: balign asm statement broken

cliffordwolf, 2017-11-18:
Consider the following assembler code:

```
j foobar
.ascii "xxxxx"
.byte 0x00
.balign 4
foobar:
nop
```

An older version of `riscv32-unknown-elf-as` produces the expected result (output of `riscv32-unknown-elf-objdump -d`):

```
00000000 <foobar-0xc>:
   0:	00c0006f          	j	c <foobar>
   4:	7878                	flw	fa4,116(s0)
   6:	7878                	flw	fa4,116(s0)
   8:	0078                	addi	a4,sp,12
	...

0000000c <foobar>:
   c:	00000013          	nop
```

But with the current version of riscv-gnu-toolchain (bf5697a) I get the following result instead:

```
00000000 <foobar-0xa>:
   0:	00a0006f          	j	a <foobar>
   4:	7878                	flw	fa4,116(s0)
   6:	7878                	flw	fa4,116(s0)
   8:	0078                	addi	a4,sp,12

0000000a <foobar>:
   a:	00000013          	nop
	...
```

The symbol foobar is not aligned to a multiple of 4. It looks like the `.balign` statement was simply ignored.

In both cases I just ran `riscv32-unknown-elf-as test.s` and then `riscv32-unknown-elf-objdump -d a.out`.

palmer-dabbelt, 2017-11-18:
`as` produces object files, not executables.  RISC-V handles alignment via linker relaxation, so object files won't have the expected alignment.  If you 'objdump -dr a.out' you should see a `R_RISCV_ALGIN` relocation.

````
$ riscv64-unknown-elf-objdump -dr test.o                                                                                                                                                                                                                                                                                                                                                                                                                                                                              test.o:     file format elf64-littleriscv                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  Disassembly of section .text:

0000000000000000 <foobar-0xa>:
   0:   00a0006f                j       a <foobar>
                        0: R_RISCV_JAL  foobar
   4:   7878                    ld      a4,240(s0)
   6:   7878                    ld      a4,240(s0)
   8:   0078                    addi    a4,sp,12

000000000000000a <foobar>:
   a:   00000013                nop
                        a: R_RISCV_ALIGN        *ABS*
        ...
````

You should see the correct alignment in an executable (ie, produced by the linker).  I haven't gotten to the specific case of the `R_RISCV_ALIGN`, but there's a bit of blog content on some other relocations and on linker relaxation

  https://www.sifive.com/blog/2017/08/21/all-aboard-part-2-relocations/
  https://www.sifive.com/blog/2017/08/28/all-aboard-part-3-linker-relaxation-in-riscv-toolchain/

wallento, 2017-11-18:
Hi @palmer-dabbelt, the bug report is based on this issue: https://github.com/cliffordwolf/picorv32/issues/50

The linker ultimately gives the error `6 bytes required for alignment to 8 bytes boundary`.

cliffordwolf, 2017-11-18:
Here is the complete error I get in the original test case:

```
/opt/riscv32i/bin/riscv32-unknown-elf-gcc -Os -ffreestanding -nostdlib -o firmware/firmware.elf \
	-Wl,-Bstatic,-T,firmware/sections.lds,-Map,firmware/firmware.map,--strip-debug \
	firmware/start.o firmware/irq.o firmware/print.o firmware/sieve.o firmware/multest.o firmware/stats.o tests/ori.o tests/xor.o tests/or.o tests/andi.o tests/lui.o tests/srai.o tests/sra.o tests/blt.o tests/bge.o tests/sub.o tests/mul.o tests/lw.o tests/lb.o tests/sw.o tests/bgeu.o tests/div.o tests/beq.o tests/auipc.o tests/rem.o tests/jalr.o tests/sll.o tests/mulhu.o tests/mulh.o tests/divu.o tests/add.o tests/slt.o tests/lbu.o tests/lhu.o tests/sh.o tests/simple.o tests/sb.o tests/bltu.o tests/slli.o tests/remu.o tests/xori.o tests/lh.o tests/srli.o tests/jal.o tests/addi.o tests/and.o tests/srl.o tests/j.o tests/bne.o tests/mulhsu.o tests/slti.o -lgcc
/opt/riscv32i/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/bin/ld: tests/auipc.o(.text+0x32): 6 bytes required for alignmentto 8-byte boundary, but only 4 present
/opt/riscv32i/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/bin/ld: can't relax section: Bad value
collect2: error: ld returned 1 exit status
```

@palmer-dabbelt  you are right, I can link the object file I get in the case of the above simple assembler input. I'll try to create a stand-alone test case that reproduces the linker error.

cliffordwolf, 2017-11-18:
Also: Maybe we can fix the "alignmentto" typo? :)
https://twitter.com/oe1cxw/status/929506482332119041

cliffordwolf, 2017-11-18:
Try this test case:

```
.text
.global _start
_start:
nop

.prname_next: 
nop
j .prname_done
j .prname_next

.test_name: 
.ascii "auipc"
.byte 0x00
.balign 4
.prname_done: 
nop

test_2:
.align 3
```

With current riscv-gnu-toolchain:

```
$ riscv32-unknown-elf-gcc -march=rv32i -Os -ffreestanding -nostdlib -o test.elf test.s
/opt/riscv32i/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccdQLHYQ.o(.text+0x1a): 6 bytes required for alignmentto 8-byte boundary, but only 4 present
/opt/riscv32i/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/bin/ld: can't relax section: Bad value
collect2: error: ld returned 1 exit status
```

With an older version:

```
$ riscv32-unknown-elf-gcc -march=rv32i -Os -ffreestanding -nostdlib -o test.elf test.s
$ riscv32-unknown-elf-objdump -d test.elf

test.elf:     file format elf32-littleriscv

Disassembly of section .text:

00010058 <_start>:
   10058:	00000013          	nop

0001005c <.prname_next>:
   1005c:	00000013          	nop
   10060:	0100006f          	j	10070 <.prname_done>
   10064:	ff9ff06f          	j	1005c <.prname_next>

00010068 <.test_name>:
   10068:	7561                	lui	a0,0xffff8
   1006a:	7069                	0x7069
   1006c:	00000063          	beqz	zero,1006c <.test_name+0x4>

00010070 <.prname_done>:
   10070:	00000013          	nop

00010074 <test_2>:
   10074:	00000013          	nop
```

jim-wilson, 2017-11-19:
On Sat, Nov 18, 2017 at 10:54 AM, Palmer Dabbelt <notifications@github.com>
wrote:

> as produces object files, not executables. RISC-V handles alignment via
> linker relaxation, so object files won't have the expected alignment. If
> you 'objdump -dr a.out' you should see a R_RISCV_ALGIN relocation.
>
> 000000000000000a <foobar>:
>    a:   00000013                nop
>                         a: R_RISCV_ALIGN        *ABS*
>         ...
>
> You should see the correct alignment in an executable (ie, produced by the
> linker). I haven't gotten to the specific case of the R_RISCV_ALIGN, but
> there's a bit of blog content on some other relocations and on linker
> relaxation
>

Except the reloc appears to be in the wrong place, after foobar when it
should be before.  The assembler appears to be confused here, but it is
confused because we are in norvc mode, and we have 2-byte alignment, which
should never happen.  One could call this programmer error.  Or perhaps we
can do something in the assembler to make it work.

Jim

jim-wilson, 2017-11-19:
On Sat, Nov 18, 2017 at 10:25 AM, Clifford Wolf <notifications@github.com>
wrote:

> Consider the following assembler code:
>
> j foobar
> .ascii "xxxxx"
> .byte 0x00
> .balign 4
> foobar:
> nop
>
>
There are some conflicting goals here.  We want to relax code at link time,
deleting unnecessary instructions, but this means that we can never know
alignment at assembly time.  Also, we have some machines that support
2-byte instructions and some that don't.  It is never safe to emit a 2-byte
nop on a target that only supports 4 byte instructions.  So the assembler
gets confused here by your code.

One way to fix this is to explicitly indicate that you are in rvc mode.
         j foobar
        .option rvc
        .ascii "xxxxx"
        .byte 0x00
        .balign 4
        .option norvc
foobar:
        nop
This produces the correct result.

Another way to fix the problem is to turn off relaxation.
.option norelax
j foobar
.ascii "xxxxx"
.byte 0x00
.balign 4
foobar:
nop
This also produces the correct result.

I don't know offhand if there is a way to make your original example work,
without breaking linker relaxation and/or rvc/norvc support.  This would
require some investigation which would take some time.

Jim

jim-wilson, 2017-11-19:
On Sat, Nov 18, 2017 at 12:35 PM, Clifford Wolf <notifications@github.com>
wrote:
> $ riscv32-unknown-elf-gcc -march=rv32i -Os -ffreestanding -nostdlib -o test.elf test.s
> /opt/riscv32i/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/bin/ld: [...]
This is the linker trying to tell you that you have a problem, because you
specifically requested no compressed instructions, and the only way we can
align this location in the text section is by emitting a compressed nop.  I
think using .option rvc/norvc around the code is the right solution here,
and remember to put the .balign 4 before the .option norvc, because we
can't emit a 2-byte nop where only 4-byte instructions are allowed, so the
alignment has to come before you exit rvc mode.

The alignmentto typo is easy to fix.  I've checked in a patch to the FSF
Binutils source tree.
    https://sourceware.org/ml/binutils-cvs/2017-11/msg00167.html
This will eventually get pulled into the riscv-gnu-toolchain tree.

Jim

cliffordwolf, 2017-11-19:
I thought the default filling byte for `.balign` is zero bytes, not nops. But apparently this is implementation defined in `.text` sections. From `as.info`:

> The second expression (also absolute) gives the fill value to be
stored in the padding bytes.  It (and the comma) may be omitted.  If it
is omitted, the padding bytes are normally zero.  However, on some
systems, if the section is marked as containing code and the fill value
is omitted, the space is filled with no-op instructions.

Is stuff like this (risc-v specific implementation defined behavior) documented anywhere?

Explicitly setting the filling byte to zero `.balign 4, 0` solves the issue with my code.

closed 2017-11-19 by cliffordwolf

jim-wilson, 2017-11-19:
On Sat, Nov 18, 2017 at 5:52 PM, Clifford Wolf <notifications@github.com>
wrote:

> I thought the default filling byte for .balign is zero bytes, not nops.
> But apparently this is implementation defined in .text sections. From
> as.info:
>
> The second expression (also absolute) gives [...]

Emitting alignment directives to align branch target labels for better
performance is a common compiler optimization.  At least it is nowadays.
So all targets with compiler support must emit nops for an alignment
directive in the text section.  Though of course for some targets, a 0 fill
is a nop and they don't have to do anything special.  Hence I would say
this is out-of-date docs that needs to be updated, not a risc-v specific
assembler feature.  The "some" should be changed to "most".

There are target dependent sections in the gas manual, but they normally
only list target dependent options and pseudo-ops.  The risc-v one only
mentions options.  It needs to be expanded to mention the .option pseudo-op
and could probably also mention a few other things.  That will have to be a
long term goal to improve the docs.  Getting things working and properly
merged upstream is the immediate concern.

I think the only target doc that mentions nops for alignment is the i386
one, and that is because they have an option to choose whether you emit for
instance a single 7-byte nop instruction or 7 1-byte nop instructions.  The
default is to emit a single multi-byte nop for every length from 1 to 14.
Though trying it, I see that it doesn't work anymore, which is no surprise,
since there is no good reason to ever use this option.

https://sourceware.org/binutils/docs-2.29/as/RISC_002dV_002dDependent.html#RISC_002dV_002dDependent

> Explicitly setting the filling byte to zero .balign 4, 0 solves the issue
> with my code.
>
That didn't occur to me, yes that works too.

Jim

cliffordwolf, 2017-11-19:
> Emitting alignment directives to align branch target labels for better
performance is a common compiler optimization.

I thought that's what `.align` is for. But now I see that `as.info` has the same paragraph for `.align`. 

Clearly the problem was that I had incorrect assumptions about the differences between the two directives. I always used something like `.align 2` to pad with nops to a multiple of 4 and `.balign 4` to pad with whatever to a multiple of 4. But now I see that it's not even standardized if the parameter to `.align` specifies the alignment in zero bits in the next address or in bytes..

So in the future I will now use `.balign 4` and `.balign 4, 0` for the two use cases. Thanks for helping me understand what went wrong here.

> The risc-v one only
mentions options.  It needs to be expanded to mention the .option pseudo-op
and could probably also mention a few other things.  That will have to be a
long term goal to improve the docs.

Yes please. :) I did not know about `.option [no]relax` and `.option [no]rvc` until you mentioned it.

jim-wilson, 2017-11-19:
On Sat, Nov 18, 2017 at 7:55 PM, Clifford Wolf <notifications@github.com>
wrote:
> Clearly the problem was that I had incorrect assumptions about the
> differences between the two directives. I always used something like .align
> 2 to pad with nops to a multiple of 4 and .balign 4 to [...]

Yes, the problem with .align is a historical one, in that different vendor
assemblers had different conventions when gas was developed long ago.  That
is why we have .balign and .p2align in gas, which are ambiguous about what
the alignment argument means.  Otherwise, align, balign, and p2align all
behave the same.

Jim