riscv-collab/riscv-gnu-toolchain #300: make linux error

Jyun-Neng, 2017-12-12:
While running make linux, it shows this error:
> checking for suffix of object files... configure: error: in /opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc:
configure: error: cannot compute suffix of object files: cannot compile
See config.log for more details.
Makefile:11138: recipe for target configure-target-libgcc failed
make[1]: *** [configure-target-libgcc] Error 1
make[1]: Leaving directory /opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1
Makefile:204: recipe for target stamps/build-gcc-linux-stage1 failed
make: *** [stamps/build-gcc-linux-stage1] Error 

Does anyone have this problem?

palmer-dabbelt, 2017-12-12:
You need to look in `config.log` (in `/opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc`) to see the real error message.  "cannot compute the suffix of object files: cannot compile" is a pretty generic error message.

Jyun-Neng, 2017-12-13:
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by GNU C Runtime Library configure 1.0, which was
generated by GNU Autoconf 2.64.  Invocation command line was

  $ /opt/riscv/riscv-tools/riscv-gnu-toolchain/riscv-gcc/libgcc/configure --srcdir=/opt/riscv/riscv-tools/riscv-gnu-toolchain/riscv-gcc/libgcc --cache-file=./config.cache --with-cross-host=x86_64-pc-linux-gnu --prefix=/opt/riscv/riscv --with-sysroot=/opt/riscv/riscv/sysroot --with-newlib --without-headers --disable-shared --disable-threads --with-system-zlib --enable-tls --disable-libatomic --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-bootstrap --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --enable-languages=c,lto --program-transform-name=s&^&riscv64-unknown-linux-gnu-& --disable-option-checking --with-target-subdir=riscv64-unknown-linux-gnu --build=x86_64-pc-linux-gnu --host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu

## --------- ##
## Platform. ##
## --------- ##

hostname = jnji-BM6660-BM6360
uname -m = x86_64
uname -r = 4.10.0-40-generic
uname -s = Linux
uname -v = #44~16.04.1-Ubuntu SMP Thu Nov 9 15:37:44 UTC 2017

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /opt/riscv/riscv/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /snap/bin
PATH: /opt/riscv/riscv/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2033: creating cache ./config.cache
configure:2239: checking build system type
configure:2253: result: x86_64-pc-linux-gnu
configure:2273: checking host system type
configure:2286: result: riscv64-unknown-linux-gnu
configure:2389: checking for --enable-version-specific-runtime-libs
configure:2402: result: no
configure:2450: checking for a BSD-compatible install
configure:2518: result: /usr/bin/install -c
configure:2534: checking for gawk
configure:2561: result: /usr/bin/gawk
configure:2661: checking for riscv64-unknown-linux-gnu-ar
configure:2688: result: riscv64-unknown-linux-gnu-ar
configure:2753: checking for riscv64-unknown-linux-gnu-lipo
configure:2780: result: riscv64-unknown-linux-gnu-lipo
configure:2845: checking for riscv64-unknown-linux-gnu-nm
configure:2872: result: /opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/nm
configure:2937: checking for riscv64-unknown-linux-gnu-ranlib
configure:2964: result: riscv64-unknown-linux-gnu-ranlib
configure:3029: checking for riscv64-unknown-linux-gnu-strip
configure:3056: result: riscv64-unknown-linux-gnu-strip
configure:3118: checking whether ln -s works
configure:3122: result: yes
configure:3139: checking for riscv64-unknown-linux-gnu-gcc
configure:3166: result:  /opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv/riscv64-unknown-linux-gnu/bin/ -B/opt/riscv/riscv/riscv64-unknown-linux-gnu/lib/ -isystem /opt/riscv/riscv/riscv64-unknown-linux-gnu/include -isystem /opt/riscv/riscv/riscv64-unknown-linux-gnu/sys-include   
configure:3435: checking for C compiler version
configure:3444:  /opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv/riscv64-unknown-linux-gnu/bin/ -B/opt/riscv/riscv/riscv64-unknown-linux-gnu/lib/ -isystem /opt/riscv/riscv/riscv64-unknown-linux-gnu/include -isystem /opt/riscv/riscv/riscv64-unknown-linux-gnu/sys-include    --version >&5
[... 185 lines cut ...]
target_noncanonical='riscv64-unknown-linux-gnu'
target_subdir='riscv64-unknown-linux-gnu'
thread_header=''
tm_defines=''
tm_file=''
tmake_file=''
toolexecdir='$(exec_prefix)/$(target_noncanonical)'
toolexeclibdir='$(toolexecdir)/lib'
unwind_header=''
vis_hide=''
with_aix_soname='aix'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "GNU C Runtime Library"
#define PACKAGE_TARNAME "libgcc"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "GNU C Runtime Library 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL "http://www.gnu.org/software/libgcc/"

configure: exit 1

Jyun-Neng, 2017-12-13:
Above is the log file.
Before building riscv64-unknown-linux-gnu-gcc, every step I have done and succeed.
But the step
`make linux `
has problem.
I have no idea if I lost something.

jim-wilson, 2017-12-13:
The interesting bit of the config.log file is here
configure:3662: checking for suffix of object files
configure:3684: /opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv/riscv64-unknown-linux-gnu/bin/ -B/opt/riscv/riscv/riscv64-unknown-linux-gnu/lib/ -isystem /opt/riscv/riscv/riscv64-unknown-linux-gnu/include -isystem /opt/riscv/riscv/riscv64-unknown-linux-gnu/sys-include -c -g -O2 conftest.c >&5
/opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/as: 106: exec: -march=rv64imafdc: not found

So there is something wrong with the /opt/riscv/riscv-tools/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/as file.  It should have a line setting ORIGINAL_AS_FOR_TARGET to the assembler that was already built and installed in $RISCV/riscv64-unknown-linux-gnu/bin/as.  You should also see this ORIGINAL_AS_FOR_TARGET setting in the build-gcc-linux-stage1/gcc/config.log file as it was computed by the configure script.

closed 2018-01-25 by jim-wilson