riscv-collab/riscv-gnu-toolchain #303: assembler does not accept accesses to user mode ustatus CSR

LeeMooreImperas, 2017-12-19:
Hi

I am a bit confiused, I tried the following assembler code
`csrr t0, ustatus`
This resulted in the error message 
`test.S:414: Error: Instruction csrr requires absolute expression`

Upon further investigation, it appears multiple registers yield the same error message, including
ustatus, uie, utvec, uepc, stval, satp, mtval

Please advise
Thx
Lee

jim-wilson, 2017-12-19:
If you use a register name that gas doesn't know, then it assumes that it is a variable name.  The csrr instruction checks for a register name first, and then an immediate operand as index number.  A variable name can't be used as an immediate operand, because it isn't an absolute expression.  This is why a lot of assemblers require that register names be prefixed with a special character, to avoid ambiguity between register names and variable names.  Requiring that symbol names be prefixed with _ helps, but only solves part of the problem, since a missing _ is still a valid symbol name, and may result in a confusing assembler error, or a linker error for a reference to an undefined symbol.

Anyways, if you have access to binutils sources, you can see the list of currently supported registers in the file include/opcode/riscv-opc.h, in the DECLARE_CSR macro calls.

This is maybe an issue of the assembler not being updated to the latest version of the priv spec yet.  Or maybe this part of the assembler was never finished.  I'll add it to my list of things that need to be fixed.

LeeMooreImperas, 2017-12-20:
Hi Jim
`
This is maybe an issue of the assembler not being updated to the latest version of the priv spec yet. Or maybe this part of the assembler was never finished. I'll add it to my list of things that need to be fixed.`

This is based upon the spec as per 1.10 priv - May/7/2017
Unfortunately there is no branch/tag under github matching this release, so it is not possible to be able to definitively checkout the tools/data/simulator which can be used to match the spec.

So best guess was to use the document date of May/7/2017 - but gas does not recognise this register, understand the latest code has now moved on from Priv ISA 1.10, and so is part of the Draft for the next release.

Many Thx
Lee

jim-wilson, 2017-12-28:
I added a patch upstream to support all priv spec 1.9.1 and 1.10 registers.
https://sourceware.org/ml/binutils/2017-12/msg00184.html
https://sourceware.org/ml/binutils-cvs/2017-12/msg00156.html

closed 2017-12-28 by jim-wilson