riscv-collab/riscv-gnu-toolchain #306: gp relaxation can cause incorrect line number for undefined symbol to be reported

TM1234, 2017-12-20:
When gp relaxation is enabled (which seems to be the default?) then an undefined symbol error can report the wrong source line number for the error. This is not the case when relaxation is disabled using -Wl,--no-relax. Is there any way for the linker to be improved to report the correct source line number when relaxation is on?

A simple example:

```
1    // rvtest.c 
2
3    static void foo(int x)
4    {
5        x++;
6    }
7
8    int main(int argc, char *argv[])
9    {
10       int i = 0;
11   
12       foo(i);
13       foo(i);
14       foo(i);
15       foo(i);
16       
17       bar(i);
18    
19       return 0;
20   }
```

The compilation warning gives the correct line number - i.e. line 17:

```
Tommy.Murphy@DUBLTC0016 D:\sandbox
> riscv64-unknown-elf-gcc -march=rv32im -mabi=ilp32 -g3 -O0 -c -Wall rvtest.c
rvtest.c: In function 'main':
rvtest.c:17:5: warning: implicit declaration of function 'bar' [-Wimplicit-function-declaration]
     bar(i);
     ^~~
```

But the link time error does not - it says line 15 instead of 17:

```
Tommy.Murphy@DUBLTC0016 D:\sandbox                                                                
> riscv64-unknown-elf-gcc -march=rv32im -mabi=ilp32 -g3 -O0 -Wall rvtest.c                        
rvtest.c: In function 'main':                                                                     
rvtest.c:17:5: warning: implicit declaration of function 'bar' [-Wimplicit-function-declaration]  
     bar(i);                                                                                      
     ^~~                                                                                          
C:\Users\TOMMY~1.MUR\AppData\Local\Temp\cc0wV5mk.o: In function `main':                           
D:\sandbox/rvtest.c:15: undefined reference to `bar'                                              
collect2.exe: error: ld returned 1 exit status  
```

Unless I disable relaxation using -Wl,--no-relax in which case both the compilation warning and the link error report the correct line number 17:

```
Tommy.Murphy@DUBLTC0016 D:\sandbox
> riscv64-unknown-elf-gcc -march=rv32im -mabi=ilp32 -g3 -O0 -Wall -Wl,--no-relax rvtest.c
rvtest.c: In function 'main':
rvtest.c:17:5: warning: implicit declaration of function 'bar' [-Wimplicit-function-declaration]
     bar(i);
     ^~~
C:\Users\TOMMY~1.MUR\AppData\Local\Temp\ccFtRmGM.o: In function `main':
D:\sandbox/rvtest.c:17: undefined reference to `bar'
collect2.exe: error: ld returned 1 exit status
```

jim-wilson, 2017-12-20:
The linker is using debug info to map addresses to line numbers.  Because relaxation changes code size, we have relocations in the debug info.  Linking with --noinhibit-exec, I see that the debug info looks correct at the end.  So the problem here is that the linker is emitting the error after performing relaxation, but before resolving all relocations, which means the line number info doesn't match the relaxed code, unless you manually interpret relocations against the line number table.  The linker isn't doing that.  Normal targets don't have relocations in the line number table, as normal targets don't change code size when relaxing, so I don't know offhand if there is an easy way to fix this.  I can put this on my to do list, but don't know when I will have a chance to look at it.

TM1234, 2017-12-20:
OK - thanks Jim.
I suspected that it might not be a simple one to fix.

closed 2021-03-08 by lazyparser

TommyMurphyTM1234, 2021-03-08:
@lazyparser can you clarify why this was closed and what the resolution, if any, was please? Thanks.

lazyparser, 2021-03-08:
> @lazyparser can you clarify why this was closed and what the resolution, if any, was please? Thanks.

Oh sorry if I closed it wrong, I was closing issues that not actively discussed or replied more than one year. Feel free to reopen it if it is still relevant.

TommyMurphyTM1234, 2021-03-08:
Thanks for the quick update @lazyparser . That's fine - I just wondered why it was closed. It probably makes sense to close it given the lack of recent activity and @jim-wilson's earlier explanation. I'll leave it closed so. Thanks again.