riscv-collab/riscv-gnu-toolchain #308: Hint instructions not decoded by riscv64-unknown-elf-objdump

cliffordwolf, 2017-12-21:
Consider this simple assembler file (`hint.s`):

```
.hword 0x0002  # C.SLLI64 rd=0 HINT
.hword 0x807e  # C.MV rd=0 rs2=31 HINT
```

The comments in above code are correct, as `spike-dasm` demonstrates:

```
$ { echo 'DASM(0x0002)'; echo 'DASM(0x807e)'; } | spike-dasm 
slli    zero, 0
mv      zero, t6
```

However, `riscv64-unknown-elf-objdump -d` does decode those hint instructions as illegal (or unknown) instructions:

```
$ riscv64-unknown-elf-as hint.s
$ riscv64-unknown-elf-objdump -d a.out

a.out:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <.text>:
   0:	0002                	0x2
   2:	807e                	0x807e
```

This makes it very confusing to read for example assembler programs that are generated by SMT solvers during formal verification of hardware.

Can we maybe have a `-M hints` switch to objdump that will make it decode valid hint instructions? That would make it so much easier to distinguish between hint instructions and illegal instructions when reading those solver-generated programs.

palmer-dabbelt, 2017-12-21:
I think we're one step ahead of you: Jim just posted a patch <https://sourceware.org/ml/binutils/2017-12/msg00168.html>.  It'll filter into the riscv-binutils-2.29 branch (and then master here) after it's been reviewed.  I think this should fix the disassembly as well as the assembly -- it won't actually say "hint" anywhere, it just treats them as valid instructions.  We could add hint aliases (like our nop alias).

jim-wilson, 2017-12-21:
c.slli64 is an RV128 instruction, and we have no RV128 support as yet, so I haven't tried to implement that one yet.  Likewise c.srli64.  I think we need a plan for how to handle rv128 support before we can do anything sensible with them.  I wanted a chance to think about that first.

The others should already be working.  I saw no special syntax defined for hint instructions, so I just handle them like regular instructions.  With current upstream binutils sources, I get

00000000 <.text>:
   0:	0002                	c.slli	zero,0x0
   2:	807e                	c.mv	zero,t6

That disassembly of the first insn as c.slli is a bug.  I added support for an x0 target, but didn't change the immediate field.  Looks like this was broken before my fix, but we didn't notice.  Looks like we never tested disassembly of illegal instructions.  If I change the testcase to
	.hword 0x0102  
	.hword 0x807e  # C.MV rd=0 rs2=31 HINT
then with a riscv-binutils-gdb build, I get

0000000000000000 <.text>:
   0:	0102                	slli	sp,sp,0x0
   2:	807e                	0x807e

which shows that this bug exists before my patch.  I will have to work on a fix for that.

See https://github.com/riscv/riscv-binutils-gdb/issues/123 for the original bug report.

cliffordwolf, 2017-12-21:
> c.slli64 is an RV128 instruction, and we have no RV128 support as yet

According to Table 12.6 of V2.2 of the User-Level ISA spec, C.SLLI64 is a hint instruction with any rs1/rd value on RV32/RV64, and a hint instruction on RV128 when rs1/rd=0:

[image]

jim-wilson, 2017-12-21:
I am aware that c.slli64 is a hint for RV32/RV64.  My point is that it is a valid instruction for RV128.  If we add support for it now, and we don't have RV128 support, then it is possible that we could make a mistake and not realize it.  Then later, when we add RV128 support, we may need an inconvenient compatibility break to fix it.  I think it makes sense to add RV128 support at the same time to avoid that potential problem.  But that is a bigger project than just adding the other hints, so I plan to tackle it separately.

jim-wilson, 2018-05-02:
riscv-gnu-toolchain has old obsolete versions of everything.  The official binutils sources are up-to-date and can be found at
  http://sourceware.org/binutils
I would suggest using them instead of the binutils in riscv-gnu-toolchain.  You can use them inside riscv-gnu-toolchain if you want.  Inside the riscv-binutils-gdb dir, you can do something like
  git remote add upstream git://sourceware.org/git/binutils-gdb.git
  git fetch upstream
  git checkout -b upstream-master --track upstream/master

We are aware that riscv-gnu-toolchain badly needs updating, but there are a number of difficult time consuming problems that need to be fixed first, and we haven't had time to deal with them yet.  The 32-bit glibc support hasn't been upstreamed yet, so updating glibc breaks all 32-bit linux support.  Updating qemu without a glibc update breaks glibc, as there were stat struct changes.  Updating binutils breaks gdb as gdb support is not fully upstream yet.  riscv-gnu-toolchain is in turn a submodule in about half a dozen other git repos, so updating riscv-gnu-toolchain means all of the other git repos need to be tested and fixed or they will end up broken.  And there are probably other problems I can't remember at the moment.  This work hasn't happened yet because we haven't had time to do it yet.

Oh, and now that gcc-8.0 has been released, we have the additional problem that riscv-gnu-toolchain can't easily track both gcc-7.3 and gcc-8.0 at the same time.  We will have to make a choice, and which ever choice we make, probably about half the users are going to be unhappy.  So someone will have to deal with that problem.  Some of these problems go away if people start building their own toolchains, from the tools versions they want, instead of assuming that riscv-gnu-toolchain is a one stop solution for everyone.

jim-wilson, 2018-05-02:
I don't believe that asking for time to fix a problem correctly is a BS argument.  I care about the quality of the work I do, and I won't apologize for that.  I also won't apologize for the fact that I'm so overworked that I haven't had time to address every problem that has been reported against binutils and gcc.  I have a list of issues and I'm trying to prioritize them and address the most important ones first.  This issue with RV128 hint instructions hasn't made it to the top of my list yet.

cliffordwolf, 2018-05-02:
Thanks for pointing me to the upstream repository. I'll check it out, but as I understand your 2nd post, this issue is not fixed there either. So I'm not quite sure what the value of updating to upstream would be in this context.

Re your 2nd post: I explicitly said that I understand that something like that can take time.

> I can understand that sometimes it takes time to fix a bug.

You have not asked for time for fixing the bug.

You have argued that it is better to leave it the way it is indefinitely (until RV128 support is added).

You have argued that this is not a bug but a missing feature.

And that *is* a first class BS argument.

And I'm not asking for an apology. I'm not even asking you to fix it right away! I'm just asking you to treat that as the bug it is and not serve my this BS about it being just a missing feature that might be added many years down the road when RV128 is added.

>  This issue with RV128 hint instructions hasn't made it to the top of my list yet.

This is not a "RV128 hint instruction". As far as RV32/RV64 is concerned this is just a valid instruction.

Insisting that it is okay to not decode this instruction correctly because it happens to have a different meaning in RV128 is *exactly* the BS argument I am talking about! And you are doubling down on it!

Do you also not decode C.FLDSP because that happens to be C.LQSP in RV128?

This is not an RV128 issue. This is an RV32/RV64 issue. Any claim to the contrary is BS.

I'm okay with it taking time. I have said that very explicitly. But do not try to feed me that BS about this not being a bug but a missing feature, or that this is a RV128 issue, not a RV32/RV64 issue. Because that's just not true!

This bug has repeatedly wasted hours of my life! But I can deal with that! What I can't deal with is being fed this BS about this not being a bug! That this is not an issue for RV32/RV64 because it's "RV128 hint instruction", even though it is just a regular instruction with NOP semantics on RV32/RV64. That's the BS argument! And it clearly is one.

aswaterman, 2018-05-02:
You’re being borderline abusive. Please chill out.

cliffordwolf, 2018-05-02:
I am sorry, you are right. I am obviously very frustrated with this issue, and being told to wait until RV128 is supported in binutils (which might be years away as far as I know), just to get correct decoding for RV32/RV64 is obviously not the kind of response I was expecting.

aswaterman, 2018-05-02:
@jim-wilson I'm happy to write a test case for all of the RVC HINTs and a patch for any that aren't working.

aswaterman, 2018-05-02:
Against the latest FSF binutils,that is.  I don't know what to do about riscv-gnu-toolchain.

cliffordwolf, 2018-05-02:
That would be great! However, my "freakout" was because, as I understand it, this is not wanted:

> If we add support for it now, and we don't have RV128 support, then it is possible that we could make a mistake and not realize it.

The argument, that I called BS, is that we should preserve the status quo and add RV128 support first, and only after that fix the hint instructions:

>  I think it makes sense to add RV128 support at the same time to avoid that potential problem.

jim-wilson, 2018-05-02:
I believe that I already have all hints supported and with testcases, except for the RV128 instructions that are RV32/RV64 hints, which I wanted to take a second look at because we have no RV128 support yet.  I then got distracted by lots of other problems and never got back to this.

By the way, I take bugs reported against FSF binutils/gcc more seriously than bugs reported here, as lots of the stuff reported here aren't actual bugs, and more people care about FSF bug reports than bugs reported here.  So I tend to work on the FSF bugs first, and then when I have time the github bugs.  The problem here is that I haven't really had time for github bugs in a while, so some of the stuff reported here isn't getting addressed.

I don't think that we need ful RV128 support before adding the missing hints.  But I would like to make sure that adding the missing hints now doesn't prevent us from adding full RV128 support latter.  This means we need some kind of a plan for how to do this, and I don't have one yet, as I've been too busy working on other stuff.

aswaterman, 2018-05-02:
Yes, I've confirmed that the only HINTs that are not [dis]assembled are the three shifts with a zero shift amount.  So there's not much to do until we take the time to understand the forward-compatibility implications of adding this feature.

I'll leave the simple test case here: https://gist.github.com/aswaterman/3431162dba909a546e4b402c4f6fb98a

cliffordwolf, 2018-05-04:
> the forward-compatibility implications of adding this feature

It's a bug. Not a missing feature.

The instruction is valid and the tools say it isn't. This is broken right now. People are using objdump as executable reference right now and get wrong results.

As I've said many times now, I can wait for a bug fix. But don't lie to yourself or to me and say it's not a bug, it's a missing feature, because clearly it is not. The tool claims to be able to decode compressed RV32/RV64 instructions and it does it incorrectly. That's pretty much the definition of a bug.

I would fix it myself, but not if I would have to maintain it as my own branch. People send me "bug reports" for PicoRV32 because their version of the tool has the bug and they trust the tool. It doesn't help me or them if I have a private branch with a bugfix.

jim-wilson, 2018-05-05:
I'm wondering what the assembly syntax should be for the c.slli64 and other related instructions.  This could be "c.slli64 a0" or "c.slli a0,64" or perhaps even something else.  I'm learning toward the first one at the moment, as the 64 is part of the opcode name in the manual, and it would be redundant to then include the shift amount as an argument.

aswaterman, 2018-05-05:
The most important thing to me is that “slli a0, a0, 64” remains a
pseudoinstruction that maps to c.slli64.

I have a slight affinity for “c.slli a0,64” since the syntax is more
regular, but I have no problem with “c.slli64 a0”, and I recognize it may
simplify the assembler a little.

jim-wilson, 2018-05-05:
On Fri, May 4, 2018 at 7:30 PM, Andrew Waterman <notifications@github.com>
wrote:

> The most important thing to me is that “slli a0, a0, 64” remains a
> pseudoinstruction that maps to c.slli64.
>

Yes, that is a given, provided that the register is not x0, which turns it
into a hint if compressed.  But since we have no rv128 support at present,
and I'm not proposing to add it at this time, this will have to wait until
later.  It is just the syntax for c.slli64 that I need now.

>
> I have a slight affinity for “c.slli a0,64” since the syntax is more
> regular, but I have no problem with “c.slli64 a0”, and I recognize it may
> simplify the assembler a little.

I suppose I could add both if I have to, but I would rather not if I can
avoid it.

Jim

jim-wilson, 2018-05-08:
I added a patch upstream to support the final 3 hint instructions that we were missing.

closed 2021-03-08 by lazyparser