riscv-collab/riscv-gnu-toolchain #311: why does gcc generate code with ebreak without any system calls being made

karimhasebou, 2017-12-25:


jim-wilson, 2017-12-25:
Can you provide a testcase?  Without a testcase I can only guess.

There is a compiler optimization enabled at -O2 that can insert ebreak instructions.  If the compiler can prove that you have a null pointer dereference, then the compiler knows that the program has undefined behavior, and will insert an ebreak to force the program to fail.  For example:

rohan:2022$ cat tmp.c
int
main (void)
{
  int *ptr = 0;
  return *ptr;
}
rohan:2023$ riscv64-unknown-linux-gnu-gcc -O2 -S tmp.c
rohan:2024$ cat tmp.s
	.file	"tmp.c"
	.option nopic
	.section	.text.startup,"ax",@progbits
	.align	1
	.globl	main
	.type	main, @function
main:
	lw	a5,0(zero)
	ebreak
	.size	main, .-main
	.ident	"GCC: (GNU) 7.2.0"
rohan:2025$ 

If you are programming in an embedded environment, and need null pointer dereferences to work for zero-page addressing, then you need to use -fno-delete-null-pointer-checks to tell the compiler that null pointer dereferences are valid on your target.  This disables several optimizations, including the one that inserts the ebreak instructions after a null pointer dereference.

closed 2018-01-01 by karimhasebou

karimhasebou, 2018-01-01:
@jim-wilson Thank you for responding. Unfortunately I am unable to provide the code I am currently working on.

rishikhan, 2018-01-19:
I see the same thing in libgomp. If you look at the dissassembly of priority_tree_remove() in priority_queue.c, you'll see it ends in an ebreak.

jim-wilson, 2018-01-19:
This is still not a bug.  The RISC-V behavior here is identical to the behavior of all other gcc ports.

ebreak is not for system calls.  ebreak is the user trap instruction.

Immediately before the ebreak is a NULL pointer dereference, and the ISO C standard says that the result of a null pointer reference is undefined.  But the libgomp code is not broken here.  GCC is an optimizing compiler.  It is doing aggressive optimizations to improve the performance of the code, and along the way it ends up with a code path with a null pointer dereference.  This code path is impossible to reach, but gcc does not know that, so it has to deliberately emit a null pointer dereference, and then it follows it with the ebreak instruction to force a trap if the load did not.

rishikhan, 2018-01-19:
Agree … I was just giving a ‘real-world’ reproducer.